tzapu/WiFiManager #461: shared webserver instance

tablatronix, 2017-11-19:
I am testing this already, but I am using a getter and a shared_ptr, not sure which way will be final.
Ill post my branch later, I was adding some callbacks and testing what kind of parameters and callouts we would want to for templating and sharing webserver control.

There are several options, make server public, add getter with shared_pointer, and af ew others.
I am unsure which is the best design pattern for this, or how we want to control ownership.

Thoughts welcome.

ageurtse, 2018-02-20:
Is there a way you could provide an example on how to run a shared webserver.

tablatronix, 2018-02-20:
Sure, when I figure out how

tablatronix, 2018-07-18:
dnsserver? dns server resolves domains

spilz87, 2018-07-18:
Hi I see 
As you can see I'm not a specialist ^^
You understand the idea : captive portal point to a "/captiveportal" page not to the "/" default page

spilz87, 2018-07-18:
An other idea could be to check if the client is connected to AP or from network , and link to different pages, if it is connected on AP the page as only few button which link to other specific pages, so the difference is only on the first page 
Do you see what I mean ? 
Maybe it's already what you do

tablatronix, 2018-07-18:
Yeah the idea is to share the webserver instance so you can access in your own code, you can do this now by just making it public, but I have not decided if that is the final route. It might be better to makde it a shared pointer so that memory gets cleaned up better.

spilz87, 2018-07-18:
A shared pointer allow you to keep the same webserver but the configuration of this webserver (.on(..) , etc.) is erase when captive portal is open, isn't it ? 
Do I understand well or I'm wrong ?
Let me know if it's not the good place to have this discussion

tablatronix, 2018-07-18:
I wouldn't think so, not sure what you mean. webportal is always on when configportal is on

spilz87, 2018-07-18:
Does the setupConfigPortal erase the setup of the server done in the setup() of the sketch ?

tablatronix, 2018-07-18:
Ah I see what you are saying, yes that resets the webserver. hmm you have to setup yours after that, wm would have to be setup to allow that to be skipped, it also gets torn down on closeconfigportal

tablatronix, 2018-07-18:
Curious , Did you add a new constructor to pass the webserver in , or use a setter?

I guess you could check by client ip, or get the clients attached to ap and check them for that ip request.

spilz87, 2018-07-18:
Not in WM, it was a test in a personal library 

Ok i will look for this function :)

spilz87, 2018-07-18:
If you remember the name, let me know ^^
Thanks

tablatronix, 2018-07-18:
https://github.com/esp8266/Arduino/issues/2681

spilz87, 2018-07-19:
Ok thanks,
I just test it, it works well, as expected only for client connected to AP. So it's possible to compare IP to know if it connected to AP or through network, and send a different first web page

spilz87, 2018-07-19:
I just test something, with my iPhone, I don't know if it's true with android and I didn't get captive portal on my computer, but the captive portal seems to request the "/hotspot-detect.html" 
Is it a standard ?...

tablatronix, 2018-10-12:
for now I am testing this out

* moving server to public, keeping it unique pointer
* adding `setWebServerCallback` that fires after webserver is setup and before route events are setup ( this allows user to override routes eg. /info )

testing added to DEV example