tzapu/WiFiManager #463: Add Parameters other than SSID and Password

MehtaTJ, 2017-11-23:
I am thankful to @tzapu  for giving this tremendous work. I want take some more parameters other than SSID and password from AutoConnectAP captive page. After selecting Configure Wifi if I want to add some more parameters, can anybody help me out there? I am novice programmer in HTML and C++. So requesting you to give detailed explanation

alx-1, 2017-11-23:
I did that with this project : https://github.com/reseauducommun/cociclo/tree/master/cociclo_transpiksel_eastern_bloc/software/cociclo_transpiksel_eastern_bloc

The first file in that directory has an example of latitude, longitude and userid added to the ssid + password. 

Good luck ++

tablatronix, 2017-11-23:
This is built in, see the parameters example.

JhonControl, 2017-12-12:
Hello, perform some tests adding parameters, maybe serve as a guide,
[](http://pdacontrolen.com/wifimanager-emoncms-oem-with-esp8266-temperature-1/)
 greetings

JhonControl, 2017-12-12:
http://pdacontrolen.com/wifimanager-emoncms-oem-with-esp8266-temperature-1/

jgarridc, 2017-12-19:
Hi, sorry about my english...
I have a problem, I added a new field in addition to pass and ssid. I am woriking with thinspeak to monitoring temperature and humity.

With de APIkey I dont have any problem, but if I want to put the channel number too, its not possible because de compilator says "**invalid conversion from 'char*' to 'long unsigned int' [-fpermissive**]"

This is te code I am using:

`Hi, sorry about my english...
I have a problem, I added a new field in addition to pass and ssid. I am woriking with thinspeak to monitoring temperature and humity.

With de APIkey I dont have any problem, but if I want to put the channel number too, its not possible because de compilator says "**invalid conversion from 'char*' to 'long unsigned int' [-fpermissive**]"`