tzapu/WiFiManager #465: What has gone wrong?

TinajaLabs, 2017-12-02:
Publish counter: 3044
Publish counter: 3045
Publish counter: 3046

Exception (3):
epc1=0x402120b7 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40000000 depc=0x00000000

ctx: sys 
sp: 3ffffc60 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffe00:  40211122 3fff1574 3ffea10f 3ffedda4  
3ffffe10:  40217600 00000301 40214f27 3ffffed0  
3ffffe20:  3ffee240 3ffedc68 3ffee360 00000014  
3ffffe30:  4021533d 3ffffed0 3ffedd90 3ffedc68  
3ffffe40:  3ffead14 00000001 40215389 3ffffed0  
3ffffe50:  616d754a 00576f6e 00040000 00000000  
3ffffe60:  3ffed930 00000000 4000050c 3fffc278  
3ffffe70:  40211b61 3fffc200 00000022 3ffffe8c  
3ffffe80:  00000000 3ffead14 ffffffcc 3ffead24  
3ffffe90:  3ffed1c8 0000002c 00000000 402160e8  
3ffffea0:  00000000 3fff1574 ffffffcc 00000000  
3ffffeb0:  00000000 3ffef188 00000000 3fffd9d0  
3ffffec0:  00000000 00000000 00000000 fffffffe  
3ffffed0:  00000000 00310b0b 00640100 00000046  
3ffffee0:  3ffead38 0000009b 3ffead4e 3ffead2c  
3ffffef0:  3ffead54 3ffead38 3ffead41 3ffead5f  
3fffff00:  00000000 00000000 3ffead65 3ffeadb9  
3fffff10:  00000000 3ffead97 00000000 00000000  
3fffff20:  00000000 00000000 0000002c 00000000  
3fffff30:  3fff0f94 40215bb3 3ffed1c8 3fff1574  
3fffff40:  00000000 3ffee360 3ffed1c8 3ffead14  
3fffff50:  3ffead14 000000c3 00000000 0000002c  
3fffff60:  00000000 000000c3 4020afcf 3ffed1c8  
3fffff70:  3ffead08 3fffdcc0 3ffe97f0 3ffe97f0  
3fffff80:  00000080 3ffed1c8 00000000 3ffef190  
3fffff90:  4020a8ff 3fffdab0 00000000 40208b77  
3fffffa0:  3ffe97f0 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

TinajaLabs, 2017-12-02:
The code I used to reveal the above error:

```
// AutoConnectWithReset-test1.ino
#include <FS.h>                   //this needs to be first, or it all crashes and burns...

#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager

long lastMsg = 0;
char msg[50];
int value = 0;

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println();

  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;

  //exit after config instead of connecting
  wifiManager.setBreakAfterConfig(true);

  //reset settings - for testing
  //wifiManager.resetSettings();

  //tries to connect to last known settings
  //if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP" with password "password"
  //and goes into a blocking loop awaiting configuration
  if (!wifiManager.autoConnect("AutoConnectAP", "password")) {
    Serial.println("failed to connect, we should reset as see if it connects");
    delay(3000);
    ESP.reset();
    delay(5000);
  }

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");

  Serial.println("local ip");
  Serial.println(WiFi.localIP());
}

void loop() {
  // put your main code here, to run repeatedly:

    long now = millis();
    if (now - lastMsg > 10000) {
      lastMsg = now;
      ++value;
      snprintf (msg, 75, "%ld", value);
      Serial.print("Publish counter: ");
      Serial.println(msg);
      // mqttClient.publish("home/anywhere/alarm-monkey/status", msg);
    }
}
```

TinajaLabs, 2017-12-02:
I've been dealing with this crash with wemos D1 mini's version 2.2.0 (ESP-12S), wemos.cc (ESP8266MOD), and non-wemos (ESP8266MOD).  I've got a couple of versions that are working OK with additional code for MQTT/PubSubClient.

I've seen the same behavior as others where a few devices fail at the same time.

I'm stuck.  What can I do to track down this problem.

Thanks,
Chris.

ldab, 2017-12-03:
When using the AutoConnectWithFSParameters.ino it crashed on me when changing the length of `char blynk_token[35]` to `char blynk_token[]`

Hope it helps.

TinajaLabs, 2017-12-05:
ldab,
Thanks for your suggestions.  I'm not using blynk but I am guessing you were storing config in FS?

re: Power supply - I use 5V 2A power supplies (basic cell phone styled charger) with my wemos d1 minis.  That should be enough for basic wifi usage.  I do have problems with power when trying to run a strand of 25 neopixels with any other sensors or wifi, etc.  But that's an ongoing issue not related to this.

One very strange behavior is that the crash also occasionally triggers a simultaneous crash in a few other wemos devices in my home.  Some are not even running wifimanager.  All are running mqtt (brokered through a raspi running node-red).  Very strange.

Again, thank you,
Chris.

ldab, 2017-12-05:
Yes, storing at FS.

I had a quick look into your code, even though I'm not too familiar with snprintf. You're storing value into msg[50] maybe it's exceeding msg size and crashing. Or value will eventually be bigger than 75bytes and crash?

TinajaLabs, 2017-12-05:
Oh.  Excellent catch.  I'll test tonight.  Thank you.

skx, 2018-01-08:
Did your test confirm that?  If so please do close the issue :)

closed 2018-01-26 by tablatronix