tzapu/WiFiManager #466: can't use lambda expression for setSaveConfigCallback

liebman, 2017-12-03:
in my `setup()` function I have:
```
    bool save_config = false;
    wm.setSaveConfigCallback([&]()
    {
        save_config = true;
    });
```
This fails with an error: `error: no matching function for call to 'WiFiManager::setSaveConfigCallback(setup()::__lambda1)'`

However I also have:
```
    wm.setAPCallback([](WiFiManager *)
    {
        configColor();
    });
```
And this works!

Any ideas?

DarkFox, 2017-12-19:
You need to call it like this:
```
    wm.setSaveConfigCallback([]()
    {
        save_config = true;
    });
```
Since setSaveConfigCallback doesn't take any arguments.

liebman, 2017-12-19:
@DarkFox correct, I used this where save_config is a local variable:
```
    bool save_config = false;
    wm.setSaveConfigCallback([&]()
    {
        save_config = true;
    });
```
It's supposed to capture the save_config variable so I don't need to make it global as its only used in this one function.  Maybe its really a tool chain bug?

DarkFox, 2017-12-19:
Yeah, I was in the same situation. I decided to just make it a static on the class. It's not like I've got multiples of it anyway. Though it is poor style.

tablatronix, 2018-02-01:
Should we not just add WiFiManager* to all callbacks?

liebman, 2018-06-19:
fixed with #584

closed 2018-08-06 by liebman