tzapu/WiFiManager #472: Auto-launching web page

bill-orange, 2017-12-13:
First off, thank you for the useful code.  I am trying to make my efforts accessible to non-hobbyists and this helps.  I do have a couple of questions.  

1. How the heck did you get the set-up page to launch automatically upon selection of the AP?  Can you direct me to the segment of you code that accomplishes the feat?  I would like to fork FSBrowser to do the same.

2. Does any portion of your code continue to run in loop() in the most basic example?  I have had an odd result with my iPhone using my existing code with your library.  Once my code is up and running all is well connecting from a WIN10 machine.  However, after a few minutes attempts to connect  to the iPhone are unresponsive.  Unless I made a subtle screw-up (away possible) , there were no code changes except for removing smartconfig and substituting in your library, applies in the most basic way.

At the beginning of loop() I added ping.ping() in an if statement that runs very 3 minutes and that seems to fix the problem.  There are lots of issues by many people with ESP8266 web server and especially the Async server becoming unresponsive but this is especially weird.  

IDE: Arduino IDE last non-beta release
Board: NODEMCU 9.0 @160 Mhz (with battery for stability)
Code; Its big, so here's a link   https://1drv.ms/f/s!AmXKqAwyCrbxh90Ne_yQ-f9rJaa0Tw 
What it should look like: http://orangecaweather.duckdns.org:8021

Thoughts?

tablatronix, 2017-12-14:
Captiveportal() function

jee1mr, 2017-12-19:
@bill-orange  The trick is, to redirect all DNS requests to your own IP and if the request came for another domain then rewrite the URL to your IP. This way the device understands that there's a captive portal running and the OS pops it up natively.

alexiszecharies, 2018-01-09:
how can i make it not pop up?

tablatronix, 2018-01-10:
In my dev branch
setCaptivePortalEnable

I hope to merge this fork soon

alexiszecharies, 2018-01-10:
I tried with your code and as you said it. Nevertheless, when i put the IP 192.168.4.1 in the browser it does not find a webpage. How am i supposed to send the data?

tablatronix, 2018-01-10:
Hmm, let me test

tablatronix, 2018-01-10:
I just committed a fix for that, thanks, sorry about that.

alexiszecharies, 2018-01-10:
Great, it worked. Thanks. There is a weird looking kind of button called No AP set. Should that be the behaviour?

tablatronix, 2018-01-10:
Screenshot? Dev is not well tested

tablatronix, 2018-01-10:
Do you mean the well status?

alexiszecharies, 2018-01-12:
[image]

alexiszecharies, 2018-01-12:
I meant the no AP set box

alexiszecharies, 2018-01-12:
Also, I am trying to run an ajax call from a webpage (not from this portal) like this:

$.ajax({
                method: 'POST',
                url: 'http://192.168.4.1/wifisave',
                data: { 
                    's': 'PFCdud3', 
                    'p': '' 
                },
                success: function(){
                    alert('wow');
                }
            });

This works in the ESP, it connects. But it returns: Failed to load http://192.168.4.1/wifisave: No 'Access-Control-Allow-Origin' header is present on the requested resource. Origin 'http://localhost:8000' is therefore not allowed access. Do you know how I can fix this?

tablatronix, 2018-01-13:
Yeah thats the status its called a well

tablatronix, 2018-01-13:
You are running this from another page? This is illegal cross domain access, there are ways around it google cors

alexiszecharies, 2018-01-13:
Yes, but I need to change code from the side of de ESP , right?

tablatronix, 2018-01-13:
Yeah not sure how you are gonna send that header without editing library, there are no callbacks available really.

tablatronix, 2018-01-13:
We could implement jsonp perhaps

alexiszecharies, 2018-01-14:
That is a really good idea

alexiszecharies, 2018-01-14:
Fixed it by adding

  server->sendHeader("Access-Control-Allow-Origin","*");
  server->sendHeader("dataType","jsonp");

in  WiFiManager::handleWifiSave.

alexiszecharies, 2018-01-15:
Do you know if it is not recommendable to use the function wifiManager.resetSettings() ?

closed 2018-01-26 by tablatronix