tzapu/WiFiManager #473: Thingspeak and wifi manager

jgarridc, 2017-12-17:
sorry about my english. I have a wifi thermostate running without problems using wifi manager, but I want to improve it. Now with my movil or tablet I can put the password, net name, **API KEY**, but its not possible to me to put the **CHANNEL** number of thingspeak, when I compile the arduino program says this "**invalid conversion from 'char*' to 'long unsigned int' [-fpermissive]**"

And I want give thaks to Tzapu for this wonderful SW.

Here my program:

`#include <FS.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>
#include <DHT.h>
#include <WiFiClient.h>
#include <WiFiServer.h>
#include <ThingSpeak.h> //librería conexión a thingspeak
#include<ESP8266WiFi.h> //librería esp8266 
//#define DHTTYPE DHT22
#define DHTTYPE DHT11
#include <DoubleResetDetector.h>
#include <LiquidCrystal_I2C.h>
#include <ArduinoJson.h>

#define DRD_TIMEOUT 100
#define DRD_ADDRESS 0

DoubleResetDetector drd(DRD_TIMEOUT, DRD_ADDRESS);

LiquidCrystal_I2C lcd(0x27, 16, 2);

const char* serverThingspeak = "api.thingspeak.com";

char myChannelNumber[8];   **-->Here I have the problem**
char myWriteAPIKey[17] = "myWriteAPIKey";
const int releCaldera = 16;
const int DHTpin = D4;
float histeresis = 1;
static unsigned long last_loop;
unsigned long int anteriorRefreshDatos = 0;
float temp= 0;
float humi= 0;
float tempDeseada = 00;
int calderaHabilitada;
int estadoRele = 0;
int contconexion = 0;

DHT dht(DHTpin, DHTTYPE);

WiFiClient client;
WiFiServer server(80);

          bool shouldSaveConfig = false;

          //callback notifying us of the need to save config
          void saveConfigCallback () {
          Serial.println("Should save config");
          shouldSaveConfig = true;
            }

void setup() {

          lcd.clear();
          lcd.begin(16,2);
          lcd.init();
          lcd.backlight();
          dht.begin();
          delay(10);
  
          pinMode(LED_BUILTIN, OUTPUT);
          Serial.begin(115200);
          Serial.println();

[... 234 lines cut ...]
      val = 1;
      } else if (req.indexOf("/tempDeseada00") != -1){
      tempDeseada = 00;
      val = 0;
      } else {
      Serial.println("peticion invalida");
      client.stop();
      return;      
      }

  // Set GPIO2 according to the request
  calderaHabilitada = val;
  client.flush();

  // Prepare the response
  String s = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n<!DOCTYPE HTML>\r\n<html>\r\nGPIO is now ";
  s += (val)?"high":"low";
  s += "</html>\n";

  // Send the response to the client
  client.print(s);
  delay(1);
  client.stop();
  Serial.println("Client disonnected");
  }

liebman, 2017-12-17:
I am assuming that your JSON config file specifies the value as a number and not a string, a string would be a quoted value - see this example: https://github.com/bblanchon/ArduinoJson/blob/master/examples/JsonParserExample/JsonParserExample.ino

FYI - this has nothing to do with WiFiManager.

jgarridc, 2017-12-18:
The Json config file its done by thingspeak, I supuse here "const char* serverThingspeak = "api.thingspeak.com";"

jgarridc, 2017-12-19:
How its possible to convert the unsigned long to a char*?

liebman, 2017-12-19:
Again I'll make an assumption.  That the this line is the you you get the error for:
```
ThingSpeak.writeFields(myChannelNumber, myWriteAPIKey);
```
Again, this is totally unrelated to WiFiManager and should be handled in a forum related to ThingSpeak.  In any case the issue is that your passing a character array, 'myChannelNumber',  to a function that expects an integer of some size.

There are functions that will convert these for you:
 * strtol()
 * atoi()

as well as other methods to do this conversion.  Some google searches for "string to int c" or "string to int c++" can give you more information.

FYI: You have marked many lines with "here I have the problem" without indicating the exact line that produced the error.  And the formatting is almost unreadable.

jgarridc, 2017-12-20:
OK, thanks a lot, sorry but I am not a programmer and I have a lot of problems with the code, but all help its good. I am goin to look for information about your answer and try, thank you very much,.