tzapu/WiFiManager #476: On Android 8.1 Pixel 2, the captive portal hangs on the main screen

udnaan, 2017-12-30:
I haven't had time to investigate further but the same seems to work on a windows computer. I'm guessing that it has something to do with how android is handling captive portal now. 

PS: I did see a line commented out in wifimanager class that says something about android captive portal. Not sure whether it's a legacy change that is no longer needed.

tablatronix, 2018-01-26:
can you provide logs from dev branch

closed 2018-01-26 by tablatronix

rohitjust24idpl, 2019-02-26:
**On Android 8.1.0**
1. Access point is appearing
2. Phone connects to AP
3. Portal is also opened quickly.
4. But the main page is hanged for few seconds and then after few seconds it appears.
5. This is quite a random behaviour and does not happens all the time. But problems appears once is 3-4 attempts.