tzapu/WiFiManager #478: Trouble with the captive portal if there is wifi credentials stored

eirikso, 2018-01-04:
This is a rather strange problem that I have managed to reproduce several times:

When connecting for the first time the portal pops up perfectly well. Fast and responsive. I choose a network and enter the password. The device restarts and connects to the wlan I entered.

Then I turn off that wlan. And reset the device. Because it can not find the wlan anymore it starts the config portal. But this time it is slow. It fails to load the buttons. The connection fails completely etc.

No matter how many times I restart etc the portal is slow and unstable.

So I upload a debug code that deletes the wifi credentials. Then upload my regular code again. And voila: the portal is fast and responsive. After testing this several times I added a WiFi.disconnect(true); inside WiFiManager.cpp. In the start of WiFiManager::setupConfigPortal(). So that the credentials is deleted if the device enters config mode.

[image]

Now it works every time.

I have tested on a WeMos D1 Mini and a WEMOS TTGO ESP8266. Actually I had the same kind of problem when testing the WiFiManager-port to ESP32 as well. 

Anyone else that have this problem?

ldab, 2018-01-04:
It happened to me as well, I'll try that and see if it helps.

tablatronix, 2018-01-04:
Known issue, dup

closed 2018-01-04 by tablatronix

eirikso, 2018-01-05:
Thank you. So it is not that strange and I am not alone...

srrsparky, 2018-05-10:
@tablatronix 
I am having this same issue. Is there a known issue number for this? I had a look through the open issues, but couldn't spot it.

tablatronix, 2018-05-16:
The issue is ap wont start or something cant find #