tzapu/WiFiManager #480: How to change the configured SSID and password

labajo, 2018-01-07:
Hi, thanks for your library.

I tried it and I configured a Wifi network properly, but I cannot change it to another one or remove or reset the SSID saved in the devices.

How can I remove that data in order to to left de device clean?

Thank you

labajo, 2018-01-07:
Sorry, I didn't see ```wifiManager.resetSettings();``` . Using this I managed to clean the Wifi credentials. The problem is that I cannot overwrite other Wifi credentials when I save them in the ConfigPortal.

tablatronix, 2018-01-07:
There is a bug that causes the new ap connection to bail if already connected, perhaps this is what you mean ?

labajo, 2018-01-07:
Yes, I think so. I saw the connection when I put the debug information visible. Thank you very much.

closed 2018-01-26 by tablatronix