tzapu/WiFiManager #481: Connecting to public wifi

ThePaceMaker, 2018-01-07:
I want to try and connect a ESP to a public wifi network that requires an email address and a code that is emailed to the user daily. After looking at the code, I can see that the connectWifi method only takes a ssid and password. Is there another method that allows for extra parameters to be added when trying to make the connection?

tablatronix, 2018-01-07:
It is probably just a web page, if its https your out of luck most likely.

adamsstephen, 2019-04-20:
I have similar question. Ap asks for email address and prompts for "connect" button to be pressed, is this possible with wifi manager?
I have included screen provided by AP support.
[image]

mahboud, 2023-04-01:
See https://github.com/tzapu/WiFiManager/issues/1513#issuecomment-1492893976