tzapu/WiFiManager #483: lwIP 2.0 and static IP - DNS issues

mdhiggins, 2018-01-10:
Using the new 2.4.0 ESP8266 Arduino release, the default lwIP settings were changed from 1.4 to 2.0

2.0 seems to introduce problems with DNS resolution when using static IP address settings

No DNS requests seem to work on 2.0 (though you can manually do a `WiFi.config()` and include the DNS server information

Additionally, after initial setup and inputting the static information if you try to active any DNS servers (such as mDNS, or ArduinoOTA) you get an Exception 28 error and the device reboots. Those services will work after rebooting if not running the initial configuration page however

Reverting to lwIP 1.4 resolves all these issues

tablatronix, 2018-01-11:
Can you confirm on staging as well?
I have been using this release gor a while and have not noticed this, i did see some lwip regressions though in the repo.

Do you have an example sketch to test?
Do you know of any issues references?

tablatronix, 2018-01-11:
https://github.com/esp8266/Arduino/issues/4078