tzapu/WiFiManager #485: NodeMCU is not connecting to Wifi after reset

netwein, 2018-01-17:
Hi,

I'm using WifiManager on my NodeMCU. After uploading the code to the NodeMCU using Arduino IDE the app works great and I'm having an Access Point which I'm configuring to use my home wifi and the NodeMCU connects to the wifi and sends the data great. Unfortunately, when disconnecting the NodeMCU from the power and then reconnecting it back the NodeMCU is not connecting to my wifi or starting the temporary AP. I've expected that it will remember the SSID and password for my wifi and connect directly to it.
Is it a bug? How can I debug this?

Thanks.

tablatronix, 2018-01-17:
setDebugOutput(boolean debug);

netwein, 2018-01-17:
I've based on the example code. 
The "debugOutput" is true by default. 
When I'm with Serial Monitor right after upload the code I can see the the logs in the serial monitor.
If I'm just connecting the NodeMCU to computer, I don't see anything in the serial monitor and I've tried by using some python script listening to the serial and all I'm getting is "1" printed in a new line...

tablatronix, 2018-01-17:
do you have a reset button ? open serial monitor, reset module you should get a bunch of stuff

netwein, 2018-01-20:
I've tried I'm not getting nothing in the serial monitor..

tablatronix, 2018-01-20:
What about your own serial.print commands?
Do you see those? Do you have serial.begin?