tzapu/WiFiManager #486: NodeMCU will not show SSID with many WiFi Networks near by

crashtestoz, 2018-01-20:
I am not sure if this is an issue with the NodeMCU v1 or the WiFiManager in Server mode but I noticed that if there are many WiFi networks near by, the device will not show the SSID of the WiFi Network.

If I move the device to a quitter part of the house it works without problems.

Any ideas?

tablatronix, 2018-01-20:
How many?

crashtestoz, 2018-01-21:
I counted 25-27 WiFi Networks with a mixture of 2.4Ghz and 5Ghz.

tablatronix, 2018-01-21:
Sounds pretty normal, I have ap scan issues at around 25-30, I think the scan actually times out on the esp hardware. But I will keep this open, and confirm.

mtiutiu, 2018-01-26:
With so many wifi networks around it's hard I guess even if you want to change the wifi channel. I made some changes with some defines for this great library so that at compile time you can change the wifi channel - I'll propose a PR when I have the time.

tablatronix, 2018-01-26:
Like a channel filter.. hmm that would certainly help