tzapu/WiFiManager #491: wifiManager.setSTAStaticIPConfig sink telegram connection

plcabgut, 2018-01-24:
hello
when i use wifimanager with static ip in this way

IPAddress ip          (192, 168,   1, 59);
IPAddress gateway(192, 168,   1,  1);
IPAddress mask     (255, 255, 255,  0);

> wifiManager.setSTAStaticIPConfig(ip, gateway, mask);

The esp8266 nodemcu with static ip don't receives nothing from telegram with universaltelegrambot library.
if I let wifimanager in dhcp mode then it works fine and receives every command from  telelegram
there might be an interaction between them. after several days trying diferents configuracion i've found that with this instruction everything go wrong. it lost the connection. 
i think that could be because of the  secure client. but i'm not sure.
The instruction _wifiManager.setSTAStaticIPConfig(ip, gateway, _mask);_ works great indeed because it put the ip statically right but then the other library dosent works propoery

```

#include <ESP8266WiFi.h>
#include <WiFiClientSecure.h>
#include <UniversalTelegramBot.h> //https://github.com/witnessmenow/Universal-Arduino-Telegram-Bot

//WIFIMANAGER
#include <DNSServer.h>            //Local DNS Server used for redirecting all rs to the configuration portal
#include <ESP8266WebServer.h>     //Local WebServer used to serve the configuration portal
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager WiFi Configuration Magic

#define BOT_TOKEN_LENGTH 46
char BOTtoken[BOT_TOKEN_LENGTH] =  "----MYTOKEN----";  

WiFiClientSecure client_secure;
UniversalTelegramBot *bot;

int Bot_mtbs = 1000; //mean time between scan messages
long Bot_lasttime;   //last time messages' scan has been done

IPAddress ip     (192, 168,   1, 59);
IPAddress gateway(192, 168,   1,  1);
IPAddress mask(255, 255, 255,  0);

void handleNewMessages(int numNewMessages) {
  for (int i = 0; i < numNewMessages; i++) {
    String chat_id = String(bot->messages[i].chat_id);
    String text_bot=bot->messages[i].text.substring(1,bot->messages[i].text.length());
    if (text_bot.equals("X")) {
      bot->sendMessage(chat_id, "X", "");
    }
    if (text_bot.equals("Y")) {
      bot->sendMessage(chat_id, "Y", "");
    }

    if (text_bot.equals("start") || text_bot.equals("help")) {
      String help= "/X \n";
      help+= "/Y";
      bot->sendMessage(chat_id, help, "");
    }
  }
}

void setup() {
  Serial.begin(115200);
  WiFiManager wifiManager;

 //IF I COMENT NEXT LINE IT WORKS FINE OTHERWISE DONT RECEIVES NOTHING
  wifiManager.setSTAStaticIPConfig(ip, gateway, mask);

  wifiManager.autoConnect("AP_door", "pwd");
  if (WiFi.status() == WL_CONNECTED) {
    bot = new UniversalTelegramBot(BOTtoken, client_secure);
    Serial.println("WiFi connected");
    Serial.print(WiFi.localIP());
    Serial.println("/");
  } 
}

void loop() {
  if (millis() > Bot_lasttime + Bot_mtbs)  {
    int numNewMessages = bot->getUpdates(bot->last_message_received + 1);
    while(numNewMessages) {
      Serial.println("got response");
      handleNewMessages(numNewMessages);
      numNewMessages = bot->getUpdates(bot->last_message_received + 1);
    }
    Bot_lasttime = millis();
  }
}
```

closed 2018-01-26 by tablatronix

evbelda, 2018-04-12:
Please, can you tell me why this issue is closed? There are any other thread solved about it?
I'm detecting same problem with WifiManager 0.12 and UniversalTelegrambot 1.1.0 and static IP

tablatronix, 2018-04-12:
Because there is no information about which esp lib or attempt to remove wifimanager and test with a basic esp static example. Also there is now a development branch to test against