tzapu/WiFiManager #493: Crash while in config Mode

swapnil823, 2018-01-27:
Thank you for such wonderful library.
Application crashes after some time. If i don't configure it in that time window and leave it open.

`*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 1
*WM: SET AP STA
*WM: 
*WM: Configuring access point... 
*WM: AutoConnectAP 
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Handle root
*WM: Request redirected to captive portal                                       
*WM: Handle root                                                                
*WM: Request redirected to captive portal                                       
*WM: Handle root                                                                
*WM: Request redirected to captive portal                                       
*WM: Handle root                                                                
*WM: Handle root                                                                
*WM: Request redirected to captive portal                                       
*WM: Request redirected to captive portal                                       
*WM: Request redirected to captive portal                                       
*WM: Handle root                                                                
*WM: Request redirected to captive portal                                       
*WM: Request redirected to captive portal                                       
*WM: Request redirected to captive portal                                       
*WM: Handle root                                                                
                                                                                
Exception (29):                                                                 
epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000
                                                                                
ctx: sys                                                                        
         sp: 3ffffc50 end: 3fffffb0 offset: 01a0                                
                                                                                
>>>stack>>>                                                                     
           3ffffdf0:  00000001 80000002 4023be9b 00000001                       
3ffffe00:  4022a5be 4022a4dc 3fff261c 40100ec6                                  
3ffffe10:  3ffeae15 401042c3 3ffee828 00000000                                  
3ffffe20:  00000018 3ffef260 3ffef1dc 00000000                                  
3ffffe30:  4022a68b 3ff20a00 3fff261c 3ffec9d2                                  
3ffffe40:  3ffec9b8 00000000 00000000 00000001                                  
3ffffe50:  00000005 088335b8 401021f2 3ffec9b8                                  
3ffffe60:  3ffec9d2 0000002c 00000008 401004f4                                  
3ffffe70:  3ffec9c2 3fff3b84 402271e4 3ff20a00                                  
3ffffe80:  3fff261c 3fffbfbc 00000015 402270b7                                  
3ffffe90:  3fffc278 40101f8c 3fffc200 00000022                                  
3ffffea0:  3ffeb6c0 fffffffe 40249fc8 00000000                                  
3ffffeb0:  40000f68 00000030 00000010 ffffffff                                  
3ffffec0:  40000f58 00000000 00000020 00000000                                  
3ffffed0:  3ffeeef0 7fffffff 00000000 3ffee470                                  
3ffffee0:  3ffee800 00000006 00000000 3fffdab0                                  
3ffffef0:  00000000 3fffdcc0 00000008 00000030                                  
3fffff00:  4022ed19 ffffffa8 00000000 00000000                                  
3fffff10:  402268f3 3ffec9b8 3ffee828 3fff261c                                  
3fffff20:  00000000 3ffec9c2 3ffec9d8 40222284                                  
3fffff30:  4023c98d 00000000 00000000 3ffec9c2                                  
3fffff40:  4023bb30 00000056 3ffec9ac 3fff3b84                                  
3fffff50:  3ffec9ac 00000000 00000008 4023b45a                                  
3fffff60:  3ffee828 3ffec9ac 3fffdcc0 3ffeaf08                                  
3fffff70:  3ffec9c8 3fff28c4 00000040 3ffeaf08                                  
3fffff80:  00000040 3ffee828 00000000 3fffdcb0                                  
3fffff90:  4023ad57 3fffdab0 00000000 40241ebf                                  
3fffffa0:  3ffeaf08 40000f49 3fffdab0 40000f49                                  
<<<stack<<<                                                                     
                                                                                
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)                                   
                                                                                
load 0x4010f000, len 1384, room 16                                              
tail 8                                                                          
chksum 0x2d                                                                     
csum 0x2d                                                                       
v4ceabea9                                                                       
~ld                                              `

tablatronix, 2018-01-31:
please try development branch and see if its reproducible

swapnil823, 2018-02-11:
Issue is resolved now. Thank you for your support. Turned out that one of the parameters being passed to library was null pointer.

closed 2018-02-11 by swapnil823