tzapu/WiFiManager #497: How to use WiFiManager only as a fallback if a certain network is not there?

probonopd, 2018-01-29:
I need to  use WiFiManager only as a fallback if a certain network (`homeNetwork`) is __not__ there.

However, if I do the following then apparently the credentials do not get saved and loaded correctly - after a reboot, `AutoConnectAP` is always spawned.

How to fix this?

```
#if defined(ESP8266)
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
#else
#include <WiFi.h>
#endif

#include <DNSServer.h>
#include <ESPAsyncWebServer.h>
#include <ESPAsyncWiFiManager.h>

AsyncWebServer server(80);
DNSServer dns;

void setup() {
  Serial.begin(115200);

  bool homeNetworkAvailable = false;

  // Check if the home network is available
  // it needs some special authentication mechanism through a captive portal
  // (not shown in this example)

  int n = WiFi.scanNetworks();
  randomSeed(ESP.getCycleCount());
  Serial.println("scan done");
  if (n == 0)
    Serial.println("no networks found");
  else
  {
    Serial.print(n);
    Serial.println(" networks found");
    for (int i = 0; i < n; ++i)
    {
      // Print SSID and RSSI for each network found
      Serial.println(WiFi.SSID(i));
      if (WiFi.SSID(i) == "homeNetwork") { // enter the ssid which you want to search
        Serial.println("The homeNetwork network is available, trying to log in");
        homeNetworkAvailable = true;

      }
    }
  }

  // ONLY if the home network is NOT available use AsyncWiFiManager
  // This works only ONCE, after a reboot I get "*WM: No saved credentials
  // FIXME

  if (homeNetworkAvailable == false) {
    AsyncWiFiManager wifiManager(&server, &dns);
    //wifiManager.resetSettings(); // reset settings - for testing
    wifiManager.setTimeout(180);

    if (!wifiManager.autoConnect("AutoConnectAP")) {
      Serial.println("failed to connect and hit timeout");
      delay(3000);
      ESP.reset();
      delay(5000);
    }
  }

  Serial.println("connected...yeey :)");

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

tablatronix, 2018-01-29:
Whats AsyncWiFiManager ?

probonopd, 2018-01-29:
Err, good point @tablatronix - I have mixed up the projects. https://github.com/alanswx/ESPAsyncWiFiManager is a fork of https://github.com/tzapu/WiFiManager/ but async. Closing here and asking there.

I suspect it has something to do with "Using last saved values, should be faster" - apparently doing a network scan confuses this. I would not be surprised if the same was true for https://github.com/tzapu/WiFiManager/ as well.

closed 2018-01-29 by probonopd