tzapu/WiFiManager #500: Document development release

tablatronix, 2018-02-02:
Create docs for changes, new features functions and known issues.

tablatronix, 2018-02-08:
`setConfigPortalBlocking`
`setShowStaticFields`
`setCaptivePortalEnable`
`setRestorePersistent`

`startWebPortal`
`stopWebPortal`
`process`

Testato, 2018-02-09:
Very good,
for the esp32 merge i think it is fundamental and, at least for me, it work very well.
Merge it in the lib will help to have new test/opinion.

Testato, 2018-02-09:
- adds signal icons, replaces percentage ( has hover titles )

On this change i do not agree, imho have the percentage is better than a simple icons whit only 4 graphic step, and on mobile lcd you cannot click for see the percentage number.
It is a very important signal, example for decide the post orientation of the Esp on a critical position far away from the wifi-router.

You should add a public function that the user can use on the .ino sketch for decide himself witch type of dsta he need ?

I repeat, imho ot is a very important info that the user should be decide whitout change the lib code.

tablatronix, 2018-02-12:
include `<rom/rtc.h>` to get esp32 info to show reset reasons

dvukovic, 2018-11-26:
Has anyone tried WiFiManager with an ESP8285 ?

tablatronix, 2018-11-26:
I have one I have not tested yet

dvukovic, 2018-11-26:
To give a little background.
I am using the Arduino1.8.5 IDE on Win7-64 bit.
The code has been compiled for an NodeMCU Lolin v3 and works properly.
I have a ESP8285 m3 board that I have been working with and have interfaces an I2C OLED display and an I2C RTC chip.
I then compile the same demo programs as the NodeMCU, but the 8285 does not what too connect to any of my devices, ( DEL laptop, iPhone or Samsung tablet) as the NodeMCU has done.
Here is a screen shot of the IDE Tools configuration.
[image]
Please let me know if you see anything I missed.
Thank You for your time.

tablatronix, 2018-11-26:
What does serial debug say?

dvukovic, 2018-11-26:
After loading the program, the serial monitor shows:
[image]
My iPhone and tablet shows the ESP8822174 as an access point, but neither are able to connect.
The above serial monitor does not change when attempting a connection.

dvukovic, 2018-11-26:
[image]

tablatronix, 2018-11-26:
maybe turn on esp debugging, so you are not able to join the ap at all?
what about the esp8266 softap example

dvukovic, 2018-11-26:
Thank You, I will create an issue.

dvukovic, 2018-11-26:
Do you have a link to a 'esp8266 softap example' or the file name if already included.

dvukovic, 2018-11-27:
I compiled for ESP8285:
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiAccessPoint/WiFiAccessPoint.ino

[image]

Still no connection:
[image]

tablatronix, 2018-11-27:
It looks like it joined and left, did you get a password prompt?

dvukovic, 2018-11-27:
Yes, but the Spinner Icon on iPhone just spins.
When I tried the Samsung tablet, it time out and exits.

tihoangyeudau, 2019-05-29:
When  I add wifimanager with custom static ip to esp8266 fastled webserver. wifimanager works and i can go 192.168.4.1 to config static ip wifi but when i done this, i cant access to webserver on the static ip. can you help me, please?

tablatronix, 2019-05-29:
post a new issue

noise5joker, 2019-06-12:
How can i add my own pages to the library captive portal?

kbickham, 2019-08-15:
I'm not sure if this is where I should post, but I'm getting some warnings on compilation. It still compiled but is untested.

the only flags it threw up are:

<details>
<summary>Details</summary>
'In file included from C:\Users\Lauren\Downloads\arduino-nightly\sketches\libraries\WiFiManager-development/WiFiManager.h:94:0,

                 from C:\Users\Lauren\Downloads\arduino-nightly\sketches\fuck2cores\fuck2cores.ino:13:

C:\Users\Lauren\Downloads\arduino-nightly\sketches\libraries\WiFiManager-development/strings_en.h:327:70: warning: missing initializer for member 'wifi_country_t::policy' [-Wmissing-field-initializers]

 const wifi_country_t WM_COUNTRY_US{"US",1,11,WIFI_COUNTRY_POLICY_AUTO};

                                                                      ^

C:\Users\Lauren\Downloads\arduino-nightly\sketches\libraries\WiFiManager-development/strings_en.h:328:70: warning: missing initializer for member 'wifi_country_t::policy' [-Wmissing-field-initializers]

 const wifi_country_t WM_COUNTRY_CN{"CN",1,13,WIFI_COUNTRY_POLICY_AUTO};

                                                                      ^

C:\Users\Lauren\Downloads\arduino-nightly\sketches\libraries\WiFiManager-development/strings_en.h:329:70: warning: missing initializer for member 'wifi_country_t::policy' [-Wmissing-field-initializers]

 const wifi_country_t WM_COUNTRY_JP{"JP",1,14,WIFI_COUNTRY_POLICY_AUTO};
'
</details>

I'll post my code, but looking at the header on the library points to thing where it includes "strings_en.h", and there it references the very end of the file. I compiled last night with this exact same set of decencies and ran okay.

tablatronix, 2019-08-15:
What version of core? Which esp chip?

kbickham, 2019-08-15:
it was v1.0.2, on a node mcu 32s. I'm using a wroom-32 for the final project but I'm trying to protect my baby. I just had to switch builds... i'm getting guru meditation errors unrelated to this, so I won't be able to test for at least 24 hours. I gotta make this stuff work.

tablatronix, 2020-01-27:
Closed this issue, this is for documentation only

tablatronix, 2020-05-18:
Known Issues/Outstanding Design

- [ ] Since autoconnect and startcp can be non blocking now, return results may be inconsistent with old code, ensure this is cleaned up and ocncise

- [ ] You can now skip wifi and save other params, wifi will be ignored, also this will alter return results as true was for connected status previously

 - [ ] save callbacks are not called if not changing wifi, unless shouldbreakafterconfig is set, this needs to be cleaned up in to a legacy handling and new handling consitancy

 - [ ] save wifi and save param can be split now, wifi still retains ip inputs, param custom params, there is no user selection here, improve where user wants params, fix up callbacks to make it clear where user is getting data back from.

- [ ] WM is NO longer persistent unless you save, atm I am reducing confusion by forcing STA mode on because that is how legacy worked, by design or not, it is what users are used to. Clarify how users want this to function, users now are responsible for user code and modes, wm can restore user modes back when starting and stopping configportal without touching their flash confgs. So if wifi was off, or ap, it will go back and they will wonder why their wifi is not on or why there is an ap when they reboot ???

- [ ] Documentation is non existent, I might work with doxygen to get some basic generated from my doc blocks, all public functions are documented fairly well. Source is Docs for now.

- [ ] DEV/ondemand example is used as my poor mans unit test, it has almost all functionality added to it and maintained.

 - [ ] Examples need to be cleaned up, I want them cleaned up into useful functions also, for example MQTT, with api token, durations, etc. littlefs

- [ ] spiffs is replaced by littlefs, update examples as such

tablatronix, 2020-12-16:
Release v2.0.3-apha Development
https://github.com/tzapu/WiFiManager/releases/tag/2.0.3-alpha

tablatronix, 2021-01-25:
- [ ] merge development into master
- [ ] delete development branch

tablatronix, 2021-01-28:
#1199

tablatronix, 2021-02-03:
Created https://github.com/tzapu/WiFiManager/wiki/Methods as basic outline for docs