tzapu/WiFiManager #501: AutoConnectWithFSParametersAndCustomIP -

kgreene1983, 2018-02-04:
Been stuck on this for a while, and I hope no one minds me posting here as a last resort.

I can't seem to get this sketch or the version without the custom IP to work. 

Once the sketch uploads the board creates an access point, but everything fails to connect to it. 

When I look at the serial monitor... it loops the following forever. (Not there some data from old project also appearing in there!) Can anybody help  

3⸮⸮⸮080 offset: 01a0

>>>stack>>>
3fff0ee0:  3fff3114 3fff31a0 3ffeff90 40208009  
3fff0ef0:  3ffe9a58 00000000 000003e8 feefeffe  
3fff0f00:  3fff23dc 3fff241c feefeffe feefeffe  
3fff0f10:  feefeffe feefeffe feefeffe feefeffe  
3fff0f20:  feefeffe feefeffe feefeffe feefeffe  
3fff0f30:  feefeffe feefeffe feefeffe feefeffe  
3fff0f40:  feefeffe feefeffe feefeffe feefeffe  
3fff0f50:  feefeffe feefeffe feefeffe feefeffe  
3fff0f60:  feefeffe feefeffe feefeffe feefeffe  
3fff0f70:  feefeffe feefeffe feefeffe feefeffe  
3fff0f80:  feefeffe feefeffe feefeffe feefeffe  
3fff0f90:  feefeffe feefeffe feefeffe 3ffeff90  
3fff0fa0:  0000007c 3fff3179 feefef0a feefeffe  
3fff0fb0:  feefeffe 3ffe8db0 feefeffe 3fff3194  
3fff0fc0:  00000200 feefeffe feefeffe feefeffe  
3fff0fd0:  feefeffe feefeffe feefeffe feefeffe  
3fff0fe0:  feefeffe feefeffe feefeffe feefeffe  
3fff0ff0:  feefeffe feefeffe feefeffe feefeffe  
3fff1000:  feefeffe feefeffe feefeffe feefeffe  
3fff1010:  feefeffe feefeffe feefeffe feefeffe  
3fff1020:  feefeffe feefeffe feefeffe feefeffe  
3fff1030:  feefeffe feefeffe feefeffe feefeffe  
3fff1040:  feefeffe feefeffe feefeffe feefeffe  
3fff1050:  feefeffe feefeffe feefeffe 3fff0050  
3fff1060:  3fffdad0 00000000 3fff0048 4020f794  
3fff1070:  feefeffe feefeffe 3fff0060 4010070c  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
vf6d232f1
~ld

mounting FS...
mounted file system
reading config file
opened config file
{"universe":0,"offset":4,"pixels":8,"leds":3,"white":0,"brightness":255,"hsv":0,"mode":0,"reverse":0,"speed":8,"position":1}
parsed json

Exception (28):
epc1=0x40212552 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

ctx: cont 
sp: 3fff0d40 end: 3fff1080 offset: 01a0

tablatronix, 2018-02-04:
erase flash using esptool

kgreene1983, 2018-02-04:
Thank you. It was driving me made.

closed 2018-02-05 by kgreene1983