tzapu/WiFiManager #502: Exeption 28: Exception 28: LoadProhibited when running Wifi Scan in Dev-Version

thomsenle, 2018-02-05:
Hey, 

very cool to see the development of the wifimanger. 
I played around a little with the development version and always get a Exeption 28 when i run the Webportal permanent. 

My main loop only consists out of
`  if (!webportalrunning) {
    wifiManager.startWebPortal();
    webportalrunning = true;
  } else     wifiManager.process();`

In the debug i can see that its scans for the networks and then crashes and reboots. Unfortunatly i have no idea where to start searching for the error.

I allready errased the flash totaly as suggested in a other issue. No change for me.

Thanks for your help, Thomas
Please let me know if you need more debug informations for testing. 

Debug output:
*WM: Starting Web Portal
*WM: HTTP server started
*WM: <- HTTP Wifi
*WM: Scan done
*WM: DUP AP: FRITZ!Box Fon WLAN 7270
*WM: DUP AP: Vodafone Hotspot
*WM: DUP AP: Vodafone Hotspot
*WM: DUP AP: Vodafone Hotspot
*WM: DUP AP: Vodafone Hotspot
*WM: DUP AP: Vodafone Homespot
*WM: DUP AP: Vodafone Homespot
*WM: DUP AP: Vodafone Homespot
*WM: DUP AP: Vodafone Homespot
*WM: AP: -57 boingboing
*WM: AP: -59 FRITZ!Box Fon WLAN 7270
*WM: AP: -67 Fritzbox-0815
*WM: AP: -67 Hallo Gast! WLAN
*WM: AP: -69 WLAN-B8BAHX
*WM: AP: -69 Telekom_FON
*WM: AP: -80 KabelBox-3760
*WM: AP: -80 Vodafone Hotspot
*WM: AP: -80 Vodafone Homespot
*WM: AP: -82 FRITZ!Box 6490 Cable
*WM: AP: -82 TP-LINK_AC1A06
*WM: AP: -87 NETGEAR88
*WM: AP: -88 KabelBox-364C
*WM: AP: -89 TP-LINK_B732
*WM: AP: -89 HITRON-3420
*WM: AP: -90 gaensebluemchen
*WM: AP: -91 Netzwerk_KW1509
*WM: AP: -92 vdr.fische.leipzig

Stack-Trace:

Exception (28):
epc1=0x402151f0 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000001 depc=0x00000000

ctx: cont 
sp: 3fff1080 end: 3fff1450 offset: 01a0

>>>stack>>>
3fff1220:  3fff12f0 00000004 3fff125c 402118c2  
3fff1230:  00000001 ff000000 3fff1274 4021179c  
3fff1240:  3fff12f0 00000004 3fff0040 4020db1d  
3fff1250:  00000000 00000000 00000000 3fff46cc  
3fff1260:  0000000f 00000003 3fff6104 40100648  
3fff1270:  00001404 3fff47f4 0000005f 00000050  
3fff1280:  3fff45fc 0000006f 00000069 33373031  
3fff1290:  00001500 3fff1314 3fff1314 402117f7  
3fff12a0:  ffffffff 0000000c 00000037 ffffffff  
3fff12b0:  00000011 00000046 00000060 00000065  
3fff12c0:  00000001 00000001 3fff12f0 4010068c  
3fff12d0:  00000001 00000001 3fff0040 4020b9ec  
3fff12e0:  00000001 00000001 3fff0040 4020ff47  
3fff12f0:  3fff4674 0000004f 0000004a 00000000  
3fff1300:  00000000 00000000 3fff600c 000000ef  
3fff1310:  000000ec 3fff6104 0000155f 00001555  
3fff1320:  00000001 00000001 3fff4284 40213664  
3fff1330:  00000000 00000000 3fff4284 4020b9e2  
3fff1340:  3fff4284 3fff38bc 3fff4284 4020ba1e  
3fff1350:  00000000 00000000 00000000 402119a0  
3fff1360:  3fff4284 3fff38bc 3fff387c 4020bab1  
3fff1370:  3fff47ac 0000000f 00000005 40209f10  
3fff1380:  3fff38bc 00000001 3fff0430 3fff38a0  
3fff1390:  00000001 00000000 0000000f 3fff4284  
3fff13a0:  00000000 00000000 3fff387c 3fff041c  
3fff13b0:  00000001 3fff38a0 3fff387c 4020bd10  
3fff13c0:  3fffdad0 00000000 3fff35ec 4020ad3d  
3fff13d0:  00000000 3fff4774 40212338 3fff0430  
3fff13e0:  3fffdad0 00000000 3fff0040 4020f8dd  
3fff13f0:  4020e3dc 00000000 3fff0040 402123f0  
3fff1400:  feefeffe 3ffe9b70 3fff27ec 4020a83f  
3fff1410:  3fffdad0 00000000 3fff00f8 4020f98b  
3fff1420:  00000000 00000000 00000001 40207194  
3fff1430:  3fffdad0 00000000 3fff0414 40212384  
3fff1440:  feefeffe feefeffe 3fff0430 4010070c  
<<<stack<<<

tablatronix, 2018-02-05:
Do you have a full sketch or can reproduce with example ?

thomsenle, 2018-02-05:
Yes, im running the following code with the artnet-lib made by rstephan.
 https://github.com/rstephan/ArtnetWifi

Used Version of the ESP8266 stuff for the IDE is 2.4.0-rc2.

`
/*
  Example, transmit all received ArtNet messages (DMX) out of the serial port in plain text.

  Stephan Ruloff 2016
  https://github.com/rstephan

*/
#if defined(ARDUINO_ARCH_ESP32)
#include <WiFi.h>
#else
#include <ESP8266WiFi.h>
#endif

#include <WiFiUdp.h>
#include <ArtnetWifi.h>
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>
bool webportalrunning = false;

WiFiManager wifiManager;

WiFiUDP UdpSend;
ArtnetWifi artnet;

void onDmxFrame(uint16_t universe, uint16_t length, uint8_t sequence, uint8_t* data)
{
  boolean tail = false;

  Serial.print("DMX: Univ: ");
  Serial.print(universe, DEC);
  Serial.print(", Seq: ");
  Serial.print(sequence, DEC);
  Serial.print(", Data (");
  Serial.print(length, DEC);
  Serial.print("): ");

  if (length > 16) {
    length = 16;
    tail = true;
  }
  // send out the buffer
  for (int i = 0; i < length; i++)
  {
    Serial.print(data[i], HEX);
    Serial.print(" ");
  }
  if (tail) {
    Serial.print("...");
  }
  Serial.println();
}

void setup()
{

  // put your setup code here, to run once:
  Serial.begin(115200);

  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;
  //reset saved settings
  //wifiManager.resetSettings();

  //set custom ip for portal
  //wifiManager.setAPStaticIPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));

  //fetches ssid and pass from eeprom and tries to connect
  //if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP"
  //and goes into a blocking loop awaiting configuration
  wifiManager.autoConnect("AutoConnectAP");
  //or use this for auto generated name ESP + ChipID
  //wifiManager.autoConnect();

  //if you get here you have connected to the WiFi
  Serial.println("connected...yeey :)");
  artnet.setArtDmxCallback(onDmxFrame);
  artnet.begin();
}

void loop()
{
  if (!webportalrunning) {
    wifiManager.startWebPortal();
    webportalrunning = true;
  } else     wifiManager.process();
  //artnet.read();
}
`

tablatronix, 2018-02-06:
cannot reproduce
Try moving WiFiManager wifiManager to global

MatteoGheza, 2021-10-30:
Having the same error (on a TTGO-T).
My code is similar to the code in the issue.
Here's a decoded stacktrace:
```
PC: 0x400014dc
EXCVADDR: 0x00000001

Decoding stack results
0x400ee42e: String::String(char const*) at C:\Users\ardui\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\WString.cpp line 35
0x400deae6: String::replace(__FlashStringHelper const*, char const*) at C:\Users\ardui\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/WString.h line 264
0x400e2545: WiFiManager::getParamOut() at C:\Users\ardui\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp line 1629
0x400e3179: WiFiManager::handleParam() at C:\Users\ardui\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp line 1305
0x4019c496: std::_Function_handler    (WiFiManager*)> >::_M_invoke(std::_Any_data const&) at c:\users\ardui\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0/functional line 600
0x400e0bca: std::function ::operator()() const at c:\users\ardui\appdata\local\arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-97-gc752ad5-5.2.0\xtensa-esp32-elf\include\c++\5.2.0/functional line 2271
0x400eb1a9: FunctionRequestHandler::handle(WebServer&, http_method, String) at C:\Users\ardui\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WebServer\src\detail/RequestHandlersImpl.h line 45
0x400eb216: WebServer::_handleRequest() at C:\Users\ardui\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WebServer\src\WebServer.cpp line 636
0x400eb395: WebServer::handleClient() at C:\Users\ardui\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WebServer\src\WebServer.cpp line 315
0x400e6fea: WiFiManager::processConfigPortal() at C:\Users\ardui\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp line 771
0x400e71c4: WiFiManager::process() at C:\Users\ardui\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp line 757
0x400d2eee: loop() at C:\Users\ardui\progetti\TallyArbiter\listener_clients\TTGO_T-listener/TTGO_T-listener.ino line 701
0x400ef360: loopTask(void*) at C:\Users\ardui\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\main.cpp line 23
0x40090b1a: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```