tzapu/WiFiManager #503: Serving pages from the filesystem

renegadeandy, 2018-02-05:
Can somebody offer an example sketch which serves the in memory pages, from the filesystem instead of the ones in memory?

I have read and tried changing the server to a SPIFFS server rather than a ESP8266WebServer, but the result is never stable. Please offer an example of how this can be done.

renegadeandy, 2018-02-05:
[WiFiManager_Test.zip](https://github.com/tzapu/WiFiManager/files/1697032/WiFiManager_Test.zip)

Upload files to spiffs flash memory and navigate to 192.168.4.1/otherpage.html

Assuming you have an otherPage.html it shuld work -- but the sketch is really unstable and rarely works! Why?

renegadeandy, 2018-02-08:
Help? Is there a community around this library?

tablatronix, 2018-02-09:
Nobody is gonna dl a zip to see your code

renegadeandy, 2018-02-09:
The code can be found here: https://bitbucket.org/renegadeandy/wifimanagerbug/

tablatronix, 2018-02-09:
maybe remove the wifimanager handlenotfound event ?

tablatronix, 2018-02-09:
oh nm you did

renegadeandy, 2018-02-09:
yep . :)

It's not that.

morrisho, 2020-05-29:
Well, I figured something out. I am able to use SPIFFS to load .html and .css files. I successfully did the root page and I am working on other pages too. However, I actually uses [khoih's](https://github.com/khoih-prog/ESP_WiFiManager) version for my mod. But the method is the same. This is the first time I contribute something on github so I am not sure how this works. Should I put my mod on a forked version or how does people normally do?

tablatronix, 2020-05-29:
fork, new branch with name of change

morrisho, 2020-05-30:
Hello, I put my mod [here](https://github.com/morrisho/WiFiManager_HtmlFile). I don't know much about html or css so I did my best I can moving the HTTP array stuff to html files. I am not sure how to do dynamic html content update so I basically dump the file text into string and replace {variables} with updated information. I googled around found that it seems AJAX can be used for this?