tzapu/WiFiManager #507: Esp8266 can't connect to wifi afeter deepsleep

acca84, 2018-02-06:
Hello, 
I have an ESP12E Amica board wich works well expected if i call esp.deepsleep(µs) AND is connected to a 3.7v battery to Vin pin.

It works well when powered by usb with or without deepsleep.
It works well on battery without deepsleep
But if i power it by battery and put it into deepsleep, it can't connect. Even if i power it back from usb.
I can see the following debug message in the serial console : 
`mounting FS...
mounted file system
reading config file
opened config file
{"domoticz_server":"192.168.1.73","domoticz_port":"8081","IDX":"1973","IDXINTER":"1964","IDXINTEROTA":"2058"}
parsed json
*WM: Adding parameter
*WM: domoticz_server
*WM: Adding parameter
*WM: domoticz_port
*WM: Adding parameter
*WM: IDX
*WM: Adding parameter
*WM: IDXINTER
*WM: Adding parameter
*WM: IDXINTEROTA
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
*WM: SET AP STA
*WM: 
*WM: Configuring access point... 
*WM: AutoConnectAP
*WM: password
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
`

then if i reconfigure it it reboots but can't connect.

I have to force a wifimanager reset to make it work
Could you help me please ?
Regards

tablatronix, 2018-02-06:
try development branch, no idea, probably one of the many bugs

acca84, 2018-02-06:
Ok, how tout do that please ?
Are tout talking about RC versions un library manager ?
Thx

acca84, 2018-02-06:
Ok. I found it

acca84, 2018-02-06:
I found my mistake, i had this code in order to reset by pushing a button :  
```
 if ( digitalRead(TRIGGER_PIN) == LOW ) {
    Serial.println("EFFACEMENT DES PARAMETRES");
    wifiManager.resetSettings();
  }
```

But i didn't pull up trigger_pin
i think voltage on trigger_pin was floating but sufficient to avoid reset, but voltage was not sufficient when powered on battery then a reset (unperfect ?) was done.
This was not a Wifimanager problem
sorry

closed 2018-02-06 by tablatronix