tzapu/WiFiManager #508: Sporadic problem with portal (Access Point)

fabiuz7, 2018-02-07:
Hi everyone,
I was working on little library to manage the wifi config, when I found this one (very good job, thanks @tzapu ).

Unfortunately, the Ap mode sometimes  doesn't work, in the sense that the client doesn't connect to the esp8266.

After a reading of the library, I discovered that you use the AP_STA mode. I don't understand why you are using this mode, because I think that should be better to separete the STA mode and the AP mode.

So I tried to rewrite few lines of the library and magically it starts to work perfectly.
The modification is very simple:
boolean startConfigPortal(...){
boolean  WiFiManager::startConfigPortal(char const *apName, char const *apPassword) {
  //setup AP
  if(WiFi.getMode()==WIFI_STA){
    WiFi.disconnect();
    // don't know if this delay is really necessary
    delay(100);
    WiFi.mode(WIFI_OFF);
    delay(100);
  }
  WiFi.mode(WIFI_AP);
  DEBUG_WM("SET AP");
........
}

Maybe my code is wrong and useless, but I'm curious about the mixed AP_STA mode :)

tablatronix, 2018-02-07:
this is already addressed in development branch

tzapu, 2018-02-07:
thank you @fabiuz7 but lately it's @tablatronix who has put an absolutely massive amount of work in
thanks

fabiuz7, 2018-02-07:
Sorry I have read your name in the github link ;) so thanks to  @tablatronix