tzapu/WiFiManager #509: ESP8266WebServer not closing after the connection to a new AP

fabiuz7, 2018-02-07:
Probably you already knowabout it, but then the WifiManager change to AP --> STA it doens't close the ESP8266WebServer, so my WebServer (in the main loop) can't restart properly. 

Are you going to fix it?

tablatronix, 2018-02-07:
What does "change from ap --> sta" mean?

fabiuz7, 2018-02-07:
I mean the mode of the WiFi: from Access Point mode to Station mode

tablatronix, 2018-02-07:
But when and where?

tablatronix, 2018-02-07:
Do you have example, are you testing on development branch ?

fabiuz7, 2018-02-07:
No, I'm not testing the dev branch, just the official Arduino version.

Assuming that the first connection to WiFi has already taken place,  I have my own Webserver the runs in the main loop. If the network disappear, yuor library is called again to switch in Access Point mode and get the new credentials. The user will put the new credentials. After this, the control flow go back to my main program ( the usual loop), but my WebServer is not going to restart. 

I checked your code and noticed that you never call server->close();  

(Sorry for poor English)

tablatronix, 2018-02-07:
ok I will look at maybe a hotfix, but there is pretty much NO development or maintenance being done on that version.

fabiuz7, 2018-02-07:
Ok, I will try the dev version, thanks again!

closed 2020-05-11 by tablatronix