tzapu/WiFiManager #510: deep linking libraries?

tablatronix, 2018-02-07:
So I usually use platformio, but I noticed that the arduino IDE does deep linking now.

Is it still necessary to include all dependencies in the sketch ?

wifimanager already includes them all, so we need these at all  in examples and documentation ?

```
#include <ESP8266WiFi.h>          //ESP8266 Core WiFi Library (you most likely already have this in your sketch)

#include <DNSServer.h>            //Local DNS Server used for redirecting all requests to the configuration portal
#include <ESP8266WebServer.h>     //Local WebServer used to serve the configuration portal
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager WiFi Configuration Magic
```

fabiuz7, 2018-02-07:
I had worked a lot on Arduino IDE in the last 2 months, I pretty sure that to can delete them from the example! However tomorrow i can try to compiled them without those #includes

tablatronix, 2018-02-08:
The esp32 branch has them removed already, if you want

fabiuz7, 2018-02-09:
Tested on Wemos D1 mini board, it works without any #include, but #include<WiFiManager.h>