tzapu/WiFiManager #511: Cannot reconnect to saved Wifi network after WiFi.disconnect() without first rebooting

horendus, 2018-02-10:
I'm finding that if I call WiFi.disconnect() and then try to reconnect with wifiManager.autoConnectNoConfigPortal() or wifiManager.autoConnect I fail to reconnect to the same wifi network again. 

The connection result are
*WM: Connection timed out
*WM: Connection result: 
*WM: 0

If I reboot my ESP8266 which calls wifiManager.autoConnectNoConfigPortal() at startup it connects without issue, so the wifi details are being stored.

If I call WiFi.begin(SSID,PWD) AFTER a WiFi.disconnect() I am able to connect without issue, but as you see I have had to specify the SSID and PWD instead of using the saved ones from WiFiManager.

Does anyone have a work around for this? My project needs to be able to enabled/disable wifi within the code and rebooting every time I want to reconnect isnt a good solution for this project.

tablatronix, 2018-02-10:
wifi.disconnect wipes config in esp8266

tablatronix, 2018-02-10:
also try development branch

horendus, 2018-02-13:
Changed WiFi.disconnect() to WiFi.mode(WIFI_OFF) 

Seems to allow me to disconnect wifi without loosing details, then re connect with wifiManager.autoConnect 

So far have had no issues during testing doing it this way, thoughts? Bad idea?

tablatronix, 2018-02-13:
you can do presistent(false); WiFi.disconnect()