tzapu/WiFiManager #512: Find port

gufur, 2018-02-11:
Hi! As many have said before me  "Exelent work". 
I want to receive and send UDP messages. For that I need the port number,  Udp.begin(localUdpPort);. Is there a function under wifimanager to get the port, or have you got any other advice to handle sending and receiving messages?
/ Gunnar

gufur, 2018-02-11:
PS. When you log on to a network via the wifimanager. Is it possible to specify a port for listening on. Or is the 4210 some how locked to ESP8266 ???

tablatronix, 2018-02-15:
What port ?

gufur, 2018-02-16:
I found out that the default port is 4210. It is the same port recommended
when you hard code SSID and password. When use the hard coded way you can
choose whatever port you feel like. I would like the same possibility in
the wifimanager.   Coding the 8266 you might need to use the port number
and used IP address in the coded application. Thats why I would like a
"command" to ask for that kind of data.
I am not an expert in network handling but when I designed my own
application, "behind" the server,  I needed the port number. Luckily enough
it was the default number so I solved it.
/ Gunnar

tablatronix, 2018-02-16:
wifimanager doesn't do anything with ports. What ports are you talking about ?
dnsserver?

closed 2018-03-06 by tablatronix