tzapu/WiFiManager #513: Esp32 testing

tablatronix, 2018-02-11:
The esp32 branch implements support for esp8266 and esp32, it is branched off development branch.

The espressif esp32 platform does not include webserver and has bugs in dnsserver.
I have a fork here that has all the needed dependancies here https://github.com/tablatronix/arduino-esp32

If you use dnsserver from espressif repo it does not work with captiveportals.

tablatronix, 2018-02-11:
#241

gepd, 2018-02-13:
I'll let here a funtional example to compile the esp32 branch with PlatformIO, it includes the patched libraries `webserver` and `dnsserver`. The `WiFiManager` library will be downloaded automatically. (needs git installed)

I hope this can save some time to anyone who like to test the branch.

[WiFiManager_esp32.zip](https://github.com/tzapu/WiFiManager/files/1718682/WiFiManager_esp32.zip)

tablatronix, 2018-02-15:
merged into development!

gepd, 2018-02-17:
I've been testing the library so far it's very stable. 

I've just missing an option to configure the hostname. I've manually solve this with:

```cpp
void WiFiEvent(WiFiEvent_t event){
    switch(event) {
        case SYSTEM_EVENT_AP_START:
            WiFi.softAPsetHostname(HOSTNAME);
            break;
        case SYSTEM_EVENT_STA_START:
            WiFi.setHostname(HOSTNAME);
            break;
        default:
            break;
    }
}

void setup(){
    WiFi.disconnect(true);
    WiFi.onEvent(WiFiEvent);
    WiFi.begin();

// checks connection and perform a call to the config portal if it's neccessary
}
```

I've update `handleWifi` to get the list of available networks asynchronously, that way the UI loads within a second and then shows the list, I'll send a PR if you are interested

The last thing I'm missing is an option to display the reason why the connection wasn't able to be perform in the last try (STA mode). You will no always have access to read the debug output.

What could be the best way to implement this?
EEPROM, SPIFFS?

tablatronix, 2018-02-17:
Yeah that was on my list of todos

tablatronix, 2018-02-17:
Do you need to store it, or can it be a lifetime variable with a getter?

tablatronix, 2018-02-17:
The goal was to get esp32 working, then add additional features as needed and add special support for esp32 stuff that is not supported in esp8266 and also polyfill some stuff that is

tablatronix, 2018-02-17:
Feel free to create issues for that because I have not had a chance to work much with it.

gepd, 2018-02-18:
I've create two issues, and will send a PR for the asynchronous fix

tablatronix, 2018-03-05:
DNS Server has been fixed in esp32 core

twinotter, 2018-04-09:
I'd like to try this, but I'm a bit confused by the mulitple webserver approaches I've found online. Tablatronix, can you tell me how you set up your webserver library? Thanks!

tablatronix, 2018-04-09:
You need https://github.com/bbx10/WiFiManager/tree/esp32

Stick in esp32 libraries

tablatronix, 2018-04-09:
No idea, you need to use development branch of wifimanager...

tablatronix, 2018-04-09:
Shit i posted the wrong link, so sorry for wasting your time.

Doh
Use this repo/development
And 
https://github.com/bbx10/WebServer_tng

tablatronix, 2018-04-09:
I updated first post for clarity

twinotter, 2018-04-09:
It looks like the missing user_interface.h is related to a fairly different structure of the esp32 library from the esp8266 library. I installed the 8266 library and found the file in the tools/sdk folder. The equivalent folder in the esp32 folder structure doesn't have user_interface.h, os_types.h, and many others...

Perhaps this is a fundamentally different way that the esp32 libraries were installed? I used the Windows install path and everything seem to be updated... and I've compiled and run code on an ESP32 dev kit C... Is there a step I'm missing?

tablatronix, 2018-04-09:
Are you using development branch? Esp32 doesnt use this file that I know of, what is including it?

twinotter, 2018-04-09:
!! Okay, so my understanding of git is clearly very limited. Not sure how to use the gui to checkout a branch. But I found a command-line that would do it and used that, so now I'm at least looking at the same files. :)

`git clone https://github.com/tzapu/WiFiManager.git --branch development --single-branch WiFiManager`

btw, the way NOT to do it: don't select the branch from the main git page and then expect that the clone link has been updated. :)

That portion of the code compiled and I'm off working on other bugs now. Thank you!

tablatronix, 2018-04-09:
Yeah the clone does the repo not a branch, to do a branch select development from be branch dropdown and download the zip. Or use girhub desktop and just change branch.

tablatronix, 2018-06-11:
Its larger... and not official either

naveenneog, 2018-07-10:
ESP32 WebServer is now available in the core PR#430 is closed