tzapu/WiFiManager #514: crash on wemos and sample firmware

pat1, 2018-02-12:
I use wemos D1 mini and pro with Arduino core for ESP8266 git master branch and Lwip v2. 
With my firmware and also with AutoConnect example I get Exception (3)

This is the tail trace for AutoConnect example with debug activated:

Request: /
 Arguments: 
*WM: Handle root
New client
method: GET url: /wifisave search: s=pat1&p=mypassword
headerName: Host
headerValue: 192.168.4.1
headerName: Referer
headerValue: http://192.168.4.1/wifi?
headerName: User-Agent
headerValue: Mozilla/5.0 (X11; Fedora; Linux x86_64) AppleWebKit/605.1 (KHTML, like Gecko) Version/11.0 Safari/605.1
headerName: Accept
headerValue: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
headerName: Upgrade-Insecure-Requests
headerValue: 1
headerName: Accept-Encoding
headerValue: gzip, deflate
headerName: Accept-Language
headerValue: it-IT
headerName: Connection
headerValue: Keep-Alive
args: s=pat1&p=mypassword
args count: 2
pos 0=@ 1 &@ 6
arg 0 key: s value: pat1
pos 7=@ 8 &@ -1
arg 1 key: p value: mypassword
args count: 2
Request: /wifisave
 Arguments: s=pat1&p=mypassword
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
wifi evt: 2
wifi evt: 7
wifi evt: 7
wifi evt: 0
wifi evt: 3
*WM: Connection result: 
*WM: 3

Exception (3):
epc1=0x4010044e epc2=0x00000000 epc3=0x00000000 excvaddr=0x40246804 depc=0x00000000

ctx: cont 
sp: 3fff0ac0 end: 3fff0e90 offset: 01a0

>>>stack>>>
3fff0c60:  00000100 40004aa0 00000100 003ff000  
3fff0c70:  00000000 400042db 3fffc718 003ff000  
3fff0c80:  3ffefd7c 0000036e 0000036e 4010020c  
3fff0c90:  00000020 40248440 3fff2f24 401009a8  
3fff0ca0:  3ffefd7c 0000031a 0000031a 4010020c  
3fff0cb0:  00000000 00000000 3fff2104 4010053a  
3fff0cc0:  00000000 00000000 3fff1ddc 4010099c  
3fff0cd0:  3ffefd7c 00000300 3fff1ddc 4020a20c  
3fff0ce0:  3ffefd7c 000002f3 3fff1ddc 40204f79  
3fff0cf0:  00000000 00000000 3fff1de0 40211d33  
3fff0d00:  00000000 3fff20f8 3fff2b54 00000000  
3fff0d10:  00000000 00000000 3fff1ddc 4020501a  
3fff0d20:  3fff1d2c 00000003 3fff1ddc 4020595f  
3fff0d30:  40239e46 3ffefb58 3fff1ddc 402059cf  
3fff0d40:  402027af 40245d2c 3fff0de0 4020a822  
3fff0d50:  00000000 3ffefb58 3fff0de0 40208b40  
3fff0d60:  00000000 00000000 00000000 00000000  
3fff0d70:  00000000 00000000 3fff18dc 401009a8  
3fff0d80:  3ffe97f0 40245d4d 3fff0dac 00000000  
3fff0d90:  00000000 3ffe8a60 3fff0de0 40208bf5  
3fff0da0:  00000000 00000000 00000000 00000000  
3fff0db0:  00000000 00000000 3ffefda0 402091ec  
3fff0dc0:  0001c200 00000000 3fff0de0 3ffefe60  
3fff0dd0:  3fffdad0 00000000 3ffefda0 40202217  
3fff0de0:  3fff1d6c 00000000 3ffe8a60 00000000  
3fff0df0:  3fff2724 0000000f 00000004 3fff2744  
3fff0e00:  0000000f 00000008 00000000 00000000  
3fff0e10:  00008fa0 40107458 00000000 40107458  
3fff0e20:  00000000 40107458 00000000 40107458  
3fff0e30:  00000000 40107458 00000000 40107458  
3fff0e40:  00000000 00000000 ffffffff fe000001  
3fff0e50:  3ffe97f0 00000000 fe010035 00000000  
3fff0e60:  00000000 0000000a 3fff2764 feefeffe  
3fff0e70:  feefeffe feefeffe 3ffefe58 4020a303  
3fff0e80:  feefeffe feefeffe 3ffefe70 40100a2c  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,7)

 ets Jan  8 2013,rst cause:4, boot mode:(1,7)

wdt reset

tablatronix, 2018-02-12:
Use development branch

tablatronix, 2018-02-12:
Can you try decoding the exception with the arduino tool?

pat1, 2018-02-12:
The trace below happen with WiFiManager development branch Lwip version 1.4 and debug disabled. 
Decoding the exception  is not so easy due to this bug:
https://github.com/me-no-dev/EspExceptionDecoder/issues/20

*WM: <- HTTP WiFi save 
*WM: Method: POST
*WM: Sent wifi save page
*WM: Connecting save WiFi
*WM: Connecting as wifi client...
*WM: wifi station disconnect
*WM: Connecting to new AP
*WM: enableSTA PERSISTENT ON
*WM: connectTimeout not set, ESP waitForConnectResult...
*WM: Connection result: WL_CONNECTED
*WM: Connect to new AP SUCCESS

Exception (3):
epc1=0x4010010e epc2=0x00000000 epc3=0x00000000 excvaddr=0x4000e8d1 depc=0x00000000

ctx: cont 
sp: 3fff0370 end: 3fff0760 offset: 01a0

>>>stack>>>
3fff0510:  3fff38c4 3fff148c 0000e648 3fff392c  
3fff0520:  3ffef64c 00003b2b 00003b2b 4010020c  
3fff0530:  00000003 5e5f2888 5e5f2894 40100690  
3fff0540:  3ffef64c 000001aa 000001aa 4020aa18  
3fff0550:  3ffef64c 000001a0 3fff124c 40204994  
3fff0560:  00000000 00000000 3fff1c7c 40100690  
3fff0570:  00000003 3fff1cd0 3fff1c7c 00000000  
3fff0580:  00000003 00000000 3fff124c 40204a46  
3fff0590:  00000001 3fff1274 3fff124c 4020536b  
3fff05a0:  40106c68 00000000 3fff124c 402053db  
3fff05b0:  00000000 00000000 3fff0680 4020bb42  
3fff05c0:  00000003 00000003 3fff0680 40207f37  
3fff05d0:  00000000 00000003 3fff0680 4020912e  
3fff05e0:  00000000 00000000 00000000 00000000  
3fff05f0:  00000000 00000000 3fff0680 00000000  
3fff0600:  00000000 00000000 3fff0680 40209cfd  
3fff0610:  3ffe8d4f 402029a0 3fff0680 3ffe8b00  
3fff0620:  00000000 00000001 3fff0680 40209da5  
3fff0630:  00000000 00000000 00000000 00000000  
3fff0640:  00000000 00000000 3fff0680 4020a734  
3fff0650:  3ffef428 40248e15 3fff0680 40207c24  
3fff0660:  0001c200 00000000 3ffef428 3ffef734  
3fff0670:  3fffdad0 00000000 3ffef674 40202077  
3fff0680:  3fff110c 00000000 40106dd8 00000000  
3fff0690:  40106dd8 00000000 40106dd8 00000000  
3fff06a0:  40106dd8 00000000 40106dd8 00000000  
3fff06b0:  40106dd8 00000000 feefef35 3fff1924  
3fff06c0:  0000000f 0000000d 3fff193c 0000000f  
3fff06d0:  00000000 3fff1954 0000000f 00000004  
3fff06e0:  3fff196c 0000000f 00000008 00000000  
3fff06f0:  00000000 0001ea6d 0001ea75 00000001  
3fff0700:  ffffffff 01000001 00010100 feefef01  
3fff0710:  3ffe8d4f 00000000 01000000 01000000  
3fff0720:  0001de89 00000000 00000005 3fff1984  
3fff0730:  feefef01 3ffef674 00000000 00000000  
3fff0740:  feefeffe feefeffe 3ffef72c 4020b5c4  
3fff0750:  feefeffe feefeffe 3ffef740 40100710  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

tablatronix, 2018-02-12:
Thanks, this is on master? Ill try to reproduce, i have been using it all week with no problems.

Have you tried erasing flash with esptool?

pat1, 2018-02-12:
> Thanks, this is on master?
as reported the first is on master the second on development branch.
for erasing flash I do not use esptool but SPIFFS.format() and wifiManager.resetSettings()

tablatronix, 2018-02-13:
I mean esp stable or staging

pat1, 2018-02-13:
esp git version master branch up-to-date

tablatronix, 2018-02-13:
osx serial not working

```
reloading settings Packages/User/deviot.sublime-settings
Exception in thread Thread-519:
Traceback (most recent call last):
  File "./python3.3/threading.py", line 901, in _bootstrap_inner
  File "./python3.3/threading.py", line 858, in run
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/serial.py", line 94, in start
    if is_available(self.port):
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/serial.py", line 182, in is_available
    serial.open()
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/pyserial/serialposix.py", line 271, in open
    self._reconfigure_port(force_update=True)
  File "/Users/shawn/Library/Application Support/Sublime Text 3/Packages/Deviot (Arduino IDE)/libraries/pyserial/serialposix.py", line 373, in _reconfigure_port
    if self._stopbits == serial.STOPBITS_ONE:
NameError: global name 'serial' is not defined
```

tablatronix, 2018-02-13:
reproduced, seems to be
  server.reset();
causing this

tablatronix, 2018-02-13:
Now I cannot reproduce, might be a combination of hardware or router beacons, there is a open issue that mentions this. I will keep looking into it and find time to decode my exceptions.

tablatronix, 2018-02-13:
this code is new
```

  if (_params != NULL){
    DEBUG_WM(F("freeing allocated params!"));
    free(_params);
  }
```

maybe try removing it and see if it helps ?

jp112sdl, 2018-02-13:
I have the same issue also on an old version of WifiManager.

And I am also using a wemos D1 mini with Arduino core for ESP8266 git master branch.

server.reset() causes an exception.

It does not make a difference if I am using lwip2 or 1.4

tablatronix, 2018-02-13:
yeah its weird it does not always do it for me, I wonder if it has to do with the socket still being in use..

tablatronix, 2018-02-14:
can you try this fix?

https://github.com/esp8266/Arduino/issues/4350

jp112sdl, 2018-02-14:
Works for me! Thanks!

tablatronix, 2018-02-15:
Good news this has been fixed

closed 2018-02-15 by tablatronix