tzapu/WiFiManager #518: Memory optimization

tablatronix, 2018-02-14:
optimize memory usage, namely RAM

- [ ] store strings in PROGMEM
- [ ] minimize use of `Strings`
- [ ] profile memory for leaks
- [ ] optimize html outputting, is it possible to flush to reduce `page` strings size, does it matter ?

tablatronix, 2018-02-14:
Made sure all debug and strings were wrapped in F(), eventually all debug strings will be moved to strings.h with precompiler flags to remove them entirely.

```
development

BEFORE
Checking program size
text	   data	    bss	    dec	    hex	filename
341992	   9648	  30192	 381832	  5d388	.pioenvs\nodemcuv2_dev\firmware.elf
Uploading 355792 bytes from .pioenvs\nodemcuv2_dev\firmware.bin to flash at 0x00000000

freeheap 37760

data  : 0x3ffe8000 ~ 0x3ffe852c, len: 1324
rodata: 0x3ffe8530 ~ 0x3ffea5b4, len: 8324
bss   : 0x3ffea5b8 ~ 0x3fff1ba8, len: 30192
heap  : 0x3fff1ba8 ~ 0x3fffc000, len: 42072

AFTER

Checking program size
text	   data	    bss	    dec	    hex	filename
345448	   8776	  30200	 384424	  5dda8	.pioenvs\nodemcuv2_dev\firmware.elf
Uploading 358368 bytes from .pioenvs\nodemcuv2_dev\firmware.bin to flash at 0x00000000

data  : 0x3ffe8000 ~ 0x3ffe852c, len: 1324
rodata: 0x3ffe8530 ~ 0x3ffea24c, len: 7452 -872
bss   : 0x3ffea250 ~ 0x3fff1848, len: 30200 +8
heap  : 0x3fff1848 ~ 0x3fffc000, len: 42936 +864

free heap 38624
```

tablatronix, 2018-02-14:
data  : 0x3ffe8000 ~ 0x3ffe852c, len: 1324
rodata: 0x3ffe8530 ~ 0x3ffea23c, len: 7436
bss   : 0x3ffea240 ~ 0x3fff1838, len: 30200
heap  : 0x3fff1838 ~ 0x3fffc000, len: 42952
*WM: getCoreVersion():          00000000
*WM: system_get_sdk_version():  2.1.0(deb1901)
*WM: system_get_boot_version(): 31
*WM: getFreeHeap():             38640

horendus, 2018-02-15:
Fantastic :)

tablatronix, 2019-11-12:
Investigate fragmentation, suspect is page html and params replace strings, user params

nikeee, 2020-08-24:
Using the development branch on my ESP32 on Platformio:
With WiFiManager:
```
RAM:   [===       ]  25.7% (used 84296 bytes from 327680 bytes)
Flash: [======    ]  60.7% (used 1113774 bytes from 1835008 bytes)
```

And without WiFiManager:
```
RAM:   [==        ]  19.9% (used 65348 bytes from 327680 bytes)
Flash: [====      ]  39.7% (used 728388 bytes from 1835008 bytes)
```
(compiled in release mode with optimizations)

Which is 385386 (~385kB) difference in flash.
I remember that the master branch was more lightweight when I used it for an ESP8266.

Is there anything I can do on my side to reduce memory usage?

tablatronix, 2020-08-25:
hmm, well all debugging is left in, they are not conditional, so removing all debug statements might help. There might be some cruft in dev since its still alpha that can be removed.. also there is a lot of info page stuff that I intend to make optional somehow, compiler directives for now probably.

nikeee, 2020-08-26:
That seems like a good idea.

What I was also thinking of is offering the user the option to supply his own HTML and CSS (not just the ability to append CSS, but all if the HTML/CSS). For my current use-case that would be easier than overwriting the existing CSS while being more compact.

tablatronix, 2020-08-26:
There IS a define check for the strings, so you can do that at compiler level if you wanted to and replace them elsewhere, problem is its still flash, in order to switch to littlefs/spiffs strings it would require code changes.

nikeee, 2020-08-26:
Do you mean this define: https://github.com/tzapu/WiFiManager/blob/development/strings_en.h#L16?

Despite the comment, does defining an own set of strings work by something like this?
```c
#define WIFI_MANAGER_OVERRIDE_STRINGS
const char HTTP_HEAD_START[] PROGMEM = // ...
// ...

#include <WiFiManager.h>
```

I feel like I'd more customization options if this function would be `virual`, so the WiFiManager could be inherited and this function could be implemented by the library user:
https://github.com/tzapu/WiFiManager/blob/31be25d54bfa814b7a51ba8fbc8b6019682f6ac8/WiFiManager.cpp#L945-L963

This might not solve the RAM/Flash issue, though.

tablatronix, 2020-08-27:
Yeah, this was the first step to abstract and organize the code so it can be refactored better. That func was one of the latest additions with that in mind.

dalbert2, 2023-04-04:
For ESP8266 users, IRAM is a critical resource; the SSL library requires a lot of free heap and anything that reduces it is a problem.  The WiFiManager is amazing, but it consumes roughly 2.3K of IRAM _even when not instantiated_.

Are there ways to reduce WiFiManager IRAM usage when not using (or even instantiating) the WiFiManager?

tablatronix, 2023-04-05:
I am assuming this is all the strings, specifically the html, try removing the HTTP_STYLE css like icons and stuff and see what you find

dalbert2, 2023-04-14:
I submitted a separate issue #1587 addressing at least two problems:

1. string arrays in wm_consts_en.h aren't properly placed in PROGMEM
   I provided examples of how to do it properly and can submit a fix if desired, but the more serious issue is: 
2. more seriously, all of the strings are declared in header files (which are themselves nested)
   so any code that includes WiFiManager.h also creates instances of those strings.
   Why aren't those strings declared in a compilation module and declared extern in header files (if needed)?

tablatronix, 2023-04-15:
#2 but there are include guards..?