tzapu/WiFiManager #519: Crash Exception(28) with ArduinoOTA() in sketch setup()

gibo77, 2018-02-15:
I am getting exception(28) with Stack Dump.

When I run just the examples, I do not get any crashing. It works fine.
But If I incorporate to ArduinoOTA, it crashes.
The ArduinoOTA example works fine by itself. But combined with WifiManager it crashes.
As per serial monitor, the WifiManager was able to connect successfully. But when the sketch executes the ArduinoOTA, it crashes.
 Do you think there is an overlapping memory/EEPROM here?

I have not done any stack trace. Maybe the gurus can do that below.
Below is the stack dump. Thanks!

*WM: Sent config page
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 3
connected...yeey :)
IP address: 192.168.0.108

Exception (28):
epc1=0x40227b83 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000098 depc=0x00000000

ctx: cont 
sp: 3fff0920 end: 3fff0e40 offset: 01a0

>>>stack>>>
3fff0ac0:  3fffc278 401022a4 3fffc200 00000022  
3fff0ad0:  40221fb0 3fff2384 3fff1fb4 3fff35f8  
3fff0ae0:  3fff30ca 00000030 00000016 ffffffff  
3fff0af0:  400044ab 3fffc718 3fff0be0 08000000  
3fff0b00:  3fff30ca 4010476e 3fff2384 3fff35f8  
3fff0b10:  0000002e 3fff12a4 3fff2384 401046e9  
3fff0b20:  3fff3084 3fff0c13 003fe000 00000030  
3fff0b30:  80000000 3fffc6fc 00000001 3fff12d9  
3fff0b40:  3fff12a4 3fff12a4 3fff3084 40104855  
3fff0b50:  40213f73 3fff3084 3fff30ca 40213f7c  
3fff0b60:  00000008 3fff0b90 3ffe8e85 3fff0bd0  
3fff0b70:  402120a4 3fff0c70 3fff0c0d 3fff30d8  
3fff0b80:  3fff3084 3fff3330 3fff12a4 4021a011  
3fff0b90:  005e0001 0000fb00 00000010 3ffe9dee  
3fff0ba0:  00000018 3fff3330 3fff3084 4021aa98  
3fff0bb0:  00000080 4000422e 3ffefd30 00000000  
3fff0bc0:  3fff0c34 00000001 00000000 00000002  
3fff0bd0:  3fff11fc 00000004 3fff332c 4021b110  
3fff0be0:  00000006 0000000e 00302078 3fff332c  
3fff0bf0:  00000000 3fff3084 3fff30f0 4021aad8  
3fff0c00:  3fff12a4 3fff0c30 00000004 3fff332c  
3fff0c10:  00000016 3fff12a4 3fff332c 4021a269  
3fff0c20:  3fff12a4 3fff0c30 00000004 40213fd8  
3fff0c30:  00000494 0104a8c0 3fff1434 40214014  
3fff0c40:  3fff3330 3fff12a4 3ffefd30 4021a33a  
3fff0c50:  3fff352c 3ffefa28 0000001c 00000020  
3fff0c60:  3fff0ccc 3fff12a4 3fff332c 4021a513  
3fff0c70:  3fff0ccc 3ffe9b7c 3fff12a4 4021a59f  
3fff0c80:  3fff0d20 00000001 3fff0d00 00000001  
3fff0c90:  3ffefa6c 3ffefa24 3ffefa6c 4020913e  
3fff0ca0:  00000000 3ffefa24 3fff31ec 401009a4  
3fff0cb0:  3ffefd30 000003d1 3fff0cf0 4020cb64  
3fff0cc0:  00000000 3ffefa24 3ffefa98 fb0000e0  
3fff0cd0:  00000000 3ffefa24 3ffefa6c 3ffefe18  
3fff0ce0:  00000000 3ffefa24 3ffefa6c 4020a60a  
3fff0cf0:  3fff31f4 009d00a5 40208e30 40209260  
3fff0d00:  00000000 00000000 3ffefab0 4020cf6c  
3fff0d10:  00000000 3ffefa6c 3ffefab0 4020ad09  
3fff0d20:  3fff3174 2d363632 4020a65c 4020d398  
3fff0d30:  3ffefab0 00000000 3ffefd58 40202659  
3fff0d40:  00000000 00000000 3ffe8b38 00000000  
3fff0d50:  3fff1f34 0000000f 00000006 3fff1f54  
3fff0d60:  0000000f 0000000a 00000000 00000000  
3fff0d70:  00001c2f 3ffe95e8 00000000 3ffe95e8  
3fff0d80:  00000000 3ffe95e8 00000000 3ffe95e8  
3fff0d90:  00000000 3ffe95e8 00000000 3ffe95e8  
3fff0da0:  00000000 00000000 ffffffff fe000001  
3fff0db0:  3ffe8ab8 00000000 fe010035 00000000  
3fff0dc0:  00000000 feefeffe feefeffe feefeffe  
3fff0dd0:  feefeffe feefeffe feefeffe feefeffe  
3fff0de0:  feefeffe feefeffe feefeffe 3fff31f4  
3fff0df0:  6c00a8c0 402024e0 40202420 feefeffe  
3fff0e00:  feefeffe feefeffe feefeffe feefeffe  
3fff0e10:  feefeffe feefeffe feefeffe 3ffefe18  
3fff0e20:  3fffdad0 00000000 3ffefe10 4020cc6f  
3fff0e30:  feefeffe feefeffe 3ffefe20 40100a28  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

gibo77, 2018-02-15:
Update: I was able to run StackTrace result below.

Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 37 results
0x40227b83: cnx_node_search at ?? line ?
0x401022a4: wDev_ProcessFiq at ?? line ?
0x40221fb0: ieee80211_output_pbuf at ?? line ?
0x4010476e: glue2esp_linkoutput at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c line 292
0x401046e9: glue2esp_linkoutput at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c line 263
0x4023db0f: wifi_station_get_connect_status at ?? line ?
0x40104855: new_linkoutput at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c line 240
0x40213f73: ethernet_output at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c line 305
0x40213f7c: ethernet_output at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c line 305
0x402120a4: __ssputs_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/nano-vfprintf.c line 180
0x4021a011: etharp_output_LWIP2 at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/etharp.c line 893
0x4021aa98: ip4_output_if_opt_src at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c line 962
0x4021b110: mem_malloc at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/mem.c line 136
0x4021aad8: ip4_output_if_opt at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c line 788
0x4021a269: igmp_send at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c line 570
0x40213fd8: do_memp_malloc_pool at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/memp.c line 231
0x40214014: memp_malloc at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/memp.c line 231
0x4021a33a: igmp_lookup_group at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c line 570
0x4021a513: igmp_start_timer at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c line 570
:  (inlined by) igmp_joingroup_netif at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c line 521
0x4021a59f: igmp_joingroup at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c line 570
0x4020913e: MDNSResponder::_listen() at C:\Users\Gilbert\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266mDNS/ESP8266mDNS.cpp line 396
0x401009a4: free at C:\Users\Gilbert\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1737
0x4020cb64: operator delete(void*) at C:\Users\Gilbert\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/abi.cpp line 84
0x4020a60a: MDNSResponder::begin(char const*) at C:\Users\Gilbert\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266mDNS/ESP8266mDNS.cpp line 396
0x40208e30: _M_manager at C:\Users\Gilbert\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266mDNS/ESP8266mDNS.cpp line 396
0x40209260: operator() at C:\Users\Gilbert\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266mDNS/ESP8266mDNS.cpp line 396
:  (inlined by) _M_invoke at c:\users\gilbert\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2071
0x4020cf6c: std::_Function_base::~_Function_base() at c:\users\gilbert\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2031
0x4020ad09: ArduinoOTAClass::begin() at C:\Users\Gilbert\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.0\libraries\ArduinoOTA/ArduinoOTA.cpp line 64
0x4020a65c: std::_Function_base::_Base_manager    (ArduinoOTAClass*)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at C:\Users\Gilbert\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.0\libraries\ArduinoOTA/ArduinoOTA.cpp line 64
0x4020d398: std::_Function_base::_Base_manager    (ArduinoOTAClass*)> >::_M_get_pointer(std::_Any_data const&) at c:\users\gilbert\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 1893
:  (inlined by) std::_Function_handler    (ArduinoOTAClass*)> >::_M_invoke(std::_Any_data const&) at c:\users\gilbert\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2071
0x40202659: setup at D:\Documents\Arduino\Interrupt_Dimmer_EEPROM/Interrupt_Dimmer_EEPROM.ino line 92
0x402024e0: _M_manager at c:\users\gilbert\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 1931
0x40202420: _M_invoke at c:\users\gilbert\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2069
0x4020cc6f: loop_wrapper at C:\Users\Gilbert\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/core_esp8266_main.cpp line 57
0x40100a28: cont_norm at C:\Users\Gilbert\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/cont.S line 109

tablatronix, 2018-02-15:
You should have been presented with a template when submitting

tablatronix, 2018-02-15:
You provided no information on your enviroment

gibo77, 2018-02-16:
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
Decoding 37 results
0x40227b83: cnx_node_search at ?? line ?
0x401022a4: wDev_ProcessFiq at ?? line ?
0x40221fb0: ieee80211_output_pbuf at ?? line ?
0x4010476e: glue2esp_linkoutput at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c line 292
0x401046e9: glue2esp_linkoutput at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c line 263
0x4023db0f: wifi_station_get_connect_status at ?? line ?
0x40104855: new_linkoutput at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c line 240
0x40213f73: ethernet_output at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c line 305
0x40213f7c: ethernet_output at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c line 305
0x402120a4: __ssputs_r at /Users/igrokhotkov/e/newlib-xtensa/xtensa-lx106-elf/newlib/libc/stdio/../../../.././newlib/libc/stdio/nano-vfprintf.c line 180
0x4021a011: etharp_output_LWIP2 at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/etharp.c line 893
0x4021aa98: ip4_output_if_opt_src at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c line 962
0x4021b110: mem_malloc at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/mem.c line 136
0x4021aad8: ip4_output_if_opt at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c line 788
0x4021a269: igmp_send at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c line 570
0x40213fd8: do_memp_malloc_pool at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/memp.c line 231
0x40214014: memp_malloc at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/memp.c line 231
0x4021a33a: igmp_lookup_group at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c line 570
0x4021a513: igmp_start_timer at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c line 570
: (inlined by) igmp_joingroup_netif at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c line 521
0x4021a59f: igmp_joingroup at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c line 570
0x4020913e: MDNSResponder::_listen() at C:\Users\Gilbert\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266mDNS/ESP8266mDNS.cpp line 396
0x401009a4: free at C:\Users\Gilbert\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266\umm_malloc/umm_malloc.c line 1737
0x4020cb64: operator delete(void*) at C:\Users\Gilbert\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/abi.cpp line 84
0x4020a60a: MDNSResponder::begin(char const*) at C:\Users\Gilbert\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266mDNS/ESP8266mDNS.cpp line 396
0x40208e30: _M_manager at C:\Users\Gilbert\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266mDNS/ESP8266mDNS.cpp line 396
0x40209260: operator() at C:\Users\Gilbert\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.0\libraries\ESP8266mDNS/ESP8266mDNS.cpp line 396
: (inlined by) _M_invoke at c:\users\gilbert\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2071
0x4020cf6c: std::_Function_base::~_Function_base() at c:\users\gilbert\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2031
0x4020ad09: ArduinoOTAClass::begin() at C:\Users\Gilbert\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.0\libraries\ArduinoOTA/ArduinoOTA.cpp line 64
0x4020a65c: std::_Function_base::_Base_manager (ArduinoOTAClass*)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at C:\Users\Gilbert\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.0\libraries\ArduinoOTA/ArduinoOTA.cpp line 64
0x4020d398: std::_Function_base::_Base_manager (ArduinoOTAClass*)> >::_M_get_pointer(std::_Any_data const&) at c:\users\gilbert\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 1893
: (inlined by) std::_Function_handler (ArduinoOTAClass*)> >::_M_invoke(std::_Any_data const&) at c:\users\gilbert\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2071
0x40202659: setup at D:\Documents\Arduino\Interrupt_Dimmer_EEPROM/Interrupt_Dimmer_EEPROM.ino line 92
0x402024e0: _M_manager at c:\users\gilbert\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 1931
0x40202420: _M_invoke at c:\users\gilbert\documents\arduinodata\packages\esp8266\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2/functional line 2069
0x4020cc6f: loop_wrapper at C:\Users\Gilbert\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.0\cores\esp8266/core_esp8266_main.cpp line 57
[... cut ...]
I have not done any stack trace. Maybe the gurus can do that below.
Below is the stack dump. Thanks!
*WM: Sent config page
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result:
*WM: 3
connected...yeey :)
IP address: 192.168.0.108  ------>The ESP8266 was able to connect, but after executing ArduinoOTA it   crashes.

Exception (28):
epc1=0x40227b83 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000098 depc=0x00000000
ctx: cont
sp: 3fff0920 end: 3fff0e40 offset: 01a0

tablatronix, 2018-02-16:
I had a similar error and commited a fix earlier today, maybe try again.
Also maybe try with esp staging (it can be a pain in arduino ide)

tablatronix, 2018-02-16:
Considering you are running 2 webservers, I am sure that has something to do with it

gibo77, 2018-02-16:
As per the serial monitoring, the ESP8266 was able to connect to the Wifi because of the IP Address being sent to serial monitor. The crash happens when executing `ArduinoOTA.begins()`.
From what I know, ArduinoOTA reserves a flash memory space that has the same size bytes of the sketch all the time. So that when an OTA is called, the new sketch will be dumped there.

It seems the WifiManager is using these same flash memory space as the ArduinoOTA. I could be wrong. But I have to say what I think is causing the crash. I have not fully studied the memory usage of WifiManager. It looks like it uses a lot of flash space due to the HTML program. Is it correct to say that WifiManager uses SPIFFS?

tablatronix, 2018-02-16:
Does ota use interrupts?

gibo77, 2018-02-20:
I am not sure of OTA does interrupts. The interrupts you see in my program is specific to the application that I am controlling with ESP8266.
So WifiManager ONLY uses PROGMEM and that is it?

Thanks for your reply!

tablatronix, 2018-02-20:
Well am working on abstracting all strings to progmem, so this might make it easier to grab from spiffs in the future by using another abstraction layer eventually.

gibo77, 2018-02-20:
I have not tried PROGMEM. But is it correct to say that since they are an extension of RAM, if you turn off the unit, the PROGMEM data will be erased as well? We all know that SPIFFS and EEPROM retain data after when removing power.

tablatronix, 2018-02-20:
https://bbs.espressif.com/viewtopic.php?t=1183

UroojAlam, 2020-06-16:
Change the webserver port to 81 or 8080 then it won't throw an exception. This worked for me without using `wifiManager.resetSetting();` , and now I can simultaneously use wifi manager, OTA, and HTTP WebUpdater in my sketch.

tablatronix, 2020-06-17:
This is old and outdated the wm webserver is now shared and can be used