tzapu/WiFiManager #524: Create quick instructions for installing development branch to arduino

tablatronix, 2018-02-16:


per1234, 2018-02-16:
Here's the most simple method:
1. Download https://github.com/tzapu/WiFiManager/archive/development.zip
1. (In the Arduino IDE) Sketch > Include Library > Add .ZIP Library... > select the downloaded file > Open

The problems with that method are:
- If they have previously installed the library via Library Manager, the master branch via "Add .ZIP Library", or manual installation this will cause two copies of the library to exist. The one with the folder name WiFiManager will have include priority. If neither folder matches the include filename then the priorities are uncertain but on my installation WiFiManager-master has priority over WiFiManager-development.
- If they have previously followed those directions and try to repeat them to update to the latest development version the installation will fail: "A library named WiFiManager-development already exists.

So manual installation may be a better choice. It's less beginner friendly but a beginner probably isn't going to be using the development version anyway:
1. Download https://github.com/tzapu/WiFiManager/archive/development.zip
1. Delete any previous installations of the WiFiManager library from {sketchbook folder}/libraries. You can find/set the location of your sketchbook folder in the Arduino IDE at **File > Preferences > Sketchbook location**.
1. Copy the folder WiFiManager-development from the downloaded file to {sketchbook folder}/libraries.
1. Restart the Arduino IDE if it's running.

Now the other option is using git to clone the repository but this is going to be a bit too involved/complex for even some reasonably experienced Arduino users. I know git well but still don't clone only for the sake of installing a library. For someone participating in development it's appropriate, though quite a few GitHub users only use the web interface to commit and don't even have Git installed on their computer. It's reasonable to add cloning instructions but I think it should be in addition to the manual installation instructions. Certainly the current [instructions ](https://github.com/tzapu/WiFiManager/tree/development#checkout-from-github)need to be improved.

tablatronix, 2018-07-29:
Esp32
https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/boards_manager.md

tablatronix, 2020-05-18:
Has this changed at all since then ?
Also I just pushed an alpha, waiting to see if arduino picks it up or not

per1234, 2020-05-19:
>Has this changed at all since then ?

No. Well, one thing has changed, but it's only related to something I mentioned in passing above:
>If neither folder matches the include filename then the priorities are uncertain

The library dependency resolution system is now very well documented:
https://arduino.github.io/arduino-cli/sketch-build-process/#dependency-resolution

Not that it helps with the project of providing development branch installation instructions.

>waiting to see if arduino picks it up or not

The tag was picked up by the indexer.

tablatronix, 2020-05-19:
I noticed that, but we only have dependencies in the examples

per1234, 2020-05-19:
WiFiManager is a dependency of any sketch that uses it, so dependency resolution is indeed relevant for anyone who has multiple copies of WiFiManager installed (as would be the case if they installed the release version via Library Manager and then installed the development version via "Add .ZIP Library", using the .zip file download from GitHub).

tablatronix, 2020-05-19:
ah I see, makes sense

tablatronix, 2020-05-19:
woah , location priority is rather interesting..

https://arduino.github.io/arduino-cli/sketch-build-process/#location-priority