tzapu/WiFiManager #525: howto call config page at runtime

ageurtse, 2018-02-16:
i'm using the development brance and need to connect to the config page at runtime. 
My problem is that at runtime i already run a webserver. 
So calling wifiManager.startWebPortal(); isn't helping.
after calling this one all is hanging. 

where i need it for is in case of needed, change the ssid and password and change static ip adres.

SomeGuyNamedJay, 2018-02-19:
I'd love to know the answer to this too. In the meantime, this is what I do:
Add a  "webpage" to call the resetSettings():

In setup:
  httpServer.on("/resetWifi", []() {
        resetWifi();
        ESP.restart();
  })

Function:
void resetWifi() {
    WiFiManager wifiManager;
    wifiManager.resetSettings();
}

For some reason I get an error if I put wifiManager.resetSettings() in the httpServer call.

ageurtse, 2018-02-20:
for me this isn't working. 
my board isn't resetting the settings. 
see log from serial monitor
as you can see it keeps resetting. 

after rebooting the device it connects back to my wifi network. 

`WiFi Started.
local ip
192.168.1.207
httpUpdater started.
webserver started.
Test Sketch v1.0 started
*WM: freeing allocated params!
*WM: unloading
Resetting wifi settings
*WM: getCoreVersion():          2_4_0
*WM: system_get_sdk_version():  2.1.0(deb1901)
*WM: system_get_boot_version(): 31
*WM: getFreeHeap():             35712
*WM: SETTINGS ERASED
*WM: enableSTA PERSISTENT ON
Done.
Reset device.

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

*WM: getCoreVersion():          2_4_0
*WM: system_get_sdk_version():  2.1.0(deb1901)
*WM: system_get_boot_version(): 31
*WM: getFreeHeap():             39416
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: wifi station disconnect
*WM: Connecting to saved AP
*WM: enableSTA PERSISTENT ON
*WM: connectTimeout not set, ESP waitForConnectResult...
*WM: Connection result: WL_CONNECTED
*WM: IP Address: 192.168.1.207
WiFi Started.
local ip
192.168.1.207
httpUpdater started.
webserver started.
Test Sketch v1.0 started
*WM: freeing allocated params!
*WM: unloading
Resetting wifi settings
*WM: getCoreVersion():          2_4_0
*WM: system_get_sdk_version():  2.1.0(deb1901)
*WM: system_get_boot_version(): 31
*WM: getFreeHeap():             37536
*WM: SETTINGS ERASED
*WM: enableSTA PERSISTENT ON
Done.
Reset device.

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

*WM: getCoreVersion():          2_4_0
*WM: system_get_sdk_version():  2.1.0(deb1901)
*WM: system_get_boot_version(): 31
*WM: getFreeHeap():             39416
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: wifi station disconnect
*WM: Connecting to saved AP
*WM: enableSTA PERSISTENT ON
*WM: connectTimeout not set, ESP waitForConnectResult...
*WM: Connection result: WL_CONNECTED
*WM: IP Address: 192.168.1.207
WiFi Started.
local ip
192.168.1.207
httpUpdater started.
webserver started.
Test Sketch v1.0 started
*WM: freeing allocated params!
*WM: unloading
Resetting wifi settings
*WM: getCoreVersion():          2_4_0
*WM: system_get_sdk_version():  2.1.0(deb1901)
*WM: system_get_boot_version(): 31
*WM: getFreeHeap():             37304
*WM: SETTINGS ERASED
*WM: enableSTA PERSISTENT ON
Done.
Reset device.

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld

*WM: getCoreVersion():          2_4_0
*WM: system_get_sdk_version():  2.1.0(deb1901)
*WM: system_get_boot_version(): 31
*WM: getFreeHeap():             39416
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: wifi station disconnect
*WM: Connecting to saved AP
*WM: enableSTA PERSISTENT ON
*WM: connectTimeout not set, ESP waitForConnectResult...
*WM: Connection result: WL_CONNECTED
*WM: IP Address: 192.168.1.207
WiFi Started.
local ip
192.168.1.207
httpUpdater started.
webserver started.
Test Sketch v1.0 started
*WM: freeing allocated params!
*WM: unloading
Resetting wifi settings
*WM: getCoreVersion():          2_4_0
*WM: system_get_sdk_version():  2.1.0(deb1901)
*WM: system_get_boot_version(): 31
*WM: getFreeHeap():             36912
*WM: SETTINGS ERASED
*WM: enableSTA PERSISTENT ON
Done.
Reset device.
`

tablatronix, 2018-02-20:
what version of esp ?

tablatronix, 2018-02-20:
oh nm, I see it

tablatronix, 2018-02-20:
sorry , let me check, you are obviously using dev branch

tablatronix, 2018-02-21:
have you tried running one webserver on another port , like 81 ?

ageurtse, 2018-02-23:
not working for me. 
below how i reset my settings, but after restarting it stil connect;s to my wifi. 
I wan't somehow to reset all settings but it doesn't work. 

> void resetwifi (){
> WiFiManager wifiManager;
> wifiManager.resetSettings();
> ESP.reset();
> }

and below this line the whole sketch. 

/*
  To upload through terminal you can use: curl -F "image=@firmware.bin" esp8266-webupdate.local/update
*/
#define FIXERASECONFIG

#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <ESP8266HTTPUpdateServer.h>
#include <WiFiManager.h> 

ESP8266WebServer server(80);
ESP8266HTTPUpdateServer httpUpdater;

const int led = 13;
byte value;

void configModeCallback (WiFiManager *myWiFiManager) {
  Serial.println("Entered config mode");
  Serial.println(WiFi.softAPIP());
  //if you used auto generated SSID, print it
  Serial.println(myWiFiManager->getConfigPortalSSID());
}

void handleRoot() {
  digitalWrite ( led, 1 );
  char temp[400];
  int sec = millis() / 1000;
  int min = sec / 60;
  int hr = min / 60;

  snprintf ( temp, 400,

"<html>\
  <head>\
    <meta http-equiv='refresh' content='5'/>\
    <title>ESP8266 Demo</title>\
    <style>\
      body { background-color: #cccccc; font-family: Arial, Helvetica, Sans-Serif; Color: #000088; }\
    </style>\
  </head>\
  <body>\
    <h1>Hello from ESP8266!</h1>\
    <p>Uptime: %02d:%02d:%02d</p>\
    [image]\
  </body>\
</html>",

    hr, min % 60, sec % 60
  );
  server.send ( 200, "text/html", temp );
  digitalWrite ( led, 0 );
}

void handleNotFound() {
  digitalWrite ( led, 1 );
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += ( server.method() == HTTP_GET ) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";

  for ( uint8_t i = 0; i < server.args(); i++ ) {
    message += " " + server.argName ( i ) + ": " + server.arg ( i ) + "\n";
  }

  server.send ( 404, "text/plain", message );
  digitalWrite ( led, 0 );
}

void resetwifi (){
WiFiManager wifiManager;
wifiManager.resetSettings();
ESP.reset();
}

void setup(void){
WiFiManager wifiManager;

  Serial.begin(115200);
  Serial.println();
  Serial.println("ESP starting .....");
  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around

  wifiManager.setShowStaticFields(true);
  wifiManager.setDebugOutput(true);

  //set callback that gets called when connecting to previous WiFi fails, and enters Access Point mode
  wifiManager.setAPCallback(configModeCallback);

  //fetches ssid and pass and tries to connect
  //if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP"
  //and goes into a blocking loop awaiting configuration
  if(!wifiManager.autoConnect()) {
    Serial.println("failed to connect and hit timeout");
    //reset and try again, or maybe put it to deep sleep
    ESP.restart();
    delay(1000);
  } 

//if you get here you have connected to the WiFi
  Serial.println("WiFi Started.");
  Serial.println("local ip");
  Serial.println(WiFi.localIP());
  
  httpUpdater.setup(&server);
  Serial.println("httpUpdater started.");

  server.on ( "/", handleRoot );
  server.on ( "/test.svg", drawGraph );
  server.on ( "/reset", resetwifi );
  server.on ( "/inline", []() {
  server.send ( 200, "text/plain", "this works as well" );
  } );
  server.onNotFound ( handleNotFound );
  server.begin();
  Serial.println("webserver started.");
  Serial.println("Test Sketch v1.0 started");
}

void loop(void){
  server.handleClient();
}

void drawGraph() {
  String out = "";
  char temp[100];
  out += "<svg xmlns=\"http://www.w3.org/2000/svg\" version=\"1.1\" width=\"400\" height=\"150\">\n";
  out += "<rect width=\"400\" height=\"150\" fill=\"rgb(250, 230, 210)\" stroke-width=\"1\" stroke=\"rgb(0, 0, 0)\" />\n";
  out += "<g stroke=\"black\">\n";
  int y = rand() % 130;
  for (int x = 10; x < 390; x+= 10) {
    int y2 = rand() % 130;
    sprintf(temp, "<line x1=\"%d\" y1=\"%d\" x2=\"%d\" y2=\"%d\" stroke-width=\"1\" />\n", x, 140 - y, x + 10, 140 - y2);
    out += temp;
    y = y2;
  };
  out += "</g>\n</svg>\n";

  server.send ( 200, "image/svg+xml", out);
}

tablatronix, 2018-02-23:
use proper code formatting

ageurtse, 2018-02-23:
don't understand what is wrong with the above code formating. 
i'm a hobby programmer, that could be the problem, sorry for that.

tablatronix, 2018-02-23:
What happens if you minimize this to a basic sketch that sets then erases config ?

ageurtse, 2018-02-23:
The example sketch AutoConnectWithReset is running fine. 
Next try to alter things, but i can't get it working. 

see code i use for testing.
Is wifiManager.resetSettings(); only working in the setup routine ?

```
#include <FS.h>          // this needs to be first, or it all crashes and burns...
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>

#define FIXERASECONFIG
ESP8266WebServer server(80);

void handleRoot() {
  server.send(200, "text/plain", "hello from esp8266!");
}

void handleReset() {
  server.send(200, "text/plain", "Resetting!");
    //Serial.println("diconnecting client and wifi");
    //client.disconnect();
    wifi_station_disconnect();

  WiFiManager wifiManager;
  wifiManager.resetSettings();
  ESP.reset();
  
}

void handleNotFound(){
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i=0; i<server.args(); i++){
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println();

  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;

  //exit after config instead of connecting
  wifiManager.setBreakAfterConfig(true);

  //reset settings - for testing
  //wifiManager.resetSettings();

  //tries to connect to last known settings
  //if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP" with password "password"
  //and goes into a blocking loop awaiting configuration
  if (!wifiManager.autoConnect("AutoConnectAP", "password")) {
    Serial.println("failed to connect, we should reset as see if it connects");
    delay(3000);
    ESP.restart();
    delay(5000);
  }

  //if you get here you have connected to the WiFi
  Serial.println("");
  Serial.print("Connected to ");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp8266")) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);

  server.on("/inline", [](){
    server.send(200, "text/plain", "this works as well");
  });
  server.on("/resetwifi", handleReset);
  server.onNotFound(handleNotFound);

  server.begin();
  Serial.println("HTTP server started");

}

void loop() {
  // put your main code here, to run repeatedly:
  server.handleClient();

}
```

tablatronix, 2018-02-23:
Thanks, fixed
Problem was persistent was off

ageurtse, 2018-02-23:
And where do i turn persistent on ?

tablatronix, 2018-02-23:
I fixed it in development

closed 2022-08-16 by ageurtse