tzapu/WiFiManager #527: Request: Configure hostname

gepd, 2018-02-18:
From #513 
Current workaround

```cpp
void WiFiEvent(WiFiEvent_t event){
    switch(event) {
        case SYSTEM_EVENT_AP_START:
            WiFi.softAPsetHostname(HOSTNAME);
            break;
        case SYSTEM_EVENT_STA_START:
            WiFi.setHostname(HOSTNAME);
            break;
        default:
            break;
    }
}

void setup(){
    WiFi.disconnect(true);
    WiFi.onEvent(WiFiEvent);
    WiFi.begin();

// checks connection and perform a call to the config portal if it's neccessary
}
```

tablatronix, 2018-02-18:
I don't really understand how hostname on esp works, is this mdns hostnames?
Does it use local.* ?

tablatronix, 2018-02-18:
Could you set softaphostname in the _apcallback and sta after autoconnect success ?

tablatronix, 2018-02-18:
so these seem to be dhcp client ids maybe ?

I still cannot figure out when to set these, some says before , some says after ap and sta are enabled.

tablatronix, 2018-02-18:
I am setting before and after for sta, well see which ones work for which chip, and adjust as needed.
Obviously there we can only set after if already connected etc.

Any tips on how to test this ?
my router shows my esp32 dhcp client name correctly, so that works.

- [ ] test esp32 ap hostname
- [ ] test esp32 sta hostname
- [ ] test esp8266 sta hostname

gepd, 2018-02-18:
Yes, it's dhcp client ids

I've tried:
```cpp
WiFi.begin();
WiFi.setHostname("name");
```

But it makes the device crash, but if you reverse it, it works
```cpp
WiFi.setHostname("name");
WiFi.begin();
```

I took that piece of code from this example, it sets AP and STA:
https://github.com/copercini/esp32-iot-examples/blob/master/WiFi_nonBlocking/WiFi_nonBlocking.ino

> Could you set softaphostname in the _apcallback and sta after autoconnect success also?
Does this need to be in core?

I'm not sure. Normally it need to be done before `WiFi.begin` But the flag `SYSTEM_EVENT_AP_START` make me doubt where it's performed

tablatronix, 2018-02-18:
I guess we could just use events also, but thats one more thing that could introduce bugs if its not needed. Well see

tablatronix, 2018-02-19:
ok I think i have a decent working solution now

tablatronix, 2018-02-19:
I am just setting hostname, and reconnecting for now if already connected, else it works fine normally.
Alternative is to restart the dhcp server, but I do not think that is exposed at all.

gepd, 2018-02-19:
I've just check it and there is not problem in STA (both names are the same) but not in AP

tablatronix, 2018-02-19:
Esp8266 has no apname config
I have not tested esp32 aphostnames yet.

It is nor working at all?

gepd, 2018-02-19:
When you are in STA
STA Hostname: `newname`
AP Hostname: `newname`

In AP:
AP hostname: `newname`
STA hostname: `espressif`

is not a big deal, but if it's fixed would be good

tablatronix, 2018-02-19:
which board?

tablatronix, 2018-02-19:
how do you test ap hostname ? I dont seem to see it coming across in wirreshark, says "server hostname not given"

tablatronix, 2018-02-19:
Might want to give it a try again.

On esp32 ap, info shows both `newname`, I have to confirm via network of course.

gepd, 2018-02-19:
I'm working with a ESP32

Now both cases are working.
I checked that from my router, and also from the `Info` page of WM

First I start the portal and check the Info page (No WiFi connection) then I connect the device to a network and check it again.

Now it's working as spected.
I don't have an esp8266 here to test it

tablatronix, 2018-02-19:
I still cannot figure out what the aphostname actually does. Shrug,

gepd, 2018-02-19:
I don't know either, for me it has not sense to change it, but as it's displayed in the info page, is a little weird to read two hostnames with different names

bitboy85, 2019-06-22:
I would also like to have this feature. Makes it more easy to open webserver by name instead of ip address. As far as i know hostname is only in station mode supported and if i remember correctly there were some issues with hostname and the esp8266 arduino library.
Can you please check with an updated one?

tablatronix, 2019-06-22:
This feature already exists.. Did you read this?

bitboy85, 2019-06-22:
What i had in mind is if using autoconfiguration that you can set a hostname in the "configureWifi" menu. Which is also saved in flash and restored during power up.
Currently its only working if hardcoded or did i miss something?

tablatronix, 2019-06-22:
yes you are correct, anything that uses flash or spiffs to store stuff has to be written by users on their own using params, wifimanager does not write to flash memory on its own by design

markg85, 2019-07-08:
I too ended up here after having hacked this very feature in my local WiFiManager along with mDSN support.

Just to figure out that the feature is there all along, just in the development branch... Which you obviously don't get when using PlatformIO..

Could you consider making a new release, please? These two features alone would already be welcome enough for a release.

tablatronix, 2019-07-08:
I will accept PR for examples on development if anyone want to write them though

markg85, 2019-07-08:
> I will accept PR for examples on development if anyone want to write them though.
> 
> I hope to release it (beta) soon, sorry about that you can use it on platformio, you can install anything on platformio using git urls

Yes, that will work. But then i lose the update feature... This is only a good solution as long as you haven't made a release. As soon as you do, it requires me (and all who used git clone) to modify it again.

tablatronix, 2019-07-09:
Can you not override libraries in platformio.ini? I have not tried yet

tablatronix, 2019-07-09:
Yeah you just add lib deps and use the custom branch
https://docs.platformio.org/en/latest/librarymanager/quickstart.html#project-dependencies

commit 9f415643 (2018-02-19): fixes #527
commit 97398d7c (2018-02-19): fix #527 for esp8266
closed 2021-01-04 by tablatronix