tzapu/WiFiManager #528: Connection result feedback in STA

gepd, 2018-02-18:
From #513 

Requesting an option to display the reason why the connection wasn't able to be performed in the last try (STA mode). You will no always have access to read the debug output.

What could be the best way to implement this?
EEPROM, SPIFFS?

@tablatronix 
> Do you need to store it, or can it be a lifetime variable with a getter?

Can be a lifetime variable
I was thinking in storing it because how will you know the connection result after a possible reset?

tablatronix, 2018-02-18:
True but I dont have the library doing resets anymore. That typically is not needed after 2.1.0 i think. Unless you mean uexpected resets?

gepd, 2018-02-18:
Yes, I mean assuming that it happens. 
I don't know if it will ever happen or if storing it will be really useful. Today I just needed to know the reason why it wasn't connecting

tablatronix, 2018-02-18:
ok I added this easiest as possible, so let me know if its not specific enough.

I am saving all connection results, not just failures, it was easier
I am saving conx result for autoconnect AND wifisave
no wifi saved uses WL_NO_SSID_AVAIL as a flag, maybe disconnected would be better..

If this is not satisfactory, I can change to getLastConxFailure, and add if(!WL_CONNECTED) conditions and only log failures, I wasn't sure if that was needed.

gepd, 2018-02-18:
I was doing something similar, and I can say now there is no need to store the result value.

I'll check your implementation, in my tests I was getting the value from `waitForConnectResult` and then I shown it when I was in AP mode again.

I guess, know if the problem was the ssid or password will be enough in most of the cases.

tablatronix, 2018-02-18:
also note
afaik STATION_WRONG_PASSWORD is not implemented in SDK last time I checked.
that is why
`WL_CONNECT_FAILED` is used for both ` STATION_CONNECT_FAIL` and `STATION_WRONG_PASSWORD`

gepd, 2018-02-19:
What do you think to add this value in the

tablatronix, 2018-02-19:
Not sure I understand

gepd, 2018-02-19:
I mean, at this moment you have a message info showing the state of the network. 

When it's not connected ([HTTP_STATUS_OFF](https://github.com/tzapu/WiFiManager/blob/development/strings_en.h#L46)), it only shows `**Not Connected** to networkname`.
It would be more useful to display
```
Not Connected to networkname
SSID not valid / Wrong Password
```

But only if `WL_NO_SSID_AVAIL` or `WL_CONNECT_FAILED`

tablatronix, 2018-02-19:
hmm, good idea

tablatronix, 2018-02-19:
It looks like esp32 doesnt give much info on failure, just disconnected, ill keep looking but i didnt  even see any useful events in the event system

gepd, 2018-02-20:
I've tested your last changes and `WL_NO_SSID_AVAIL` works as expected, with a wrong password I only get `WL_CONNECT_FAILED`

Is the `WL_STATION_WRONG_PASSWORD` flag recently introduced by espressif?

tablatronix, 2018-02-20:
In esp8266 it has always been there but not working. There is no equiv in esp32, there is an auth fail, but i only see it triggered when ap doesnt exist not on password falure, in fact i see no events for wrong password, just the default which is disconnected.

tablatronix, 2018-02-20:
> In step 4.2, the Wi-Fi connection may fail because, for example, the password is wrong, the AP is not found, etc. In a case like this, <SYSTEM_EVENT_STA_DISCONNECTED> will arise and the reason for such a failure will be provided. For handling events that disrupt Wi-Fi connection, please refer to phase 6.

gepd, 2018-02-20:
I got it, the event needs an extra argument to get the reason:

```cpp
#include <WiFi.h>

void WiFiEventHandler(WiFiEvent_t event, system_event_info_t info)
{
  Serial.printf("Got Event: %d\n", event);

  switch(event) {
    case SYSTEM_EVENT_STA_DISCONNECTED:
        Serial.printf("Reason: %d\n", info.disconnected.reason);
        break;
    }
}

void setup()
{
  Serial.begin(115200);
  
  WiFi.begin();
  WiFi.onEvent(WiFiEventHandler);
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    delay(500);
  }
  Serial.println();
}

void loop()
{
}
```

tablatronix, 2018-02-20:
I just commited that

tablatronix, 2018-02-20:
oddly sometimes I get no events at all, which seems odd, even it the router was not compatible or adhoc, I would think it would fire something.
```
*WM: Connecting to saved AP
*WM: connectTimeout not set, ESP waitForConnectResult...
*WM: Connection result: WL_DISCONNECTED
*WM: lastconxresult: WL_DISCONNECTED
```

tablatronix, 2018-02-20:
ok I just hacked it in using  ` static uint8_t _lastconxresulttmp;` and copying it into _lastconxresulttmp as needed. Should work for now.

tablatronix, 2018-02-20:
sometimes I see AUTH_FAIL events, but connection result comes back as WL_NO_SSID_AVAIL
So i am not entirely sure how reliable this will be, I saw some similar posts from others about these not being reliable always.

```
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 5 - STA_DISCONNECTED
[D][WiFiGeneric.cpp:298] _eventCallback(): Reason: 201 - AUTH_FAIL
*WM: Event: SYSTEM_EVENT_STA_DISCONNECTED, reconnecting
*WM: .
*WM: .
*WM: .
*WM: Connection result: WL_NO_SSID_AVAIL
*WM: lastconxresult: WL_NO_SSID_AVAIL
```

tablatronix, 2018-02-20:
sometimes I get this instead

```
          // [D][WiFiGeneric.cpp:298] _eventCallback(): Reason: 2 - AUTH_EXPIRE
          // *WM: Connection result: WL_DISCONNECTED          
```

not sure if this counts as wrong_password or not, probably not

gepd, 2018-02-20:
I think it does. When I set a wrong pass I always get:

```
*WM: connectTimeout not set, ESP waitForConnectResult...
*WM: EVENT: WIFI_REASON: AUTH_EXPIRE
*WM: EVENT: WIFI_REASON: AUTH_FAIL
*WM: Connection result: WL_CONNECT_FAILED
```

tablatronix, 2018-02-20:
But this is WL_DISCONNECTED  not WL_CONNECT_FAILED

tablatronix, 2018-02-20:
oh ok i see now

```
        } else if(reason == WIFI_REASON_AUTH_EXPIRE) {
            if(WiFi.getAutoReconnect()){
                WiFi.begin();
            }
```

This is changing the event status to disconnect, because it is auto reconnecting and losing the event somehow

tablatronix, 2018-02-20:
I am trying to propose a fix for this, but I cannot figure out why this only runs 3 times, and not in a loop.
if WIFI_REASON_AUTH_EXPIRE status never gets updated, and there is no other event to stop the loop, so why does it only run 3 times ?

```
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:298] _eventCallback(): Reason: 2 - AUTH_EXPIRE
WM: EVENT: WIFI_REASON: 2
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:298] _eventCallback(): Reason: 2 - AUTH_EXPIRE
WM: EVENT: WIFI_REASON: 2
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:298] _eventCallback(): Reason: 2 - AUTH_EXPIRE
WM: EVENT: WIFI_REASON: 2
*WM: Connection result: WL_DISCONNECTED
```

tablatronix, 2018-02-20:
ohhh duh, i never actually looked at this code

waitforconnectresult is the only thing preventing infinite loops in the event handler.. ugh
 while((!status() || status() >= WL_DISCONNECTED) && i++ < 500) {

tablatronix, 2018-02-20:
Yeah so this actually runs forever, unless you change the mode which we do. hmm

tablatronix, 2018-02-20:
well anyways, auto_expire could be auth took too long and other reasons, so it is probably not a good idea to tell users wrong password unless we know it for certain

tablatronix, 2018-02-20:
https://github.com/espressif/arduino-esp32/issues/234

tablatronix, 2018-02-20:
I suppose we could change wrong passwor dot Auth Failure to cover more bases

gepd, 2018-02-20:
Yes, It may be better.
I tried the latest changes and all seems to be working well

closed 2018-03-06 by tablatronix