tzapu/WiFiManager #531: clean up examples

tablatronix, 2018-02-19:
Too many examples, simplify

- AutoConnect
- AutoConnectWithFeedback
- AutoConnectWithFeedbackLED
- AutoConnectWithFSParameters
- AutoConnectWithFSParametersAndCustomIP
- AutoConnectWithReset
- AutoConnectWithStaticIP
- AutoConnectWithTimeout
- OnDemandConfigPortal
- OnDemandWebPortal

tablatronix, 2018-02-19:
- AutoConnect
- FSParameters
- OnDemandConfigPortal
- OnDemandWebPortal
- kitchenSink

2delarosa, 2018-03-20:
I'm looking for an example of setting the ESP hostname when in station mode.

tablatronix, 2018-03-20:
#570

tablatronix, 2018-06-29:
- [ ] Move all autoconnect into subfolder
- [ ] Move basic autoconnect to simple example ( minimal )
- [ ] reduce variations, remove all custom code from examples
- [ ] add custom example containing kitchen sink of custom options for configportal
  * custom examples
   * set hostname
- [ ] add non blocking example for configportal and webportal, maybe make it smart, call configportal if not connected, and webportal if paired with ap, use selection logic in the example so reduce number of files.

joko2020, 2021-07-22:
> setBreakAfterConfig

Hi @tablatronix  I need help to use `setBreakAfterConfig` , 

What i want to do is  disable config portal or try to connect to wifi in background mode, so the program can continue to void loop. can this function be used?
thank you

tablatronix, 2021-07-24:
Yes that is non blocking mode see the examples

joko2020, 2021-07-24:
Ok thanks