tzapu/WiFiManager #532: Esp module not reconnecting by itself automatically to wifi with WiFiManager

jplavoiemtl, 2018-02-19:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Development

Hardware:			ESP-12e

Core Version:      	

### Description

Strange, the esp does not reconnect automatically to WiFi.  When my esp is running it connect to WiFi no problem.  I have a WiFi status of 3 which is connected.  Serial.println(WiFi.status());

Then I deactivate WiFi in my router to simulate a WiFi problem.  After I reactivate WiFi on the router, the esp does not reconnect to WiFi.  I check status at 1 second intervals and I get 6 for a WL_DISCONNECTED.

I don't understand why the module does not reconnect automatically like is should.  Any help please, thanks.

### Settings in IDE

Module:  			NodeMcu,

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

tablatronix, 2018-02-19:
try

`WiFi.setAutoReconnect(true);`

jplavoiemtl, 2018-02-19:
Thanks, I just tried and added in my setup loop: 
WiFi.setAutoReconnect(true);
Still no re-connection to WiFi and same WL_DISCONNECTED 6 value looping around in my serial monitor window...

tablatronix, 2018-02-19:
But it connects on reboot?

tablatronix, 2018-02-19:
Did you try development branch yet?

jplavoiemtl, 2018-02-19:
Yes, it connects fine on reboot.  Only not reconnecting after I cut WiFi on the router.
I tried the development branch today and I also tried the Master branch that I downloaded today.  Same results, no automatic WiFi re-connection...  :-(

tablatronix, 2018-02-19:
What version of esp core are you using might be a bug

jplavoiemtl, 2018-02-19:
Good point.  In my Arduinio IDE librairies I see ESP8266 Built-in version 1.0.0 installed.  I suppose I should get the new one at https://github.com/esp8266/Arduino and install the latest right?  I'll do that tonight and install the latest 2.2.0.  I will test and report after.  We may be on a good track here, thanks.

tablatronix, 2018-02-19:
2.4.0 your not even running the latest sdk on that version.

jplavoiemtl, 2018-02-20:
Might be a a basic question but how do you install the lastest esp core in Arduino IDE?  I've downloaded the Arduino-master.zip file from https://github.com/esp8266/Arduino.  In IDE I go to Add a .zip library and select that file.  But I get an error message that the file does not contain a valide library...

jplavoiemtl, 2018-02-20:
A new test running with latest 2.4.0 esp8266 core.  Same results, no WiFi re-connection after interruption.  

I'm confused with this problem.  This is a very basic setup: WiFi manager connects initially after power-up.  The unit runs fine.  Then with a simple WiFi interruption there is no re-connection?

tablatronix, 2018-02-20:
No idea that functionality is controlled by esp code, maybe remove wifimanager and just see what happens with a begin in setup then unplug and turn on esp debugging

jplavoiemtl, 2018-02-20:
I found the problem, it is a bug in esp core 2.4.0!  I did revert back to esp core 2.3.0 and the WiFi reconnects fine and WiFiManager works like a charm.  I built this module last year with core 2.3.0 and the WiFi got screwed with the latest 2.4.0 core.  So I guess you could close my issue here.  

Do you recommend me opening a bug report issue on the esp8266 core github or someone will eventually correct it?  It's a really important issue basic WiFi re-connection and I wonder how come it didn't surface in the reports on the new 2.4.0 core.

tablatronix, 2018-02-20:
Yes, see if one exists already though

jplavoiemtl, 2018-02-20:
Thanks for your help in resolving the issue to this problem @tablatronix

tablatronix, 2018-02-21:
be sure to link to any bugs you find so we know

jplavoiemtl, 2018-02-21:
There is an issue [#4397](https://github.com/esp8266/Arduino/issues/4397) I opened on the esp8266 core group.

closed 2018-02-21 by tablatronix