tzapu/WiFiManager #534: getValue() why not setValue()

zafrirron, 2018-02-20:
Probably my misunderstanding...:

Currently from Docs and Examples, handling custom configs done by save config hadler definition and managing FS (json) writes by app developer responsibility.

Since I guess most cases when one uses addParameter method he intend to save it in config, so I would expect this to happen inside the manager by the manager (same like any other wifi params).
In case the app needs to update the config param, I would expect a setValue() method, that would update (and write to FS) the config param.
 
A derived case is the ability to create, set, update a config params by the app without displaying it in manager config page, so a flag should be available for this.....

tablatronix, 2018-02-20:
yeah i was wondering why no setter, but noticed no ones asked for one.
sounds good

tablatronix, 2018-03-06:
This is added to development branch, have you been able to test?

closed 2018-03-09 by tablatronix

radeonorama, 2018-09-30:
Sorry to bring this one back up but is there an example usage for setValue()? 

I want the config page to show the newly saved params and not the default ones and this looks like it might be exactly what i need

zen85, 2020-05-14:
> Sorry to bring this one back up but is there an example usage for setValue()?
> 
> I want the config page to show the newly saved params and not the default ones and this looks like it might be exactly what i need

https://github.com/tzapu/WiFiManager/issues/1050

tablatronix, 2020-05-14:
- [ ] make sure there is an example for this