tzapu/WiFiManager #535: Getting error conecting to MQTT Broker using ESP8266-01 -  WiFiManager v0.12.0  and PubSubClient v2.6.0

xchelox, 2018-02-21:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Development

Hardware:			esp01

Core Version:      	2.4.0, staging

### Description
Hi! Before say anything i whant to give you thanks for this great library. Now, i have a problem and i cant find the solution. Maybe you can help me. Using normal wifi connection i can connect with MQTT Broker, but using WiFiManager i cant get connected up. The MQTT Broker's IP and Port are ok, sience i can connect it using just wifi. I love the WiFiManager library but i just need to solve this issue. Can you help me please?

Many thabks in advance.
Regards

### Settings in IDE

Module:  			
"Generic ESP8266 Module"

Additional libraries: 
#include <ESP8266WiFi.h>
#include <WiFiManager.h>
#include <PubSubClient.h>

### Sketch

#include <ESP8266WiFi.h>
#include <WiFiManager.h>
#include <PubSubClient.h>

WiFiClient espClient;
const char* MQTT_server = "192.168.0.5";
PubSubClient client(espClient);

void mqttCallback(char* topic, byte* payload, unsigned int length) {
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)payload[i]);
  }
  Serial.println();

  Serial.print("ret: ");
  Serial.println((int)payload[1]);
  if( (int)payload[1] == 49)//49 es el ascii del 1 -> es muy tarde y no quiero seguir renegando con este cast
  {
    Serial.println("Prendiendo led");
    // Please don't send more that 10 values per second.
  }
  else 
  {
    Serial.println("Apagando led");
    // Please don't send more that 10 values per second.
  }
}

void mqttReconnect() {
    // reconnect code from PubSubClient example
    
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection at: ");
    Serial.print(MQTT_server);
    Serial.print(":");
//    Serial.print(atoi(mqtt_port));
    Serial.print(" ... ");
    // Create a random client ID
    String clientId = "ESP8266Client-";
    clientId += String(random(0xffff), HEX);
    // Attempt to connect
    if (client.connect(clientId.c_str())) {
      Serial.println("connected");
      // Once connected, publish an announcement...
      //client.publish("event", "hello world");
      // ... and resubscribe
      client.subscribe("boton");
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 3 seconds");
      // Wait 5 seconds before retrying
      delay(3000);
    }
  }
}

void setup() {
   Serial.begin(115200);
   WiFiManager wifiManager;
  wifiManager.setTimeout(180);

  if(!wifiManager.autoConnect("AutoConnectAP")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    ESP.reset();
    delay(5000);
  } 

  Serial.println("connected...yeey :)");
  client.setServer(MQTT_server,1883);
  delay(10000);
  client.setCallback(mqttCallback);
}

void loop() {
  if (!client.connected()) {
    mqttReconnect();
  }
  client.loop();
}

### Debug Messages

```
messages here
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 3
*WM: IP Address:
*WM: 192.168.0.104
connected...yeey :)
Attempting MQTT connection at: 192.168.0.5: ... failed, rc=-2 try again in 3 seconds
Attempting MQTT connection at: 192.168.0.5: ... failed, rc=-2 try again in 3 seconds
Attempting MQTT connection at: 192.168.0.5: ... failed, rc=-2 try again in 3 seconds
Attempting MQTT connection at: 192.168.0.5: ... failed, rc=-2 try again in 3 seconds
Attempting MQTT connection at: 192.168.0.5: ... failed, rc=-2 try again in 3 seconds
Attempting MQTT connection at: 192.168.0.5: ... failed, rc=-2 try again in 3 seconds
Attempting MQTT connection at: 192.168.0.5: ... failed, rc=-2 try again in 3 seconds
Attempting MQTT connection at: 192.168.0.5: ... failed, rc=-2 try again in 3 seconds

tablatronix, 2018-02-21:
Hmm odd
Did you try development branch yet?

xchelox, 2018-02-21:
Not yet, you sugest me to try it? Do you think that on new branch it will
be solved?

Regards

xchelox, 2018-02-21:
What barnch do you think that i shuld test?

Thanks
Regards

tablatronix, 2018-02-21:
development

tablatronix, 2018-02-21:
please use code fences for your code in case someone wants to copy it

SteveRMann, 2019-08-14:
Any resolution to this?  I am having the same issue.

tablatronix, 2019-08-14:
try development branch ?

SteveRMann, 2019-08-15:
It was my error, I had two Wemos boards running the same sketch, which meant that the second sketch would not connect to the MQTT server.

```
// The constructor MUST be unique on the network.
WiFiClient telephoneClient;
PubSubClient client(telephoneClient);
```

tablatronix, 2019-08-15:
is the obj name the default host name???

SteveRMann, 2019-08-15:
No.  My project is a telephone prop and while the hostnames are "telephone" and "telephone1", I forgot to make the WiFiClient constructor names unique.  One is the working model and the second is on my bench for further development. (Is a project ever "done"?)

Two problems with WiFiManager have me putting this branch away and going back to hard-coding the WiFi credentials.  I am attracted to WiFiManager because the end users will not have access to the sketch, or even know what an Arduino IDE is.  It has to be truly turnkey.
Problems:
1) I can't change the credentials.  I have two WiFi LANs here, one for everyday use and one for testing.  If I start the AP then enter the second LAN ssid and password, everything appears to work, but the Wemos connects to the original LAN, "already connected", as if I did nothing.  I added WiFi.disconnect();  before starting the AP, and that seems to fix that issue.  But it has received light testing.
2) The WiFiManager doesn't handle a bad password well.  In my testing it never returns to where it was called and there is no way to tell the user that the password isn't accepted.
3) (This is the final nail in the coffin) I realized that the MQTT broker credentials were still hard-coded and adding the MQTT broker login credentials to the AP website is simply way above my pay grade.

But, overall, I will be using WiFiManager in future projects where I don't need the MQTT PubSub client.

tablatronix, 2019-08-15:
All those are easily solvable, and I am not sure what the first one is, there are no known save bugs in the development branch, you should open a new issue for that and provide logs.

No idea what #2 means as development branch specifically lets you get the last status for that very reason..

SteveRMann, 2019-08-15:
"Provide logs"- where are the logs you would like to see?
2) I prefer to work with stable releases through the IDE Library Manager.  If there is a way to get the status of the connect, I am not seeing it.  I do see a "4" in the comm screen whi, I think, is the bad password error, but I don't see a method of getting that error status in my sketch.  Even so, I have no way to tell the user that the password is bad.  I would expect to see this from the WiFi Manager in an AP web page.
3) do you have a specific example?

tablatronix, 2019-08-16:
serial output with debugging on..

example minimal code to reproduce?