tzapu/WiFiManager #537: wifiManager.autoConnect crashes ESP on MDNS.begin with Fatal exception 28(LoadProhibitedCause):

chadham, 2018-02-21:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Development
Version 0.12.0 (per Arduino Library Manager)
Hardware:			NodeMcu 1.0 (ESP12-E module)

Core Version:      	Version 0.12.0 (per Arduino Library Manager)

### Description

Problem description: If wifiManager.autoconnect is used to connect the esp8266, it crashes when MDNS.begin(..)  is called

### Settings in IDE

Module:  			NodeMcu 1.0 (ESP12-E module)

Additional libraries: 
ESP8266mDNS.h
### Sketch
`#include <Arduino.h>
#include <ESP8266mDNS.h>
#include <WiFiManager.h>

WiFiManager wifiManager;

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);
  WiFi.disconnect();
  wifiManager.autoConnect("DEVICE", "123456789");
  /*
  WiFi.begin("TrojanBackdoorVirus", "1234567890123");
  while (WiFi.status() != WL_CONNECTED) { 
    Serial.print("."); 
    delay(1000);
  }
  */
  Serial.print("\nConnected to "); Serial.print(WiFi.SSID()); Serial.print(" ,IP address is "); Serial.print(WiFi.localIP()); Serial.print(" and signal strength is "); Serial.println(WiFi.RSSI());

  if (!MDNS.begin("hello")) { Serial.println("Error setting up MDNS responder!"); }
  else { Serial.println("Started MDNS"); }
  
}

void loop() {}
`

### Debug Messages
USING WIFIMANAGER.AUTOCONNECT
```
rl d⸮⸮| ⸮l⸮|⸮$⸮c|⸮⸮⸮⸮;⸮c⸮c⸮⸮g'⸮lng⸮⸮⸮bx⸮⸮l;l{lx⸮o⸮⸮d⸮⸮#g⸮|d⸮⸮#⸮⸮no⸮ $⸮⸮l ⸮ogl n{⸮⸮⸮gc⸮ls⸮⸮n#⸮d⸮⸮⸮c⸮l ⸮⸮g⸮*WM: 
*WM: AutoConnect
bcn 0
del if1
mode : sta(2c:3a:e8:43:17:f7)
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
mode : sta(2c:3a:e8:43:17:f7) + softAP(2e:3a:e8:43:17:f7)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
*WM: SET AP STA
*WM: 
*WM: Configuring access point... 
*WM: DEVICE
*WM: 123456789
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
add 1
aid 1
station: b0:70:2d:82:d3:90 join, AID = 1
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
scandone
*WM: Scan done
*WM: DUP AP: TrojanBackdoorVirus
*WM: DUP AP: TrojanBackdoorVirus
*WM: HP-Print-06-LaserJet 200
*WM: -40
*WM: TrojanBackdoorVirus
*WM: -54
*WM: Iro
*WM: -66
*WM: ASUS
*WM: -89
*WM: Sent config page
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
scandone
switch to channel 6
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 5
cnt 

connected with TrojanBackdoorVirus, channel 6
dhcp client start...
ip:10.0.1.111,mask:255.255.255.0,gw:10.0.1.1
*WM: Connection result: 
*WM: 3
station: b0:70:2d:82:d3:90 leave, AID = 1
rm 1
bcn 0
del if1
pm open,type:2 0
mode : sta(2c:3a:e8:43:17:f7)
sl

Connected to TrojanBackdoorVirus ,IP address is 10.0.1.111 and signal strength is -59
Fatal exception 28(LoadProhibitedCause):
epc1=0x40225ba3, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000098, depc=0x00000000

Exception (28):
epc1=0x40225ba3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000098 depc=0x00000000

ctx: cont 
sp: 3fff0670 end: 3fff0aa0 offset: 01a0

>>>stack>>>
3fff0810:  fffffffe 00000000 4010100f 3fff284c  
3fff0820:  4021ffd0 3fff1d5c 3fff145c 3fff3044  
3fff0830:  3fff2dfa 40231e47 3fff284c 3fff1344  
3fff0840:  00000000 40220c83 3ffee1ec 40106b9c  
3fff0850:  3fff2dfa 40104456 3fff1d5c 3fff3044  
3fff0860:  0000002e 3fff0f00 3fff1d5c 401043d1  
3fff0870:  3fff2db4 3ffe973d 3fff184f 40222a3f  
3fff0880:  3ffee8f8 3ffee9f0 40224b34 3fff0f35  
3fff0890:  3fff0f00 3fff0f00 3fff2db4 4010453d  
3fff08a0:  40211f93 3fff2db4 3fff2dfa 40211f9c  
3fff08b0:  00000008 3fff08e0 4022451b 3ffeea10  
3fff08c0:  3ffee9f0 40224b34 3ffee8f8 3fff2e08  
3fff08d0:  3fff2db4 3fff3368 3fff0f00 40218031  
3fff08e0:  005e0001 4000fb00 60000200 3fff0990  
3fff08f0:  00000018 3fff3368 3fff2db4 40218ab8  
3fff0900:  40222b38 3ffee1ec 00000000 00000000  
3fff0910:  3fff0984 00000001 00000000 00000002  
3fff0920:  3fff0e60 00000004 3ffe8a81 40219130  
3fff0930:  4023a9ba 3ffe8a95 3ffe8a81 3fff3364  
3fff0940:  00000000 3fff2db4 3fff2e20 40218af8  
3fff0950:  3fff0f00 3fff0980 00000004 3fff3364  
3fff0960:  00000016 3fff0f00 3fff3364 40218289  
3fff0970:  3fff0f00 3fff0980 00000004 40211ff8  
3fff0980:  00000494 0104a8c0 3fff105c 40212034  
3fff0990:  3fff3368 3fff0f00 3fff2ebc 4021835a  
3fff09a0:  3fff3314 3ffef76c 3fff0a50 00000020  
3fff09b0:  3fff0a1c 3fff0f00 3fff3364 40218533  
3fff09c0:  3fff0a1c 3ffe98d4 3fff0f00 402185bf  
3fff09d0:  3fff0900 00000001 3fff0a50 00000001  
3fff09e0:  3ffef724 3ffef768 3ffef724 402025ce  
3fff09f0:  00000000 3ffef768 3fff0a40 40100690  
3fff0a00:  3ffef98c 000003af 00000000 4020ad78  
3fff0a10:  00000000 3ffef768 3ffef750 fb0000e0  
3fff0a20:  00000000 3ffef768 3ffef724 3ffefa70  
3fff0a30:  00000000 3ffef768 3ffef724 4020386a  
3fff0a40:  3fff2d3c 3fff0a70 40202234 402026f0  
3fff0a50:  00000000 00000000 3ffef9b0 4020a1e0  
3fff0a60:  3fffdad0 3ffef768 3ffef9b0 40202128  
3fff0a70:  3ffe93b8 6f01000a 00000000 feefeffe  
3fff0a80:  feefeffe 00000000 3ffefa68 4020ae84  
3fff0a90:  feefeffe feefeffe 3ffefa80 40100710  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
v4ceabea9
~ld
scandone
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 0
mode : sta(2c:3a:e8:43:17:f7) + softAP(2e:3a:e8:43:17:f7)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
*WM: SET AP STA
*WM: 
*WM: Configuring access point... 
*WM: DEVICE
*WM: 123456789
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
```

Works fine when using WiFi.being(said, password)

tablatronix, 2018-02-22:
Well for one you are removing wifi credentials every time you reboot
`  WiFi.disconnect();` deletes credentials !

tablatronix, 2018-02-22:
cannot reproduce with any combination of versions

chadham, 2018-02-22:
Wifi.disconnect call was intentional as I want to force wifiManager to load each time (for testing this issue, will be removed once fixed).

I stripped it down to the minimal code per above, and am able to reproduce on 2 NodeMCU's to eliminate (or at least reduce) the chance of flash corruption on one. 

Also deleted and downloaded latest version from here.

Any chance Im on a older firmware version on the nodeMCU? How do I check the firmware version?

tablatronix, 2018-02-22:
What is your esp core?

chadham, 2018-02-22:
I’m not sure. It’s a stock wemos nodemcu dev board. How do I tell what the core version is?

tablatronix, 2018-02-22:
are you using esp staging, or 2.4.0 ?

chadham, 2018-02-22:
Im sorry, but I don't understand the question.
If you are asking about the WifiManager version, its whatever you get when you download the library from the clone or download button.

chadham, 2018-02-22:
its 2.4.0

tablatronix, 2018-02-22:
k

tablatronix, 2018-02-22:
still cannot reproduce, works for me

```
*WM: Connecting as wifi client...
scandone
switch to channel 11
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1

connected with SSID, channel 11
dhcp client start...
cnt 
ip:10.80.21.143,mask:255.255.240.0,gw:10.80.16.1
*WM: Connection result: 
*WM: 3
station: dc:0c:5c:e2:24:41 leave, AID = 1
rm 1
bcn 0
del if1
pm open,type:2 0
mode : sta(2c:3a:e8:43:1f:5e)

Connected to SSID ,IP address is 10.80.21.143 and signal strength is -65
Started MDNS
```

tablatronix, 2018-02-22:
If you use arduino ide you can install esp exception decoder tool and decode that exception. Also erase with esptool.

Not sure what else to suggest

chadham, 2018-02-22:
Still happening. Here is the stack trace

`Decoding 59 results
0x4022a558: ieee80211_output_pbuf at ?? line ?
0x4023d2cf: pp_attach at ?? line ?
0x4023d2cf: pp_attach at ?? line ?
0x40104456: glue2esp_linkoutput at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c line 292
0x401043d1: glue2esp_linkoutput at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c line 263
0x4010453d: new_linkoutput at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c line 240
0x4021c523: ethernet_output at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c line 305
0x4021c52c: ethernet_output at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c line 305
0x402225c1: etharp_output_LWIP2 at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/etharp.c line 893
0x40223048: ip4_output_if_opt_src at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c line 962
0x402236bc: mem_malloc at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/mem.c line 136
0x40223088: ip4_output_if_opt at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c line 788
0x40222819: igmp_send at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c line 570
0x4021c588: do_memp_malloc_pool at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/memp.c line 231
0x4021c5c4: memp_malloc at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/memp.c line 231
0x402228ea: igmp_lookup_group at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c line 570
0x40222ac3: igmp_start_timer at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c line 570
:  (inlined by) igmp_joingroup_netif at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c line 521
0x40222b4f: igmp_joingroup at /home/david/dev/esp8266/origin/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/igmp.c line 570
0x402118b2: MDNSResponder::_listen() at /Users/mchadha/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/libraries/ESP8266mDNS/ESP8266mDNS.cpp line 262
0x40100690: free at /Users/mchadha/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/umm_malloc/umm_malloc.c line 1737
0x402142d0: operator delete(void*) at /Users/mchadha/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/abi.cpp line 57
0x40212dd2: MDNSResponder::begin(char const*) at /Users/mchadha/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/libraries/ESP8266mDNS/ESP8266mDNS.cpp line 262
0x402115a4: _M_manager at /Users/mchadha/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/libraries/ESP8266mDNS/ESP8266mDNS.cpp line 262
0x402119d4: operator() at /Users/mchadha/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/libraries/ESP8266mDNS/ESP8266mDNS.cpp line 262
:  (inlined by) _M_invoke at /Users/mchadha/Library/Arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/xtensa-lx106-elf/include/c++/4.8.2/functional line 2071
0x40213890: String::~String() at /Users/mchadha/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/WString.cpp line 746
0x40209035: BIOTMQTT::discoverMQTTServer() at /var/folders/4w/bd288hf91n50d8r8tcm8cmf00000gn/T/arduino_build_958116/sketch/BIOTMQTT.cpp line 81
0x40106cc0: vPortFree at /Users/mchadha/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/heap.c line 47
0x402530d0: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x4023d2cf: pp_attach at ?? line ?
0x4010020c: _umm_free at /Users/mchadha/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/umm_malloc/umm_malloc.c line 1291
0x40100670: _umm_realloc at /Users/mchadha/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/umm_malloc/umm_malloc.c line 1641
:  (inlined by) realloc at /Users/mchadha/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/umm_malloc/umm_malloc.c line 1713
0x402138eb: String::changeBuffer(unsigned int) at /Users/mchadha/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/WString.cpp line 746
0x40213937: String::reserve(unsigned int) at /Users/mchadha/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/WString.cpp line 746
0x402530d0: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40213c1b: String::concat(char const*, unsigned int) at /Users/mchadha/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/WString.cpp line 746
0x40106db5: __wrap_spi_flash_read at /Users/mchadha/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/core_esp8266_phy.c line 267
0x402530d0: sleep_reset_analog_rtcreg_8266 at ?? line ?
0x40213c98: String::concat(char) at /Users/mchadha/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/WString.cpp line 746
0x402077f2: ArduinoJson::Internals::JsonWriter   >::writeRaw(char) at /Users/mchadha/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson/Deserialization/../Serialization/JsonWriter.hpp line 68
0x4020788f: ArduinoJson::Internals::JsonWriter   >::writeString(char const*) at /var/folders/4w/bd288hf91n50d8r8tcm8cmf00000gn/T/arduino_build_958116/sketch/BIOTDevice.cpp line 209
[... cut ...]
0x40209589: BIOTMQTT::begin(BIOTDevice*) at /var/folders/4w/bd288hf91n50d8r8tcm8cmf00000gn/T/arduino_build_958116/sketch/BIOTMQTT.cpp line 38
0x40208941: setup at /Users/mchadha/Documents/Arduino/libraries/TaskScheduler/src/TaskScheduler.h line 600
0x402143dc: loop_wrapper at /Users/mchadha/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/core_esp8266_main.cpp line 57
0x40100710: cont_norm at /Users/mchadha/Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/cont.S line 109`

tablatronix, 2018-02-22:
Whats all that json stuff?

chadham, 2018-02-22:
Replaced test harness stack above

tablatronix, 2018-02-22:
This looks like an mdns error

chadham, 2018-02-22:
Interesting question is why it works if I hardcode the ssid and password and fails if I connect through wifimanager

tablatronix, 2018-02-22:
We use webserver and dnsserver, we switch mode several times?

Try making sure the wifi is at the right state or a known state, before starting mdns, set mode etc.

I think there is a mdns debug flag in the files.

Not entirely sure whats going on here, but that stack seems far from wifimanager.

Another thing you can try put wifimanager in a different scope.

tablatronix, 2018-02-22:
What lwip do you have selected in arduino?

chadham, 2018-02-22:
Thanks, let me try checking the state.

Should  I not assume that wifimanager.Connect blocks until WiFi is successfully setup?

Ie do u wait for connection success confirm before returning?

tablatronix, 2018-02-22:
Master, Yes by default, development has a lot more features to choose

tablatronix, 2018-02-22:
Actually Lwip might only be an option in staging esp.

chadham, 2018-02-22:
I’ll check WiFi state once back at my computer but if you only release on successful connection, count if that’s what is causing it. Stay tuned.

chadham, 2018-02-22:
What is lwip and how do I check?

tablatronix, 2018-02-22:
Have you tried esptool erase ?

chadham, 2018-02-22:
No. You think it’s corruption in the flash? I tied two different esp’s, same issue.

Let me read up on esptool erase

chadham, 2018-02-22:
Looks complex .. a couple of quick questions..

1. Im on a Mac, any issues running esptool?
2. Once I erase, what do I need to do? Can I upload my sketch through Arduino's IDE as normal or do I need to put a boot loader or something else back on the ESP via ESP tool?
3. Im assuming I can brick the device .. in which case I should take the time to understand what Im doing vs just doing it immediately..

appreciate any guidance

chadham, 2018-02-22:
iwi variant is V2 Prebuilt

chadham, 2018-02-22:
Checked and confirmed WiFi.Status() is reporting connected (3)

tablatronix, 2018-02-23:
Esptool just erases flash
Needs python 
Esptool.py -p /dev/tty/... erase_flash

Upload fine after

tablatronix, 2018-02-23:
There are corruption bugs in 2.4.0

chadham, 2018-02-24:
issue remains after erase_flash....

Not sure what else to try here...

tablatronix, 2018-02-24:
maybe someone else will reproduce, I have not been able to.
I even added mdns to the development branch and it worked fine.

tablatronix, 2018-02-24:
do you get this on release as well?

https://github.com/tzapu/WiFiManager/releases/tag/0.12

tablatronix, 2018-02-24:
maybe post to esp issues and someone will know what those are.

lwip ieee80211_output_pbuf etc

tablatronix, 2018-03-14:
Is this still an issue?

MarcFontaine, 2018-03-14:
MDNS did not work for me with 2.4.0 and work with 2.4.1

chadham, 2018-03-14:
Yes :(

tablatronix, 2018-03-14:
Try 2.4.1 or staging maybe. Still not seeing this, could it be router specific?

myk3y, 2018-04-05:
Im getting the same Exception 28 with MDNS and WiFiManager - 2.4.0 and 2.4.1 using the Arduino IDE.

Might try PlatformIO.

tablatronix, 2018-04-05:
hmm sketch code?
What chip revision ?

tablatronix, 2018-04-05:
There are 1 or 2 known issues with esp32 I have found, I have implemented some workarounds in the library, are you seeing this with the latest commits , have you pulled recently ?

I suspect this is a race condition issue , which makes sense since wm exposes it, and normal use does not, since wm switches modes back and forth alot.

myk3y, 2018-04-06:
Not using an esp32 - esp12e based Wemos mini lite.

Remove wifimanager from the equation and the exact same code works perfectly

Regards,

Mike