tzapu/WiFiManager #539: Allow changing where debug output goes.

liebman, 2018-02-27:
In my case I have a GPS connected to Serial and send all diagnostic output to Serial1.  I have this working and can create a pull request.  (actually was about to submit a PR against master and saw that you changed the contributing guide)

Suggesting: ```setDebugStream(Stream& dbg_stream);``` Making this work: ```wm.setDebugStream(Serial1);```

tablatronix, 2018-02-27:
#540

closed 2018-02-27 by tablatronix

cjcharles0, 2018-08-21:
@tablatronix Hello, I found this fix in the development branch but not in master, is this something I should just port myself, or is there a plan to put it into master?

Its exactly what Im after, but given the last git comment was 'well shit' I fear the branch is probably not very reliable yet? (As great as I know it will end up eventually)

Thanks in advance team!