tzapu/WiFiManager #540: optional debug port

tablatronix, 2018-02-27:
https://github.com/tzapu/WiFiManager/commit/0ae9ca3d558d3945d39a1129ea14d8c45dcc87f9

tablatronix, 2018-02-27:
`WiFiManager::WiFiManager(Stream& consolePort)`

closed 2018-02-27 by tablatronix

cjcharles0, 2018-08-21:
@tablatronix Hello, I found this fix in the development branch but not in master, is this something I should just port myself, or is there a plan to put it into master?

Its exactly what Im after, but given the last git comment was 'well shit' I fear the branch is probably not very reliable yet? (As great as I know it will end up eventually)

Thanks in advance team!

tablatronix, 2018-08-21:
no plan, its reliable, but still being beta tested and still fixing esp stuff that changes,

tablatronix, 2018-08-21:
now that esp8266 has a new release
and esp32 is officially released and supported in arduino ide, I just need to fix up some examples I an reworking.