tzapu/WiFiManager #541: Feature - Option to continue AP Mode and  load SPIFFS

akshaim, 2018-03-01:
----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Master
Hardware:			ESP-12e (NodeMCU) 

Kindly add the option in configuration page to continue operation in AP mode. Currently I am not able to load the webpage from SPIFF unless connected to WiFi Network.

tablatronix, 2018-03-01:
you mean leave the ap on ?
that's already supported

#500

akshaim, 2018-03-03:
Thanks for the quick response.  I want to access the application in ESP8266 Flash even when there is no WiFi AP nearby.
Currently, I am able to launch the app only after WiFi manager exits, which happens only after ESP8266 connects to a WiFi AP. An option in configuration page to continue in AP mode to restart webserver and launch the local app could help.

tablatronix, 2018-03-03:
this all exists in the development banch, not sure exactly what you want to do.