tzapu/WiFiManager #542: ESP32 Testing doesn't get past WiFi.enableSTA(true)

renegadeandy, 2018-03-01:
I can't seem to get very far using the development branch.

### Basic Infos
Using ESP32 dev board oooouuuu board.

#### Hardware
WiFimanager Branch/Release:  Development

Hardware:			ESP32

Core Version:      	2.4.0, staging

### Description

When I try to incorporate the dev WiFiManager into an old project which ran fine on esp8266, It seems to hang and crash directly after WiFi.enableSTA(true);

My call from main sketch is as follows:
```

 //Start WiFi Setup
  Serial.println("Setting up WiFi...");
  WiFiManager wifiManager;
  wifiManager.autoConnect("AP Setup", NULL);

```

The added Debug to autoconnect is as follows:
```

boolean WiFiManager::autoConnect(char const *apName, char const *apPassword) {
  DEBUG_WM(F("AutoConnect"));
 //digitalWrite(apLED,HIGH);
  // attempt to connect using saved settings, on fail fallback to AP config portal
    DEBUG_WM(F("2"));
  WiFi.enableSTA(true);
    DEBUG_WM(F("22"));
  _usermode = WIFI_STA;
  DEBUG_WM(F("222"));
```

Note my debug output is:
```
Setting up WiFi...
*WM: AutoConnect 
*WM: 2 

```
As we can see, we never hit the *WM: 22 message. Is there some kind of setup I need to do? Do I need to declare I am on an ESP32 to the library somehow?

### Settings in IDE

Module:  			ESP32 DEV BOARD

tablatronix, 2018-03-01:
Whats the debug from wifimanager???

renegadeandy, 2018-03-01:
How do I turn that on? Not sure what you mean?

tablatronix, 2018-03-01:
It should be on, unless you are setting it to off in your code.
It outputs lots of info

renegadeandy, 2018-03-01:
I see no debug output at all in my code, apart from what i've shown to you above?

tablatronix, 2018-03-01:
hmmm

tablatronix, 2018-03-01:
works for me

```
*WM: AutoConnect 
*WM: ESP32 event handler enabled 
*WM: Connecting as wifi client... 
*WM: wifi station disconnect 
*WM: Connecting to saved AP 
*WM: connectTimeout not set, ESP waitForConnectResult... 
WM: EVENT: WIFI_REASON: 201
*WM: Event: SYSTEM_EVENT_STA_DISCONNECTED, reconnecting
*WM: Connection result: WL_NO_SSID_AVAIL
*WM: lastconxresult: WL_NO_SSID_AVAIL
*WM: wifi station disconnect 
*WM: Disabling STA 
*WM: Enabling AP 
*WM: StartAP with SSID:  AutoConnectAP
*WM: AP has anonymous access! 
*WM: SoftAP Configuration 
*WM: -------------------- 
*WM: ssid:             AutoConnectAP
*WM: password:         
*WM: ssid_len:         13
*WM: channel:          1
*WM: authmode:        
*WM: ssid_hidden:     
*WM: max_connection:   4
*WM: beacon_interval:  100(ms)
*WM: -------------------- 
*WM: AP IP address: 192.168.4.1
*WM: setupConfigPortal 
*WM: Starting Web Portal 
*WM: HTTP server started 
*WM: Scan done 
*WM: Config Portal Running, blocking, waiting for clients... 
```

renegadeandy, 2018-03-01:
Ahh, well you see you never get past WiFi.enableSTA(true); so the rest of the debug output never gets executed!

tablatronix, 2018-03-01:
ohh, well that is not wifimanager then, there is nothing in that function to get stuck, is your module crashing?

tablatronix, 2018-03-01:
enable esp debugging also maybe?

renegadeandy, 2018-03-01:
No, although I have made a couple of alterations, and now I see 

Brownout detector was triggered....

Wondering if this is a USB power problem...

renegadeandy, 2018-03-01:
Yes! This was the problem, a really rubbish USB Hub, time to put that one back into the garage, sorry tablatronix!

closed 2018-03-01 by renegadeandy

tablatronix, 2018-03-01:
Yeah power issue, below 300 ma or so, the wifi radio cannot turn on

tablatronix, 2018-03-01:
added a catch for this

renegadeandy, 2018-03-01:
Fantastic - thanks mate!