tzapu/WiFiManager #543: AutoConnectWithFSParametersAndCustomIP.ino forgets reverts to AP after power cycle

fverschure, 2018-03-02:
#### Hardware
WiFimanager Branch/Release:  0.12
Hardware:			wemos d1 mini
Core Version:      	2.4.0, staging
### Description

If i use the example AutoConnectWithFSParametersAndCustomIP.ino it reverts to AP mode after a power cycle. Is this by design or am i missing something?
The custom parameters are retrieved successfully from the SPIFFS

tablatronix, 2018-03-02:
esp8266 or esp32?

fverschure, 2018-03-02:
esp8266, on wemos d1 mini.
using version 0.12 in arduino ide. Arduino ide version: 1.8.5

After setting the settings the JSON is succesfully written and can be read again on reboot, however if I remove the power and start the ESP again it powers up the AP again and seems to have forgotten its SSID and password.

Tried on multiple modules with the same result.

tablatronix, 2018-03-02:
no idea, try development version, lots of bugs were fixed

fverschure, 2018-03-02:
tried development version. Look totally different, but result is the same. 

``
*WM: No saved credentials, skipping wifi 
*WM: Connection result: WL_NO_SSID_AVAIL
*WM: lastconxresult: WL_NO_SSID_AVAIL
``

tablatronix, 2018-03-02:
sometimes config wont save if memory is corrupt, you could try erasing flash with esptool.

fverschure, 2018-03-02:
Thanks for your help. This did the trick indeed. Would love if this would be implemented inside the arduino IDE..

I do love the Wifimanager library!

tablatronix, 2018-03-02:
I think there is a nodemcu flasher that has a ui that will erase also

fverschure, 2018-03-04:
@tablatronix Thanks for the hint. I know the program. I guess  I prefer the commandline tool instead. Works a lot better.

I found another issue with the WifiManager Development branch. 
After saving settings it does not save the settings to spiffs.
Somehow saveconfig callback is not being set by wifimanager.

This does work in the Master branch build but there i am running into other issues.

See what happens with the development branch
`
*WM: Sent wifi save page 
*WM: Connecting to a new AP 
*WM: Connecting as wifi client... 
*WM: wifi station disconnect 
*WM: Connecting to new AP 
*WM: enableSTA PERSISTENT ON 
*WM: connectTimeout not set, ESP waitForConnectResult... 
*WM: Connection result: WL_CONNECTED
*WM: lastconxresult: WL_CONNECTED
*WM: Connect to new AP [SUCCESS] 
*WM: Got IP Address: 
*WM: 192.168.2.44 
*WM: disconnect configportal 
*WM: config portal exiting 
`

With the Master branch the SPIFFS file is being written, see below

`*WM: WiFi save
*WM: Parameter
*WM: server
*WM: mqtt.local
*WM: Parameter
*WM: port
*WM: 8080
*WM: Parameter
*WM: blynk
*WM: <REMOVED>
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 3
Should save config
connected...yeey :)
saving config
{
  "mqtt_server": "mqtt.local",
  "mqtt_port": "8080",
  "blynk_token": "<REMOVED>",
  "ip": "192.168.2.44",
  "gateway": "192.168.2.254",
  "subnet": "255.255.255.0"
}local ip
`
Any idea what is causing this to happen with the development build?

tablatronix, 2018-03-05:
The master should not also have that bug. Odd.
Ill test this, i have not yet tested hooks

tablatronix, 2018-03-05:
ok well settings don't get saved to spiffs, they are saved to flash by the esp sdk.

tablatronix, 2018-03-05:
I see
callback is only called on failure and if setBreakAfterConfig true.
Ill fix tomorrow if its a bug

tablatronix, 2018-03-06:
ok fixed the callback, it should be called like it was before, on save and failure (if breakafter)

fverschure, 2018-03-06:
Thanks! @tablatronix, This fixes the saving to SPIFFS indeed.

However I do still run into the following:
After configuring the AP via wifimanager it connects to wifi and runs the code successfully. 
When i reset the device via the resetbutton it also returns to the previous configured AP. 
However. If i unplug the device and leave it unpowered for several minutes and power it up again it looses the configured wifi settings and starts wifimanager again. 

I guess this is not expected behavior? I tested it with several modules and cleaned the flash with esptool before. 

When flashing the wifi credentials in the code this does not happen with the modules, so I am ruling out the flash  to be worn out with these modules.

tablatronix, 2018-03-06:
suggestions

add a 3 second delay at setup, WiFi.printDiag(Serial), and see what it outputs, see if your credentials are there first.

tablatronix, 2018-03-06:
If you still can reproduce, create a new issue for this

closed 2018-03-06 by tablatronix

fverschure, 2018-03-09:
Hi, 
thanks for your answer.
with a 5 seconds delay at startup the microcontroller does not forget the stored access point.

tablatronix, 2018-03-09:
That is odd, it should not take that long to startup wifi and it should happen before setup

tablatronix, 2018-03-09:
do you have a simple sketch ?
Does it happen with autoconnect example?

fverschure, 2018-03-09:
I will check this weekend. My sketch is based on the example sketch with custom variables. My hunch is that it is this combination which is causing this behavior.

tablatronix, 2018-03-09:
Are you using static ip?

fverschure, 2018-03-22:
Hi, Sorry for the large delay. Was a bit occupied last weeks. I do not use static IP.

tablatronix, 2018-03-22:
Can you create a new issue for this particular issue with code if you have