tzapu/WiFiManager #544: ESP8266 does not reconnect after exiting promiscuous mode

kubark42, 2018-03-02:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  0.12.0

Hardware:			Huzzah

Core Version:      	2.4.0

### Description

When promiscuous mode is turned on and then off, WifiManager will no longer connect.

### Settings in IDE

Module:  			Huzzah ESP8266

### Sketch

```cpp
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>         //https://github.com/tzapu/WiFiManager

WiFiManager wifiManager;

void setup() {
  // put your setup code here, to run once:
  Serial1.begin(115200);
  Serial1.setDebugOutput(true); // Turn on WIFI debug info

  wifiManager.autoConnect("ssid", "password");

  //if you get here you have connected to the WiFi
  Serial1.println("connected...yeey :)");
}

void loop() {
  enableWifiMonitorMode();
  Serial1.print("before (ms): ");
  Serial1.println(millis());
  enableWifiClient();
  Serial1.print("after (ms): ");
  Serial1.println(millis());
}

void enableWifiMonitorMode()
{
  // Send ESP into promiscuous mode. At this point, it stops being able to connect to the internet
  Serial1.println("Turning on wifi monitoring.");
  //    wifi_set_opmode(WIFI_STA);
  wifi_set_promiscuous_rx_cb(receiveWifiSniff_cb);
  wifi_promiscuous_enable(true);
}

void enableWifiClient()
{
  // Send ESP into promiscuous mode. At this point, it stops being able to connect to the internet
  Serial1.println("Turning off wifi monitoring.");
  wifi_promiscuous_enable(false);

  if (!wifiManager.autoConnect("ssid", "password")) {
    Serial1.println("failed to connect, timed out...");
  }

}

void receiveWifiSniff_cb(uint8_t *buf, uint16 len)
{
}

```

### Debug Messages

```
rscandone
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with SSID, channel 1
dhcp client start...
ip:192.168.86.109,mask:255.255.255.0,gw:192.168.86.1
connected...yeey :)
Turning on wifi monitoring.
state: 5 -> 0 (0)
rm 0
before(ms): 4158
Turning off wifi monitoring.
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with SSID, channel 1
dhcp client start...
pm open,type:2 0
```

tablatronix, 2018-03-06:
Try deveopment, master is not supported

henriquebol, 2018-05-22:
Same problem here. Did you find a solution?