tzapu/WiFiManager #545: Server not working after portal connection. Only after manual reset.

jwktje, 2018-03-03:
### Basic Infos
Webserver responses only work on a reset after successfully connecting to a Wifi Network via the portal.

#### Hardware
WiFimanager Branch/Release:  ESP32

Hardware:		ESP32 (Wemos Lolin32)

### Description

I'm using the bbx10 webserver and the ESP32 branch of WifiManager. Everything works great and stable since running these versions. But my only problem is that my server responses get a ERR_CONNECTION_REFUSED. When I reset the module and it already has the credentials, everything runs fine.

### Sketch

```
//Init webserver
WebServer server(80);

void setup() {
  
  //Begin Serial
  Serial.begin(115200);
 
  //Manager setup
  WiFiManager wifiManager;
  wifiManager.setBreakAfterConfig(true);
  //wifiManager.resetSettings();
  
  if (!wifiManager.autoConnect("ConfigureWifi")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again
    ESP.restart();
    delay(5000);
  }

  Serial.print ( "Connected with IP address: " );
  Serial.println ( WiFi.localIP() );

  server.begin();
  
  server.on("/", handle_root);
```

### Debug Messages

```
*WM: Free heap:  139096
*WM: ESP-IDF version:  v3.1-dev-239-g1c3dd23f-dirty
*WM: freeing allocated params!
*WM: unloading
*WM: -> captive.apple.com
*WM: <- Request redirected to captive portal
*WM: <- HTTP Root
*WM: -> 192.168.4.1
*WM: -> captive.apple.com
*WM: <- Request redirected to captive portal
*WM: <- HTTP Root
*WM: -> 192.168.4.1
*WM: -> captive.apple.com
*WM: <- Request redirected to captive portal
*WM: <- HTTP Root
*WM: -> 192.168.4.1
*WM: -> captive.apple.com
*WM: <- Request redirected to captive portal
*WM: <- HTTP Wifi
*WM: Free heap:  132700
*WM: ESP-IDF version:  v3.1-dev-239-g1c3dd23f-dirty
*WM: freeing allocated params!
*WM: unloading
*WM: Free heap:  132708
*WM: ESP-IDF version:  v3.1-dev-239-g1c3dd23f-dirty
*WM: freeing allocated params!
*WM: unloading
*WM: Scan done
*WM: AP:  -52 Network-redacted
*WM: AP:  -79 Network-redacted
*WM: AP:  -91 Network-redacted
*WM: Sent config page
*WM: <- HTTP Root
*WM: -> 192.168.4.1
*WM: -> captive.apple.com
*WM: <- Request redirected to captive portal
*WM: <- HTTP Root
*WM: -> 192.168.4.1
*WM: NUM CLIENTS: 1
*WM: -> captive.apple.com
*WM: <- Request redirected to captive portal
*WM: <- HTTP WiFi save 
*WM: Method: POST
*WM: Sent wifi save page
*WM: Connecting to a new AP
*WM: Connecting as wifi client...
*WM: wifi station disconnect
*WM: Connecting to new AP
*WM: connectTimeout not set, ESP waitForConnectResult...
*WM: Free heap:  131700
*WM: ESP-IDF version:  v3.1-dev-239-g1c3dd23f-dirty
*WM: freeing allocated params!
*WM: unloading
*WM: Free heap:  132008
*WM: ESP-IDF version:  v3.1-dev-239-g1c3dd23f-dirty
*WM: freeing allocated params!
*WM: unloading
*WM: Connection result: WL_CONNECTED
*WM: Connect to new AP [SUCCESS]
*WM: Got IP Address:
*WM: 192.168.2.23
*WM: disconnect configportal
*WM: disconnect configportal - softAPdisconnect failed
*WM: Free heap:  145384
*WM: ESP-IDF version:  v3.1-dev-239-g1c3dd23f-dirty
*WM: freeing allocated params!
*WM: unloading
*WM: Free heap:  145580
*WM: ESP-IDF version:  v3.1-dev-239-g1c3dd23f-dirty
*WM: freeing allocated params!
*WM: unloading
*WM: config portal exiting
Connected with IP address: 192.168.2.23
```

jwktje, 2018-03-03:
I found this gist but the BBX10 webserver misses this function it seems.
https://gist.github.com/tzapu/ecc0759829d30d5a6152

tablatronix, 2018-03-03:
what is in your loop ?
`*WM: unloading` this means your deconstructor is running, why is the class init and deinit over and over?

jwktje, 2018-03-03:
My loop is this:
```
timer.run(); //Run non-wifi related timed functions
timeClient.update(); //For updating the NTP time i'm using
  handleTheButton(); //Handling button inputs, not wifi related
  handleTheClient(); //Only wifi related function
```
The only relevant piece in my code is just calling this:
```
server.handleClient();
```
Other

tablatronix, 2018-03-03:
try the latest commit

jwktje, 2018-03-03:
Which branch? I'm on ESP32 now and pulled yesterday

tablatronix, 2018-03-03:
is it possible the webserver just doesn't support instances properly ?

jwktje, 2018-03-03:
Maybe yeah. I'm not fully up to speed on how the WebServer lib handles things. The gist I commented is for ESP8266 and the respective WebServer which has a reset function apparently.

Could I simulate this function somehow myself without changing the BBX10 Lib? As the Lib functions fine otherwise

tablatronix, 2018-03-03:
reset() is not part of the library, is a the smart pointer method.

std::unique_ptr<ESP8266WebServer> server;
server.reset(new ...)

tablatronix, 2018-03-03:
Hmm ill try to test this myself and see what i can come up with.

tablatronix, 2018-03-04:
for some reason this doesnt work

`server.reset(new my_WebServer(WiFi.localIP(), 80));`
it is running the wrong constructor and thinking ip is int, so port gets set to `xxx`. of the ip instead of 80

just remove it it should work

`  server.reset(new my_WebServer();`

here my gist for esp32 esp8266

https://gist.github.com/tablatronix/e57dc3096ea311dd0e59aa512b8fe6e7

tablatronix, 2018-03-04:
there is no wifiserver constructor for this

`_server(addr, port)`
so it takes the addr as port.

jwktje, 2018-03-04:
Just tried exactly what you put in the gist this but still the problem remains. The server keeps giving ERR_CONNECTION_REFUSED just after configuring the wifi. And after a reset it does handle routes fine. Obviously this was tested with `wifiManager.resetSettings()` commented out as otherwise a manual reset would remove the credentials so it could never work.

Anything else I could test? I don't even mind paying a small bug bounty for this or something? Would be awesome to get this patched.

tablatronix, 2018-03-05:
You used the gist verbatim?

tablatronix, 2018-03-05:
Do me a favor, port scan your esp. what os are you on i can tell you how

jwktje, 2018-03-05:
Just did a portscan on osx. This is the result:
```
Port Scan has started…

Port Scanning host: 192.168.2.23

Port Scan has completed…
```

tablatronix, 2018-03-05:
I am assuming you port scanned it after "HTTP Started"

jwktje, 2018-03-05:
Yep, but that does only get logged during the capacitive portal phase like so
`*WM: HTTP server started `

There is no HTTP started logged afterwards when resetting my server.

When I port scan on the AP ip when connected to it's own AP network I get the following, as expected:
```
Port Scan has started…

Port Scanning host: 192.168.4.1

	 Open TCP Port: 	80     		http
```

tablatronix, 2018-03-05:
The gist says
"Serial.println("HTTP server started");"

jwktje, 2018-03-05:
Yes I tested that first and then went back to my own sketch for testing. But the relevant calls are identical in order. Just to be sure, I ran the gist again. This is the log
```
*WM: Sent config page 
*WM: -> captive.apple.com 
*WM: <- Request redirected to captive portal 
*WM: <- HTTP Root 
*WM: -> 192.168.4.1 
*WM: Scan is cached 
*WM: <- HTTP WiFi save  
*WM: Method: POST
*WM: Sent wifi save page 
*WM: Connecting to a new AP 
*WM: Connecting as wifi client... 
*WM: wifi station disconnect 
*WM: Connecting to new AP 
*WM: connectTimeout not set, ESP waitForConnectResult... 
*WM: Connection result: WL_CONNECTED
*WM: lastconxresult: WL_CONNECTED
*WM: Connect to new AP [SUCCESS] 
*WM: Got IP Address: 
*WM: 192.168.2.23 
*WM: disconnect configportal 
*WM: disconnect configportal - softAPdisconnect failed 
*WM: config portal exiting 
connected...yeey :)
HTTP server started
192.168.2.23
*WM: freeing allocated params! 
*WM: unloading 
```

Portscan results are as I explained before. I get port 80 when connected to the AP. And it finds nothing after connecting. 

Anything else I can help to test? I did pull the latest commits yesterday from the Development branch

tablatronix, 2018-03-05:
I will try to test again on my side, because it was working for me before,  I do not have a working ap at the moment, so it might take a bit. For some reason esp32 doesn't see my hotspot like esp8266 does.

tablatronix, 2018-03-05:
were you also running the core dnsserver?

tablatronix, 2018-03-06:
Still no reproduce, even on port 80.
works for me, on captive portal save, and autoconnect success.

I added a 1 second delays before server.reset, not sure if that increased stability or not.

this helps testing also

```cpp
  server->on("/erase", []() {
    server->send(200, "text/plain", "erasing");
    WiFiManager wifiManager();
    wifiManager.erase();
    delay(1000);
    ESP.restart();
  });
```

jwktje, 2018-03-06:
Nope I just checked and I was using bbx10/DNSServer_tng as the DNS server. This might be the problem? I will check again with the core one. I remember needing the bbx10 DNS server as tipped in an issue a while back. But maybe core is stable enough now for esp32. 

Will report back soon! (tonight probably)

tablatronix, 2018-03-06:
I doubt it, dnsserver was just updated in core yesterday. Just making sure I am on the same page.

jwktje, 2018-03-06:
Could this just be a flaw in the ESP32 chip with this setup though? I just tested again with the latest commits for WiFiManager and nothing but the core libs and the bbx10 webserver. Both my code and your Gist, but no luck.

jwktje, 2018-03-06:
Wow wow wow wait. Just tested without explicitly including the WebServer.h and now it looks like it's working. I only commented out like so `#include <WebServer.h>`. Maybe the ESP32 arduino environment now has a more stable WebServer?

tablatronix, 2018-03-06:
That would not matter wm includes the webserver, there is no webserver in esp32 at all.

How exactly are you testing this ? Wipe settings, connecting to ap, saveing, then trying to connect to your webserver and it fails ?

Did you try enabling debug flag -DDEBUG_ESP_HTTP_SERVER and see what happens?

jwktje, 2018-03-06:
Yep exactly that is how I've been testing for the last hour. I will try the debug flag now and continue testing

jwktje, 2018-03-06:
I'm now getting the logs below with verbose debugging. It shows the AP_STOP event after I'm trying to reset my WebServer. Maybe this could be causing trouble? 
```
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 7 - STA_GOT_IP
*WM: Connection result: WL_CONNECTED
*WM: lastconxresult: WL_CONNECTED
*WM: Connect to new AP [SUCCESS] 
*WM: Got IP Address: 
*WM: 192.168.2.23 
*WM: disconnect configportal 
*WM: disconnect configportal - softAPdisconnect failed 
*WM: config portal exiting 
Server Started!
*WM: freeing allocated params! 
*WM: unloading 
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 14 - AP_STOP
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 14 - AP_STOP
```

tablatronix, 2018-03-06:
did you ever add that delay I mentioned?

jwktje, 2018-03-06:
Yep that's in my sketch. It is the only relevant change I made which is now causing it to work a majority of the times. As I now realize that uncommenting the WebServer library makes no difference.

Should I just increase it to a crazy amount of seconds and check my mileage?

tablatronix, 2018-03-07:
I just started getting crashed on my esp32 with this code for some reason.

I had to remove this 
```
  // server->on("/inline", []() {
  //   server->send(200, "text/plain", "this works as well");
  // });

````

whats more interesting is this

```
Exception Cause: Not found

0x40078000: ?? ??:0
0x40078000: ?? ??:0
0x40078a28: ?? ??:0
0x4000c46c: ?? ??:0
0x4000c477: ?? ??:0
0x400827d4: start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c:328
0x400829c0: call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c:203
0x4007887a: ?? ??:0
0x400788e1: ?? ??:0
0x40078a21: ?? ??:0
0x40078b4a: ?? ??:0
0x40007c31: ?? ??:0
0x4000073d: ?? ??:0
```

jwktje, 2018-03-07:
Hmm. Well, I am a webdeveloper first, and Arduino enthusiast second. So maybe I'm misunderstanding.

But does this indicate just instability in the Arduino Core from espressif? I mean, random Exceptions/crashes when using function as they are intended. Isn't that just a result of core bugs?

tablatronix, 2018-03-07:
it goes away when changing various pieces of code, so its probably unrelated, maybe a compiler bug

tablatronix, 2018-03-14:
I posted this to esp32 lib, still no response

aj-ptw, 2018-04-07:
is my issue of not being able to run wifiManager at the same time as a WebServer on port 80 related to this? I tried doing a server.stop() but that does not seem to help.

aj-ptw, 2018-04-07:
Should I open another issue for that?

tablatronix, 2018-04-07:
There are  already issues discussing the exact same thing, you cant run 2 webservers ...

tablatronix, 2018-04-07:
Maybe make a simple sketch and paste it?

tablatronix, 2018-04-07:
Did you test my gist?

aj-ptw, 2018-04-08:
@tablatronix hi!

I just ripped out all the unneeded code to make a simple example!

https://github.com/PushTheWorld/OpenBCI_Air_Shield_Library/blob/esp32/examples/ESP32WiFiManager/EPS32WiFiManager.ino

You can see see in [line 243](https://github.com/PushTheWorld/OpenBCI_Air_Shield_Library/blob/a1f97f92324a5267e0e7b0372bb5b20ec4a12800/examples/ESP32WiFiManager/EPS32WiFiManager.ino#L243) where there is a lambda function call on a server route to start the wifi manager (a flag is set true, 200 response sent to client) then later [in loop](https://github.com/PushTheWorld/OpenBCI_Air_Shield_Library/blob/a1f97f92324a5267e0e7b0372bb5b20ec4a12800/examples/ESP32WiFiManager/EPS32WiFiManager.ino#L361) the wifi manager is started.

anwarbashir, 2018-06-15:
Hi, I am using Platform.IO with development version of WifiManager (tzapu) and WebServer (bbx10). I am exclusively using ESP32. I am able to obtain a WiFi connection, but after accessing via AP the WiFi on my phone does not revert and I have to reset it manually. This is okay but a bit cumbersome. I think on the ESP8266 it reverts automatically. I think that the reason for this behaviour is down to a {ERROR} disconnect configportal - softAPdisconnect FAILED (But I may be wrong)? In any case can someone advise me, is it possible for the phone to reconnect to WiFi afterwards and if so what can I do to enable this functionality. The console log is provided below:

--- Miniterm on COM5  9600,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
b����␂␄ ␃�␗␙�$�d'�␀3v�E (20370) wifi: esp_wifi_disconnect 847 wifi not start
*WM: Disabling STA
*WM: Enabling AP
*WM: StartAP with SSID:  thing2car
*WM: AP has anonymous access!
*WM: AP IP address: 192.168.4.1
*WM: Starting Web Portal
*WM: HTTP server started
*WM: WiFi Scan done
*WM: Config Portal Running, blocking, waiting for clients...
*WM: NUM CLIENTS: 1
*WM: <- HTTP Wifi
*WM: Scan is cached
*WM: 4 networks found
*WM: DUP AP: Virgin Media
*WM: AP:  -45 VM575052-2G
*WM: AP:  -45 Virgin Media
*WM: AP:  -93 VM2135198
*WM: <- HTTP WiFi save
*WM: process connect
*WM: Connecting as wifi client...
*WM: Connecting to new AP: VM575052-2G
*WM: connectTimeout not set, ESP waitForConnectResult...
*WM: Connection result: WL_CONNECTED
*WM: Connect to new AP [SUCCESS]
*WM: Got IP Address:
*WM: 192.168.0.11
*WM: disconnect configportal
*WM: [ERROR] disconnect configportal - softAPdisconnect FAILED
*WM: config portal exiting
connected...yeey
Attempting MQTT connection...connected

tablatronix, 2018-06-15:
When was the last time you pulled in esp32 lib ?

anwarbashir, 2018-06-16:
@tablatronix - it was some time ago, seems from your comment that I need the latest, so I will look at how to download and install the latest. If you could advise a link that would be useful - thanks.

UroojAlam, 2020-06-16:
For me, changing the web server port to 81 and 8080 without using `wifiManager.resetSetting();` worked.