tzapu/WiFiManager #546: Crash setting alternate debug output

liebman, 2018-03-03:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Development

Hardware:			ESP-12f

Core Version:      	latest git

### Description

Changing serial using `WiFiManager wifi(Serial1);` crashes in the destructor.

### Settings in IDE

Module:  			NodeMcu

Additional libraries:  None

### Sketch
Standard auto-connect sketch modified to use Serial1 instead of Serial

```cpp
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager

void setup() {
    // put your setup code here, to run once:
    Serial.begin(115200);

    //WiFiManager
    //Local intialization. Once its business is done, there is no need to keep it around
    Serial1.begin(115200);
    WiFiManager wifiManager(Serial1);
    //reset saved settings
    //wifiManager.resetSettings();
    
    //set custom ip for portal
    //wifiManager.setAPStaticIPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));

    //fetches ssid and pass from eeprom and tries to connect
    //if it does not connect it starts an access point with the specified name
    //here  "AutoConnectAP"
    //and goes into a blocking loop awaiting configuration
    wifiManager.autoConnect("AutoConnectAP");
    //or use this for auto generated name ESP + ChipID
    //wifiManager.autoConnect();

    
    //if you get here you have connected to the WiFi
    Serial1.println("connected...yeey :)");
}

void loop() {
    // put your main code here, to run repeatedly:
    
}
```

### Debug Messages

**messages from Serial:**
```
tail 8
chksum 0x2d
csum 0x2d
va2e7c7d0
~ld
*WM: freeing allocated params! 
*WM: unloading 

Exception (3):
epc1=0x40100326 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40249f7d depc=0x00000000

ctx: cont 
sp: 3fff0350 end: 3fff0710 offset: 01a0

>>>stack>>>
3fff04f0:  00000009 0000000c 3fff0550 3ffe911b  
3fff0500:  3ffe8b43 3fff0550 3fff0600 3fff05b0  
3fff0510:  0000001c 0000000a 3ffef5fc 3ffef5fc  
3fff0520:  00000010 0000001c 00000000 40100980  
3fff0530:  402072c4 402072b1 3fff0550 3ffe8a60  
3fff0540:  402477dd 3fff05b0 3ffef608 3ffe8a60  
3fff0550:  00000010 00000000 3ffe911a 4020cc3c  
3fff0560:  3ffe911b 00000000 3fff05b0 3ffef6e4  
3fff0570:  3fffdad0 0001c200 3fffeff0 401004e1  
3fff0580:  3ffe911a 3ffef3d8 3fff1984 401009a8  
3fff0590:  3ffe911b 3ffef608 3fff0600 4020b714  
3fff05a0:  40247836 3ffe911b 3fff0600 402022d6  
3fff05b0:  3fffdad0 00000013 3ffef608 4020b701  
3fff05c0:  4023c146 00000001 feefeff6 4010053a  
3fff05d0:  3fffdad0 0001c200 3fff0600 4010099c  
3fff05e0:  0001c200 0000001c 3fff0600 40203aef  
3fff05f0:  3fffdad0 0001c200 3ffef608 402021c4  
3fff0600:  00000000 00000000 00060200 00000000  
3fff0610:  feef0000 401074a8 00000000 401074a8  
3fff0620:  00000000 401074a8 00000000 401074a8  
3fff0630:  00000000 401074a8 00000000 401074a8  
3fff0640:  00000000 feef5035 3fff16b4 0000000f  
3fff0650:  00000006 3fff16d4 0000000f 00000000  
3fff0660:  3fff16f4 0000000f 00000000 3fff1714  
3fff0670:  0000000f 00000000 00000000 00000000  
3fff0680:  00000000 00000000 00000001 3fff1734  
3fff0690:  0000000f 00000003 feefef03 00000000  
3fff06a0:  00000000 feefef07 ffffffff 01000001  
3fff06b0:  01010100 01000100 feefef01 3ffe911b  
3fff06c0:  3ffe911b 0000effe 01000000 00000000  
3fff06d0:  00000000 feefeffe feefeffe feef0201  
3fff06e0:  3ffef608 00000000 00000000 feefeffe  
3fff06f0:  feefeffe 00000000 3ffef6dc 4020c5cb  
3fff0700:  feefeffe feefeffe 3ffef6f0 40100a2c  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```
**messages from Serial1:**
```
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
va2e7c7d0
~ld
*WM: AutoConnect 
*WM: Connecting as wifi client... 
*WM: wifi station disconnect 
*WM: Connecting to saved AP 
*WM: enableSTA PERSISTENT ON 
*WM: connectTimeout not set, ESP waitForConnectResult... 
*WM: Connection result: WL_CONNECTED
*WM: lastconxresult: WL_CONNECTED
*WM: IP Address: 192.168.0.17
connected...yeey :)
*WM: freeing allocated params! 

Exception (3):
epc1=0x40100326 epc2=0x00000000 epc3=0x00000000 excvaddr=0x40249f7d depc=0x00000000
sp: 3fff0350 end: 3fff0710 offset: 01a0
3fff04f0:  00000009 0000000c 3fff0550 3ffe911b  
3fff0500:  3ffe8b43 3fff0550 3fff0600 3fff05b0  
3fff0510:  0000001c 0000000a 3ffef5fc 3ffef5fc  
3fff0520:  00000010 0000001c 00000000 40100980  
3fff0530:  402072c4 402072b1 3fff0550 3ffe8a60  
3fff0540:  402477dd 3fff05b0 3ffef608 3ffe8a60  
3fff0550:  00000010 00000000 3ffe911a 4020cc3c  
3fff0560:  3ffe911b 00000000 3fff05b0 3ffef6e4  
3fff0570:  3fffdad0 0001c200 3fffeff0 401004e1  
3fff0580:  3ffe911a 3ffef3d8 3fff1984 401009a8  
3fff0590:  3ffe911b 3ffef608 3fff0600 4020b714  
3fff05a0:  40247836 3ffe911b 3fff0600 402022d6  
3fff05b0:  3fffdad0 00000013 3ffef608 4020b701  
3fff05c0:  4023c146 00000001 feefeff6 4010053a  
3fff05d0:  3fffdad0 0001c200 3fff0600 4010099c  
3fff05e0:  0001c200 0000001c 3fff0600 40203aef  
3fff05f0:  3fffdad0 0001c200 3ffef608 402021c4  
3fff0600:  00000000 00000000 00060200 00000000  
3fff0610:  feef0000 401074a8 00000000 401074a8  
3fff0620:  00000000 401074a8 00000000 401074a8  
3fff0630:  00000000 401074a8 00000000 401074a8  
3fff0640:  00000000 feef5035 3fff16b4 0000000f  
3fff0650:  00000006 3fff16d4 0000000f 00000000  
3fff0660:  3fff16f4 0000000f 00000000 3fff1714  
3fff0670:  0000000f 00000000 00000000 00000000  
3fff0680:  00000000 00000000 00000001 3fff1734  
3fff0690:  0000000f 00000003 feefef03 00000000  
3fff06a0:  00000000 feefef07 ffffffff 01000001  
3fff06b0:  01010100 01000100 feefef01 3ffe911b  
3fff06c0:  3ffe911b 0000effe 01000000 00000000  
3fff06d0:  00000000 feefeffe feefeffe feef0201  
3fff06e0:  3ffef608 00000000 00000000 feefeffe  
3fff06f0:  feefeffe 00000000 3ffef6dc 4020c5cb  
3fff0700:  feefeffe feefeffe 3ffef6f0 40100a2c  

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
```

Decoded stack:
```
Exception Cause: 3  [LoadStoreError: Processor internal physical address or data error during load or store]

0x40100326: check_poison at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c:821
0x40249f7d: sleep_reset_analog_rtcreg_8266 at ??:?
0x40100980: realloc at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c:1720
0x402072c4: ESP8266WiFiSTAClass::begin() at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp:552
0x402072b1: ESP8266WiFiSTAClass::begin() at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/libraries/ESP8266WiFi/src/ESP8266WiFiSTA.cpp:552
0x402477dd: sleep_reset_analog_rtcreg_8266 at ??:?
0x4020cc3c: Print::write(unsigned char const*, unsigned int) at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp:87
0x401004e1: check_poison at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c:857
 (inlined by) check_poison_block at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c:847
0x401009a8: free at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c:1739
0x4020b714: Print::print(char const*) at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp:87
0x40247836: sleep_reset_analog_rtcreg_8266 at ??:?
0x402022d6: DEBUG_WM<const __FlashStringHelper*, char const*> at /Users/chris.l/Documents/Arduino/libraries/WiFiManager/WiFiManager.cpp:1814
0x4020b701: Print::write(char const*) at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp:87
0x4023c146: wifi_get_opmode at ??:?
0x4010053a: get_unpoisoned at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c:942
0x4010099c: free at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c:1726
0x40203aef: WiFiManager::~WiFiManager() at /Users/chris.l/Documents/Arduino/libraries/WiFiManager/WiFiManager.cpp:1814
0x402021c4: setup at /var/folders/d7/fkhv_1ns19d8d3sqsttcrj880000gn/T/arduino_modified_sketch_290966/AutoConnect.ino:28
0x401074a8: sntp_get_current_timestamp at ??:?
0x401074a8: sntp_get_current_timestamp at ??:?
0x401074a8: sntp_get_current_timestamp at ??:?
0x401074a8: sntp_get_current_timestamp at ??:?
0x401074a8: sntp_get_current_timestamp at ??:?
0x401074a8: sntp_get_current_timestamp at ??:?
0x4020c5cb: loop_wrapper at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/core_esp8266_main.cpp:57
0x40100a2c: cont_norm at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/cont.S:109
```

tablatronix, 2018-03-03:
confirmed

tablatronix, 2018-03-03:
pretty sure I am chaining constructors wrong

```C++
// constructors
WiFiManager::WiFiManager(Stream& consolePort):_debugPort(consolePort){
  WiFiManagerInit();
}
```

tablatronix, 2018-03-03:
Someone else can fix this properly, I am not sure the correct syntax for delegated constructors, and I couldn't get it to work.

tablatronix, 2018-03-03:
should work for now at least

liebman, 2018-03-03:
Works for me now.

tablatronix, 2018-03-04:
Thanks, I figured the constructors out, I was doing it backwards, but I am going to leave the init function the way it is, makes it cleaner if I decide to add more constructors.

liebman, 2018-03-04:
I found if you erase the flash to force it to run the captive portal, it still crashes, but not when its been configured before. I used the same sketch but erased the flash first.  Does not happen if `WiFiManager wifiManager(Serial1);` is not used.

```
load 0x4010f000, len 1384, room 16 
tail 8
chksum 0x2d
csum 0x2d
va2e7c7d0
~ld
*WM: AutoConnect 
*WM: Connecting as wifi client... 
*WM: wifi station disconnect 
*WM: No saved credentials, skipping wifi 
*WM: Connection result: WL_NO_SSID_AVAIL
*WM: lastconxresult: WL_NO_SSID_AVAIL
*WM: wifi station disconnect 
*WM: Disabling STA 
*WM: Enabling AP 
*WM: StartAP with SSID:  AutoConnectAP
*WM: AP has anonymous access! 
*WM: SoftAP Configuration 
*WM: -------------------- 
*WM: ssid:             AutoConnectAP
*WM: password:         
*WM: ssid_len:         13
*WM: channel:          1
*WM: authmode:        
*WM: ssid_hidden:     
*WM: max_connection:   4
*WM: beacon_interval:  100(ms)
*WM: -------------------- 
*WM: AP IP address: 192.168.4.1
*WM: setupConfigPortal 
*WM: Starting Web Portal 
*WM: HTTP server started 
*WM: Scan done 
*WM: Config Portal Running, blocking, waiting for clients... 
*WM: NUM CLIENTS: 0 
*WM: -> captive.apple.com 
*WM: <- Request redirected to captive portal 
*WM: <- HTTP Root 
*WM: -> 192.168.4.1 
*WM: Scan done 
*WM: -> captive.apple.com 
*WM: <- Request redirected to captive portal 
*WM: <- HTTP Root 
*WM: -> 192.168.4.1 
*WM: Scan is cached 
*WM: -> captive.apple.com 
*WM: <- Request redirected to captive portal 
*WM: <- HTTP Root 
*WM: -> 192.168.4.1 
*WM: Scan is cached 
*WM: <- HTTP Wifi 
*WM: Scan is cached 
*WM: DUP AP: ThicketLand
*WM: AP:  -29 ThicketLand
*WM: AP:  -33 T3WQBGGB
*WM: AP:  -41 DIRECT-7C-HP ENVY 5660 series
*WM: AP:  -73 TC8717T24
*WM: AP:  -80 FBI Surveillance Van 
*WM: AP:  -80 WIFID216BA
*WM: AP:  -81 Bob's Guest Network
*WM: AP:  -81 Bob's Network
*WM: AP:  -82 Srrrico
*WM: AP:  -87 DIRECT-0WM2070 Series
*WM: AP:  -89 DIRECT-45-HP ENVY 4520 series
*WM: AP:  -91 ZARATE_2GEXT
*WM: Sent config page 
*WM: -> captive.apple.com 
[... 79 lines cut ...]
0x40107198: vPortFree at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/heap.c:40
0x4023c4ba: system_param_load at ??:?
0x4023c486: system_param_load at ??:?
0x4023c4d7: system_param_load at ??:?
0x4020cc3c: std::function<void ()>::swap(std::function<void ()>&) at ??:?
0x401004e1: check_poison at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c:857
 (inlined by) check_poison_block at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c:847
0x401007f2: calloc at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c:1695
0x4020b714: Print::print(int, int) at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/Print.cpp:87
0x40247836: sleep_reset_analog_rtcreg_8266 at ??:?
0x402022d6: std::_Function_base::_Base_manager<std::_Bind<std::_Mem_fn<void (WiFiManager::*)()> (WiFiManager*)> >::_M_manager(std::_Any_data&, std::_Any_data const&, std::_Manager_operation) at /Users/chris.l/Documents/Arduino/libraries/WiFiManager/WiFiManager.cpp:1814
0x4023c3ce: system_param_save_with_protect at ??:?
0x4010053a: get_unpoisoned at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c:942
0x4010099c: free at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/umm_malloc/umm_malloc.c:1726
0x40203aef: WiFiManager::~WiFiManager() at /Users/chris.l/Documents/Arduino/libraries/WiFiManager/WiFiManager.cpp:1814
0x402021c4: setup at ??:?
0x401074a8: sntp_get_current_timestamp at ??:?
0x401074a8: sntp_get_current_timestamp at ??:?
0x401074a8: sntp_get_current_timestamp at ??:?
0x401074a8: sntp_get_current_timestamp at ??:?
0x401074a8: sntp_get_current_timestamp at ??:?
0x401074a8: sntp_get_current_timestamp at ??:?
0x4020c5cb: strnlen_P at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/pgmspace.cpp:148
0x40100a2c: cont_norm at /Users/chris.l/Documents/Arduino/hardware/esp8266com/esp8266/cores/esp8266/cont.S:109
```

tablatronix, 2018-03-04:
Cant reproduce this. Still working for me using staging.

Are you using any parameters?

tablatronix, 2018-03-04:
https://gist.github.com/tablatronix/b966719f8da88f50a5dbb59c9d38e1fb

liebman, 2018-03-05:
Not using any parameters.

tablatronix, 2018-03-14:
Has this been resolved, is this still an issue?

liebman, 2018-08-06:
No longer an issue.

closed 2018-08-06 by liebman