tzapu/WiFiManager #550: How do I connect disconnect esp from a network with out resting it.

Ameengray, 2018-03-08:
The esp is connected to my home network. I used the WiFi manager connected my esp to my home network. Now I want to connect the esp to another network in my home.the only way I can do that is by Turing off my home network.(which is not something I want to do) in such cases how can we have esp8266 start a new access point so that we can connect to another network in the same range ? Help out thanks.

Ameengray, 2018-03-08:
Connect to another access point while esp is already connected to an access point in the same range. How to disconnect and reconnect without turning of the access point esp is is connected to in the first place.

tablatronix, 2018-03-08:
You should be able to just begin(blah)

Ameengray, 2018-03-08:
@tabatronix I did not get you. I fairly new to this. Is there a solution for this issue ?

trailblazer2006, 2018-03-11:
You can adapt [this sample](https://github.com/tzapu/WiFiManager/tree/master/examples/OnDemandConfigPortal) to your scenario.