tzapu/WiFiManager #553: ESP32 Captive Portal not loading, multiple OSes

fcjr, 2018-03-14:
### Basic Infos

Hi,  
I just pulled the latest development build of WiFiManager (as well as the latest ESP32 including the latest DNSServer library (as suggested in https://github.com/espressif/arduino-esp32/issues/1186), however when running The AutoConnect Example (and others) I do not get a captive portal on any device.  On Windows 10 and macOS (both latest) I am able to navigate to 192.168.4.1 or another non https page and it resolves to the ESP32, however no captive portal pops up on load.  On android (8.1.0, latest) the issue seems to be more problematic, the device does not see any internet connection so it defaults to LTE and does not allow navigation to the ESP32 (even though it stays connected to the AP).

Are there some configuration steps I might be missing?

Thanks! - FC

#### Hardware
WiFimanager Branch/Release:     Development

Hardware:     espressif ESP32 DevKitC

Core Version:     latest

tablatronix, 2018-03-14:
Dnsserver is no longer required, use the one in core

fcjr, 2018-03-14:
Hey @tablatronix, thanks for the reply!  I am using the DNSServer library from the core, i am using the latest version of the core and am having the results stated above.

tablatronix, 2018-03-14:
confirmed, ondemand config portal works, autoconnect does not.
Should be somehting dumb, will fix

fcjr, 2018-03-24:
Hi @tablatronix, thanks for your reply.  I realized that the issue on my mac and pc was due to my antivirus blocking captive portals. (Sorry about that!)  However there still seems to be an issue with android devices finding the captive portal.  I've tried on both a Google Pixel and a Galaxy S8. 

The Debug output simply states the following on connect:
*WM: NUM CLIENTS: 0
*WM: NUM CLIENTS: 1

The devices do seem to be getting the dns information properly however (see examples from pixel at end of this post).  When attempting to navigate to a non https page (using chrome), or even directly via 192.168.4.1, the phones don't seem to even be making requests to the ESP.  I have tested the devices on other captive portal systems (such as the guest wifi functionality on my home router, and both android phones pick up those portals fine).

If there is any more information I may be able to give you to help determine the problem please let me know!

[image]
[image]
[image]

tablatronix, 2018-03-24:
I do not have an android device to test unfortunately. I will try to obtain one.
When I had that problem it was a dns issue or firewall issue, Ill take a look at wireshark and see if everything looks ok, but its working for me everytime.

treii28, 2018-04-01:
I've also been experiencing this trouble for some time. I have plenty of extra systems, wifi adapters, etc. around as well as multiple android and other devices. (no iOS but I can gain access to some if needed). I've been needing a pop-up style portal for over a year and have been searching and/or posting occasionally to try to find a solution.
I see the 'help wanted' and 'Needs Feedback'. I've only tinkered with wireshark a little, but if you can give me an idea what you want to see, I will do anything I can to get you data.

nicogon, 2018-04-02:
On android only works if you try to open a http page  (not https). And is not automatic as in ios or a pc.
I've been researching about this, and it seems the problem is that android dont like that captive portal ip is in the same network. i've tested with a functional captive portal that dns resolves correctly (for android captive  https://www.gstatic.com/generate_204 as an external ip) but it seems there is some kind of http redirection of the request to the captive portal.

nicogon, 2018-04-02:
dns service should give an external ip and then do some kind of http redirection (that i havent been able to do it with esp8266). Do you know if there is a way of intercepting http packet to an external ip and deliver a 302 redirection to the captive portal?

tablatronix, 2018-04-02:
Yeah I will keep tracking this here https://github.com/espressif/arduino-esp32/issues/1037#issuecomment-377962531
no need to continue posting here, as this is clearly not a wm issue.

I will keep it open in case we can find a workaround or solution for us, then update the captiveportal examples and vice versa.

I don't understand why this stuff is hard to come by, there should just be a developer whitepaper on this for andoid that clearly specifies what the deal is and how to get captive portals working.

JohnnyPP, 2018-04-22:
@nicogon How did you exactly do it? I have been struggling with this problem for about two months.

treii28, 2018-04-29:
I think the gist of it is that whatever the device will reject local IP dns responses for addresses it knows - or at least rejects local ips it knows for something it knows should be on the internet at large. I've been trying to find either a code example or even a written explanation of just what things like OpenWRT do to make them work. My suspicion is that they don't simply redirect DNS like the arduino/esp code used to do but instead it actually intercepts internet traffic going to any other address than it's own and answers it as though it were the address in question.
The ESP code core has lwip in with it to handle the IP stack. If I could find a good code example somewhere, I might be able to stab blindly around in the dark with my limited knowledge of C++ to try to do something similar until a given client was properly re-directed.

tablatronix, 2018-06-29:
Anyone figure anything new out?

mattncsu, 2018-08-06:
On your Android device, go to Advanced wifi options and disable the option to "Switch to mobile data - Switch to mobile data whenever your Wi-Fi internet connection is slow or unstable."

kbickham, 2019-08-20:
I got this working by setting ip to 8.8.8.8 ot 8.8.4.4 and using port 53

nabeelkirmani, 2019-11-26:
_Newbie here, hence the non-standard way of posting solution.._

**Workaround** 

In file [WiFiManager.cpp](https://github.com/tzapu/WiFiManager/blob/master/WiFiManager.cpp), in function `void WiFiManager::handleNotFound()` replacing the `400` response code to `200` at [Line 738](https://github.com/tzapu/WiFiManager/blob/master/WiFiManager.cpp#L738) i.e.
 
replacing `server->send ( 404, "text/plain", message );` 
with `server->send ( 200, "text/plain", message );`

solves the issue of pop-up not appearing on my _Windows 10_ device. I tested this on the [AutoConnect](https://github.com/tzapu/WiFiManager/blob/master/examples/AutoConnect/AutoConnect.ino) example in [Sloeber IDE](https://eclipse.baeyens.it/nightly.php) on [NodeMCU 1.0](https://en.wikipedia.org/wiki/NodeMCU) board.

I don't know why it worked but it did, and if someone can explain kindly do.

Whereas for my _Android P_ & _Windows 10_ devices,
[CaptivePortal.ino](https://github.com/esp8266/Arduino/blob/master/libraries/DNSServer/examples/CaptivePortal/CaptivePortal.ino) sketch works and brings up the pop-up captive portal.

tablatronix, 2019-11-26:
Thats interesting indeed, all requests should return the portal if its active..
Do you have or can you obtain serial logs of this ?

That handler should never be reached.

nabeelkirmani, 2019-11-27:
On building the project in the previously mentioned environment -
```
10:33:20 **** Incremental Build of configuration Release for project tzapu ****
"C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Do all objcopy commands'
"C:\Sloeber\arduinoPlugin\packages\esp8266\tools\python3\3.7.2-post1/python3" "C:\Sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.6.1/tools/elf2bin.py" --eboot "C:\Sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.6.1/bootloaders/eboot/eboot.elf" --app "C:\Users\nabeel\Documents\sloeber-wrkspc\tzapu/Release/tzapu.elf" --flash_mode dio --flash_freq 40 --flash_size 4M --path "C:\Sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506/bin" --out "C:\Users\nabeel\Documents\sloeber-wrkspc\tzapu/Release/tzapu.bin"
Creating BIN file "C:\Users\nabeel\Documents\sloeber-wrkspc\tzapu/Release/tzapu.bin" using "C:\Users\nabeel\Documents\sloeber-wrkspc\tzapu/Release/tzapu.elf"
"C:\Sloeber\arduinoPlugin\packages\esp8266\tools\python3\3.7.2-post1/python3" "C:\Sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.6.1/tools/signing.py" --mode sign --privatekey "/private.key" --bin "C:\Users\nabeel\Documents\sloeber-wrkspc\tzapu/Release/tzapu.bin" --out "C:\Users\nabeel\Documents\sloeber-wrkspc\tzapu/Release/tzapu.bin.signed" --legacy "C:\Users\nabeel\Documents\sloeber-wrkspc\tzapu/Release/tzapu.bin.legacy_sig"
"C:\Sloeber\arduinoPlugin\packages\esp8266\tools\python3\3.7.2-post1/python3" "C:\Sloeber\/arduinoPlugin/packages/esp8266/hardware/esp8266/2.6.1/tools/sizes.py" --elf "C:\Users\nabeel\Documents\sloeber-wrkspc\tzapu/Release/tzapu.elf" --path "C:\Sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506/bin"
Executable segment sizes:

IROM   : 276360          - code in flash         (default or ICACHE_FLASH_ATTR) 

IRAM   : 26716   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 

DATA   : 1244  )         - initialized variables (global, static) in RAM/HEAP 

RODATA : 2068  ) / 81920 - constants             (global, static) in RAM/HEAP 

BSS    : 29360 )         - zeroed variables      (global, static) in RAM/HEAP 

'Finished building: tzapu.hex'
' '
'Building target: tzapu'
'Printing size:'
"C:\Sloeber\arduinoPlugin\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-4-b40a506/bin/xtensa-lx106-elf-size" -A "C:/Users/nabeel/Documents/sloeber-wrkspc/tzapu/Release/tzapu.elf"
C:/Users/nabeel/Documents/sloeber-wrkspc/tzapu/Release/tzapu.elf  :
section                                                                                                                                                      size         addr
.data                                                                                                                                                        1244   1073643520
.noinit                                                                                                                                                         4   1073644764
.text                                                                                                                                                         252   1074790400
.irom0.text                                                                                                                                                276360   1075843088
.text1                                                                                                                                                      26464   1074790652
.rodata                                                                                                                                                      2068   1073644768
.bss                                                                                                                                                        29360   1073646840
.debug_frame                                                                                                                                                47664            0
.debug_info                                                                                                                                               1152832            0
.debug_abbrev                                                                                                                                              121376            0
.debug_aranges                                                                                                                                              13344            0
.debug_ranges                                                                                                                                               30688            0
.debug_line                                                                                                                                                276298            0
.debug_str                                                                                                                                                 232529            0
.comment                                                                                                                                                     4680            0
.xtensa.info                                                                                                                                                   56            0
[... cut ...]
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
```
My default browser opens automatically and -
```
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
```
it redirects to the captive portal.

On opening HTTPS and HTTP links, it is the same story as with the Android Pie device.

mariano22, 2021-09-01:
I found a similar behaviour when using WifiServer library:

When providing HTTP server for captive portal you generally have two options:

- Using WebServer class and set handlers for roots. The parsing and handling of HTTP header is done by the library.

> Use WifiServer and receive the HTTP message with header included and handler everything by your own.

If you are doing number 1 and pretend using by connecting with an Android device I suggest adding a handler for the error case showing the captive portal:

server.onNotFound(handleRoot);

I noticed that for Android 10, to properly detect the captive portal and make the phone to show "Touch to log in the network" message is necesary. If not you must navigate to any HTTP address and let the DNS do the job for you (not HTTPS).

tablatronix, 2021-09-02:
So is this a problem with these dns failing and refusing to redirect? Ill look again but I am still confused how we are hitting the 404