tzapu/WiFiManager #554: Has anyone tested WiFiManager with ESP8266 board manager ver 2.4.1?

bill-orange, 2018-03-14:
Using a WEMOS D1 Mini Pro with WiFiManager after a clean full erase of flash, WiFi manager does not launch AP mode but launched STA mode with IP 0.0.0.0.  I redacted board manager 2.4.1 to 2.3 and my code worked again.

Can anyone reproduce this?

tablatronix, 2018-03-14:
not seeing that either

EveningStarNM, 2018-03-30:
I've used the v0.12.0 library in the Arduino IDE on Wemos D1 R2 and Wemos D1 Mini. It works, if you workaround some bugs that may occur when using char* arrays to set the SSID and password for the access point. (Don't try to do any operations on char* arrays, such as strcat, or sprintf. If you need such functionality, specify a length for the arrays when declaring them, and make sure that the length of the password array is between 8 and 63 characters.)